espressif/esp-idf #7064: 4.3 / master - ble_gap_master_set_timer is not defined when configuring nimBLE for peripheral only

bpietsch, 2021-05-24:
On both the release/4.3 and master branches, I have my project configured to use nimBLE with peripheral mode enabled and central mode disabled in my sdkconfig. I get the following build failure because ble_gap_master_set_timer is only defined when central mode is enabled (error message below is from master (c13afea635adec735435961270d0894ff46eef85).

/Users/brian/Tools/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:4616:9: error: implicit declaration of function 'ble_gap_master_set_timer'; did you mean 'ble_gap_master_timer'? [-Werror=implicit-function-declaration]
         ble_gap_master_set_timer(duration_ticks);
         ^~~~~~~~~~~~~~~~~~~~~~~~
         ble_gap_master_timer

RahulEspressif, 2021-05-24:
Hi @bpietsch  , Thanks for reporting this issue. Can you please share your sdkconfig used to reproduce this issue ?

bpietsch, 2021-05-24:
Sure thing, here you go!
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/6529950/sdkconfig.txt)

RahulEspressif, 2021-05-24:
Hi @bpietsch  .. As you are disabling central role, CONFIG_BT_NIMBLE_ROLE_OBSERVER  too should be disabled.  Please disable this option from menuconfig and confirm if you still observe the issue.

bpietsch, 2021-05-24:
Yes, that resolved the issue, thank you. Given that I made no changes to the sdkconfig when updating from 4.2 (where it compiled fine), I feel there should be a change here as there is simply no way I could have known this. Also, it shouldn't be that easy to set two sdkconfig options that are inherently incompatible with each other. Perhaps a compile time check for the incompatible options with a #error directive. Thank you for the extremely quick response and I appreciate your help! Thank you.

closed 2021-05-31 by Alvin1Zhang