espressif/esp-idf #7065: nimble crash on sdk v4.2.1 &v4.3 beta3 when executing ble_gap_conn_rssi

zhongpang, 2021-05-24:
Hi esp team!
I developed a BLE central application on ESP32 based nimble in past year, the nimble worked well until I upgraded the SDK to 4.2 (I also tried v4.3 beta 3, the result was the same).
My device supported to connect to multiple BLE peripherals concurrently, the problem is: After established one connection with a BLE Device, esp32 doing ble_gap_conn_rssi regularly(10 seconds) on the device, and sometimes, the ble_gap_conn_rssi caused the esp32 crash! Here's the output of the trace log, for reference:
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x401bf5da  PS      : 0x00060530  A0      : 0x801191fc  A1      : 0x3fff4690  
0x401bf5da: put_le16 at /Users/zhonghui/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/src/endian.c:28

A2      : 0x00000001  A3      : 0x00000c35  A4      : 0x00000000  A5      : 0x00060323  
A6      : 0x00000001  A7      : 0x3ffb00cc  A8      : 0x801206e2  A9      : 0x3fff4670  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3fff3400  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000001  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace:0x401bf5d7:0x3fff4690 0x401191f9:0x3fff46b0 0x401193e5:0x3fff46d0 0x401188ad:0x3fff46f0 0x401188cf:0x3fff4720 0x4011eaae:0x3fff4740 0x400d9c35:0x3fff4760 0x40092919:0x3fff4780
0x401bf5d7: put_le16 at /Users/zhonghui/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/src/endian.c:24

0x401191f9: ble_hs_hci_cmd_send at /Users/zhonghui/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_cmd.c:80

0x401193e5: ble_hs_hci_cmd_send_buf at /Users/zhonghui/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_cmd.c:125

0x401188ad: ble_hs_flow_tx_num_comp_pkts at /Users/zhonghui/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_flow.c:99

0x401188cf: ble_hs_flow_event_cb at /Users/zhonghui/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_flow.c:120

0x4011eaae: ble_npl_event_run at /Users/zhonghui/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:121
 (inlined by) nimble_port_run at /Users/zhonghui/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:81

0x400d9c35: blecent_host_task at /Users/zhonghui/Source/manding/mbox/mbox_esp32/main/Task_Device_NB.c:2305

0x40092919: vPortTaskWrapper at /Users/zhonghui/esp/esp-idf/components/freertos/xtensa/port.c:143

RahulEspressif, 2021-05-25:
Hi @zhongpang , Thanks for reporting the issue.

StoreProhibited is something trying to write to non-existent memory, like accessing a NULL pointer. I tried applications of 4.3 sdk by modifying it to call ble_gap_conn_rssi . But i couldn't reproduce the issue.   Can you please:

1. Share some snippet of the example code that i can try to reproduce this issue ?
2. It is mentioned that in latest SDK versions you are observing the issue. May i know which previous SDK versions were not showing this behaviour for the same application at your end ?

zhongpang, 2021-05-27:
Hi @RahulEspressif,

Thanks for the reply!
1. Here the sigle line code which randomly caused the crash:
   ret =   ble_gap_conn_rssi(dev_config->conn_id,
                            &dev_config->value_signal);
2. In the call stack, the line in my App 2305 is: nimble_port_run();
3. By disable "Enable Host Flow control" in menuconfig, the problem disappeard.
4. In this project, I used esp-idf started from 4.0, bluedroid stack, then nimble stack, it was stable on 4.1 in my memory, and started to encounter various of problems start from 4.2 rc, my device is a BLE -> 4G hub, so besides the niuble problem, I also been blocked by uart2 problem(which used to drive my 4G module...) for couple days.
Plese let me know if you need more information, thx.

RahulEspressif, 2021-05-27:
Hi @zhongpang  , 

>> By disable "Enable Host Flow control" in menuconfig, the problem disappeard.

There is a known issue in recent releases causing unexpected protocol behaviours. We suggest to either use this option ( of disabling host flow control ) OR wait for the next official release where we have added fix for same.

Thanks,
  Rahul

AxelLin, 2021-05-28:
@RahulEspressif 
Is this the same issue as https://github.com/espressif/esp-idf/issues/7026 ?
If yes, the release/v4.2 branch already has the fix for testing.

dhrishi, 2021-06-18:
@zhongpang ESP-IDF v4.2.2 is released that has the fix for NimBLE and it is functional as expected. You can find it here: https://github.com/espressif/esp-idf/releases/tag/v4.2.2

Please give it a try and let us know in case of any issues

closed 2021-08-18 by Alvin1Zhang