espressif/esp-idf #7066: Can not connect  ESP32-C3 ble when enable modem sleep

u-tec-com, 2021-05-24:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-C3-MINI-1]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.0.1
- Operating System:     [Windows]
- (Windows only) environment type: [PowerShell].
- Using an IDE?: [VSCODE]
- Power Supply:         [USB]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior
Connect the device when the device in ble with modem sleep
### Actual Behavior
Can not connect to device with modem sleep. and works well without modem sleep
### Steps to reproduce

1. step1
Enable BLE function and start advertising.
Check the advertise in Mobile App (BLE Scanner), this is ok.
Connect the device 
fail...

u-tec-com, 2021-05-25:
set max_freq_mhz and min_freq_mhz with 160MHZ in pm_config, with modem sleep.
the connection is ok now

    pm_timer_args.arg = NULL;
    pm_timer_args.callback = pm_timer_callback;
    esp_timer_create(&pm_timer_args, &pm_timer);

#if CONFIG_PM_ENABLE
    // Configure dynamic frequency scaling:
    // maximum and minimum frequencies are set in sdkconfig,
    // automatic light sleep is enabled if tickless idle support is enabled.
#if CONFIG_IDF_TARGET_ESP32
    esp_pm_config_esp32_t pm_config = {
#elif CONFIG_IDF_TARGET_ESP32S2
    esp_pm_config_esp32s2_t pm_config = {
#elif CONFIG_IDF_TARGET_ESP32C3
    esp_pm_config_esp32c3_t pm_config = {
#endif
            .max_freq_mhz =  CONFIG_ESP32C3_DEFAULT_CPU_FREQ_MHZ,
            .min_freq_mhz =  CONFIG_ESP32C3_DEFAULT_CPU_FREQ_MHZ, 
#if CONFIG_FREERTOS_USE_TICKLESS_IDLE
           .light_sleep_enable = true
#endif
    };
    ESP_ERROR_CHECK( esp_pm_configure(&pm_config) );

AxelLin, 2021-05-25:
> ## Environment
> * Development Kit:      [ESP32-DevKitC]
> * Kit version (for WroverKit/PicoKit/DevKitC): [v4]
> * Module or chip used:  [ESP32-C3-MINI-1]
> * IDF version (run `git describe --tags` to find it):
>   // v3.2-dev-1148-g96cd3b75c

I don't think you can use v3.2-dev for ESP32-C3.
What is the **git describe --tags** output?

u-tec-com, 2021-05-25:
> I don't think you can use v3.2-dev for ESP32-C3.
> What is the **git describe --tags** output?

Sorry, the code is master branch. v4.4-dev-1406-g59a723854
I has fix it in bug report

xiewenxiang, 2021-06-09:
@u-tec-com 

Version: v3.2-dev-1148-g96cd3b75c , is three years ago, it's don;t support ESP32C3.
```c
commit 96cd3b75cd81d4a778bdd430424012f583e1d4c8 (HEAD -> master)
Merge: 66fd7e9cc3 5585f699d1
Author: Angus Gratton <angus@espressif.com>
Date:   Tue Sep 25 08:24:29 2018 +0800

    Merge branch 'feature/mbedtls_mem_alloc_options' into 'master'
    
    mbedtls: configurable options for controlling dynamic memory allocations
    
    See merge request idf/esp-idf!3343

```

version: v4.4-dev-1406-g59a723854.

I can't find this commit in ESP-IDF, could you please post the link to this commit.

Alvin1Zhang, 2021-08-18:
@u-tec-com Would you please help share if any updates or details for this issue? Thanks.

u-tec-com, 2021-08-18:
with external 32k crystal, the connection is more stable. thanks.

closed 2021-08-18 by u-tec-com