espressif/esp-idf #7067: MQTT client Non-ssl scheme error

sumathi98, 2021-05-25:
## Environment

    Development Kit : ESP32-DevKitC
    Module or chip used: ESP32-WROOM-32U
    IDF version : v4.4.0
    Build System : idf.py
    Compiler version : 8.4.0
    Operating System : Linux
    Using an IDE? : No
    Power Supply : external 5V

## Problem Description

Iam trying to connect esp32 to Node-red server over MQTT protocol,  I need Mqtt with tls (secured) and using ssl_mutual_auth example as base code. I have replaced my client.key, client.crt, ca.crt and ca.key files also changed the host IP and publish/subscribe topics in main file. Iam able to connect to wifi network but unable to connect to Mqtt server, getting below error.

## Debug Logs
<27>[0;32mI (3629) MQTTS_EXAMPLE: [APP] Free memory: 234076 bytes<27>[0m<\r><\n>
<27>[0;33mW (3629) MQTT_CLIENT: SSL related configs set, but the transport protocol is a non-SSL scheme, transport = 0<27>[0m<\r><\n>
<27>[0;32mI (3639) MQTTS_EXAMPLE: Other event id:7<27>[0m<\r><\n>

chegewara, 2021-05-25:
Please post mqtt configuration here.

sumathi98, 2021-05-25:
Below is the mqtt configuration,
const esp_mqtt_client_config_t mqtt_cfg = {
        .host = "54.215.xxx.xxx",
        .client_cert_pem = (const char *)client_cert_pem_start,
        .client_key_pem = (const char *)client_key_pem_start,
        .cert_pem = (const char *)server_cert_pem_start,
    };

chegewara, 2021-05-25:
From IP app cant get what transport you are using. Try to add this:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/mqtt.html#_CPPv4N24esp_mqtt_client_config_t9transportE

sumathi98, 2021-05-25:
Thanks, have added other configs,
        .transport = 2,
	.port = xxxx,
	.client_id = "NODE_RED"
Certificates are verified and then connection fails...

DEBUG LOG:
<27>[0;32mI (5129) MQTTS_EXAMPLE: [APP] Free memory: 234092 bytes<27>[0m<\r><\n>
<27>[0;32mI (5139) MQTTS_EXAMPLE: Other event id:7<27>[0m<\r><\n>
W (6899) wifi:<ba-add>idx:0, tid:0, ssn:12, winSize:64<\r><\n>
<27>[0;31mE (7629) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x50<27>[0m<\r><\n>
<27>[0;32mI (7629) esp-tls-mbedtls: Certificate verified.<27>[0m<\r><\n>
<27>[0;31mE (7629) esp-tls: Failed to open new connection<27>[0m<\r><\n>
<27>[0;31mE (7639) TRANSPORT_BASE: Failed to open a new connection<27>[0m<\r><\n>
<27>[0;31mE (7639) MQTT_CLIENT: Error transport connect<27>[0m<\r><\n>
<27>[0;32mI (7649) MQTTS_EXAMPLE: MQTT_EVENT_ERROR<27>[0m<\r><\n>
<27>[0;31mE (7649) MQTTS_EXAMPLE: Last error reported from esp-tls: 0x8010<27>[0m<\r><\n>
<27>[0;31mE (7659) MQTTS_EXAMPLE: Last error reported from tls stack: 0x50<27>[0m<\r><\n>
<27>[0;32mI (7659) MQTTS_EXAMPLE: Last errno string (Success)<27>[0m<\r><\n>
<27>[0;32mI (7669) MQTTS_EXAMPLE: MQTT_EVENT_DISCONNECTED<27>[0m<\r><\n>

sumathi98, 2021-05-26:
Need help for troubleshooting "MBEDTLS_ERR_SSL_FATAL_ALERT_MESSAGE (-0x7780)",

Debug log:
<27>[0;31mE (40165) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780<27>[0m<\r><\n>
<27>[0;32mI (40165) esp-tls-mbedtls: Certificate verified.<27>[0m<\r><\n>
<27>[0;31mE (40165) esp-tls: Failed to open new connection<27>[0m<\r><\n>
<27>[0;31mE (40175) TRANSPORT_BASE: Failed to open a new connection<27>[0m<\r><\n>
<27>[0;31mE (40185) MQTT_CLIENT: Error transport connect<27>[0

chegewara, 2021-05-26:
https://forums.mbed.com/t/error-0x7780-during-handshake/6883
https://www.google.com/search?q=0x7780&oq=0x7780&aqs=chrome..69i57&sourceid=chrome&ie=UTF-8

david-cermak, 2021-05-29:
Hi @sumathi98 

Macro `MBEDTLS_SHA256_C ` is enabled by default in IDF (see https://github.com/espressif/esp-idf/blob/master/components/mbedtls/port/include/mbedtls/esp_config.h#L2046), but I think this is unrelated, as your problem likely points a key exchange negotiation.

As @chegewara noted above, the error `MBEDTLS_ERR_SSL_FATAL_ALERT_MESSAGE` means that the server and the client didn't find a good common cypher for key exchange. To find out more, we would have to know which methods the client (ESP32) offers and which are supported by your server.

* As for the clients methods, you can either check your sdkconfig (look for `CONFIG_MBEDTLS_KEY_EXCHANGE...` authentication modes that are turned on), or enable mbedtls debug logs (component-config -> mbedtls -> `MBEDTLS_DEBUG`)  or open a wireshark and simple checkout the `Client Hello` message, where all the offered cyphersuites are listed.
* As for the servers methods, you can use some online SSL service (e.g. https://www.ssllabs.com/ssltest/) if you're using a public server to connect (or check your configuration if you're running a local server as your mqtt broker)

Alvin1Zhang, 2021-06-30:
@sumathi98 Thanks for reporting. Would you please help share if any updates or details for this issue? Thanks.

sumathi98, 2021-07-05:
@Alvin1Zhang  No I wasn't able to fix the issue. For now Iam using mqtt with TCP, but in future I will need to implement mqtt with TLS.

david-cermak, 2021-07-26:
@sumathi98 Could you please follow the steps from https://github.com/espressif/esp-idf/issues/7067#issuecomment-850844781 and share the results?

Alvin1Zhang, 2021-08-31:
@sumathi98 Thanks for reporting. Do you have any updates per the suggestion by @david-cermak ? Thanks.

Alvin1Zhang, 2021-09-07:
Thanks for reporting, would close due to short of feedback, feel free to reopen.

closed 2021-09-07 by Alvin1Zhang