espressif/esp-idf #7069: problem with length from esp_http_client_read_response(...) in esp_client module

pdemianczuk07, 2021-05-25:
## Environment

- Module or chip used:  [ESP32-WROOM-32D|
- IDF version v4.4-dev-1254-g639e7ad49
## Problem Description
Hi,
I use example eth basic with W5500 by SPI; I use local network with router
I have tried read data from server (local on my computer) ;

On 4 attempt is everything ok
on 5 attempt I have read inncorect part of data and I cannt contiunue process

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"
#define MAX_HTTP_RECV_BUFFER 512*2*91
#define LEN_BUFF 1024*8
char buff[LEN_BUFF] = {0};
char buffer[(MAX_HTTP_RECV_BUFFER + 1)] = {0};
void http_get_frame3_stream(char *tst_received, short size_send){

    esp_http_client_config_t config3 = {
    	.host = ADRES_HOST,
		.path = WEB_URL,
		.port = ADRES_PORT,
		.buffer_size = 10*1024,
    };
    esp_http_client_handle_t client3 = esp_http_client_init(&config3);
    esp_err_t err;
    if ((err = esp_http_client_open(client3, 0)) != ESP_OK) {
        ESP_LOGE(TAG, "Failed to open HTTP connection: %s", esp_err_to_name(err));
        return;
    }
    int content_length =  esp_http_client_fetch_headers(client3);
    ESP_LOGI(TAG, "esp_http_client_fetch_headers: %d B", content_length );
    int total_read_len = 0, read_len;
    int pack_len = 0;
    while ( content_length != total_read_len){

    	if ((content_length- total_read_len) > LEN_BUFF){
    		pack_len = LEN_BUFF;
    	}
    	else {
    		pack_len = content_length- total_read_len;
    	}
    	ESP_LOGI(TAG, "try read_len = %d \t", pack_len);
        read_len = esp_http_client_read_response(client3, buff, pack_len);
        if (read_len <= 0) {
            ESP_LOGI(TAG, "Error read data ; err : %x" , read_len);
        }
        buffer[read_len] = 0;
        ESP_LOGI(TAG, "read_len = %d", read_len);
        total_read_len +=read_len;
    }
	ESP_LOGI(TAG, "HTTP Stream reader Status = %d, content_length = %d",
                   esp_http_client_get_status_code(client3),
                   esp_http_client_get_content_length(client3));

    esp_http_client_close(client3);
    esp_http_client_cleanup(client3);

}

called:

	int counter = 0;
		while (1==1){
			printf("%05d ", ++counter);
			http_get_frame3_stream(test, test_size);

			vTaskDelay(500 / port);
		}
```

[image]

ryan4volts, 2021-06-03:
I (think I) found a similar issue. v4.1.1 works fine, though v4.2 and greater do not. Can you test this with v4.1.1 and see if you get the same results?

shubhamkulkarni97, 2021-07-14:
@pdemianczuk07 @ryan4volts Is this issue reproducible with WiFi interface or only with ethernet?

Alvin1Zhang, 2021-08-09:
@pdemianczuk07 Thanks for reporting and will close due to short of feedback, feel free to reopen wit more updates. Thanks.

closed 2021-08-09 by Alvin1Zhang