espressif/esp-idf #7070: Get Common Name from a certificate loaded in ESP32 automatically.

nexstem-siddhant, 2021-05-25:
Is there a method via which I can extract the CN from the certificate in esp directly, instead of manually passing it. Every single device that I have needs to have a separate device certificate and the device name on Azure IOT hub is derived from the common name of the certificate in the device. It is becoming very tedious now, since I have to keep track of more than 50 certificates and their common names, and have to manually change it the firmware before flashing the program onto esp.

A alternative that I am currently exploring is that i create a NVS partition which contains the common name. I can then use the mass_mfg tool to create different nvs partition files for each device using the csv, which is automatically generated by a python script which loops through all the certificates and gets their common name and stores it in the csv.

negativekelvin, 2021-05-25:
Mbedtls has x509 certificate parsing functions

mahavirj, 2021-06-04:
@nexstem-siddhant 

> Is there a method via which I can extract the CN from the certificate

Please see following pointer that may help:

https://github.com/espressif/esp-idf/blob/c13afea635adec735435961270d0894ff46eef85/components/coap/port/coap_mbedtls.c#L301

>  I can then use the mass_mfg tool to create different nvs partition files for each device using the csv

This sounds all good and something that we would also recommend.

mahavirj, 2021-06-08:
Hopefully that answers your question. Please feel free to re-open if you need additional information or raise a topic on esp32.com forum

closed 2021-06-08 by mahavirj