espressif/esp-idf #7071: PPPoS Uart baudrate 921600 APB_CLK

persan666, 2021-05-25:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version (run ``git describe --tags`` to find it):
    // v4.3-beta3-195-g6be10fab0
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [Yes (Eclips)]
- Power Supply:         [Battery]

## Problem Description
 I am using a modified version of the PPPoS example and trying to get baud rate 921600 to work when communicating with a U-Blox SARA-R5 modem.

The PPPos example uses the following settings for the UART clock (UART_SCLK_REF_TICK):

```cpp
    /* Config UART */
    uart_config_t uart_config = {
        .baud_rate = config->baud_rate,
        .data_bits = config->data_bits,
        .parity = config->parity,
        .stop_bits = config->stop_bits,
#if CONFIG_IDF_TARGET_ESP32 || CONFIG_IDF_TARGET_ESP32S2
        .source_clk = UART_SCLK_REF_TICK,
#else
        .source_clk = UART_SCLK_XTAL,
#endif
        .flow_ctrl = (config->flow_control == MODEM_FLOW_CONTROL_HW) ? UART_HW_FLOWCTRL_CTS_RTS : UART_HW_FLOWCTRL_DISABLE
    };
```

But using this setting, I get an un-accurate baud rate.

Instead I tried changing the clock source to APB_CLK and the the baud rate is then correct but then I get the following error due to that the second parameter to this function is 1, changing it to 0 do not cause the error but then I get un-correct UART readings :
```cpp
uart_set_rx_timeout(esp_dte->uart_port, 1); 
```
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x00000000  PS      : 0x00060230  A0      : 0x00000000  A1      : 0x3ffdefe0
A2      : 0x3ffdcd78  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x3f4160c4
A6      : 0x3ffdf214  A7      : 0x00000000  A8      : 0x800f53fe  A9      : 0x3ffdef90
A10     : 0x3ffdcedc  A11     : 0x00000004  A12     : 0x00000000  A13     : 0xffffffff
A14     : 0x00000004  A15     : 0x3ffc5350  SAR     : 0x00000010  EXCCAUSE: 0x00000014
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff

Backtrace:0xfffffffd:0x3ffdefe0

ELF file SHA256: 8cecbba466a8a8c8

Rebooting...

persan666, 2021-05-27:
@AxelLin I saw from another issue [4838](https://github.com/espressif/esp-idf/issues/4838) that you also had problem using baud rate 921600. Did you manage to get it to work? Did you use it together with pattern detection?

AxelLin, 2021-05-28:
@persan666 

The uart baudrate 921600 APB_CLK is mostly working for me on ESP32-WROOM-32E.
I said **mostly** working, because sometimes the uart still receives unexpected data.
I didn't report the issue because it's difficult to reproduce, but as far as
I can tell, receving wrong data can happen with baudrate 921600.
It's difficult to reproduce, as a result it's difficult to verify if a fix really helps.

ESP-YJM, 2021-06-18:
@persan666 Sorry for reply late. From the description, this is a pure UART problem. You can bring out the PPPoS module and run the UART example separately for verification. I verified that the baud rate is 921600, the soure clock is set to APB_CLK(UART_SCLK_REF_TICK only support at most 115200), and the uart_events example is OK, and the pattern can be detected.

Alvin1Zhang, 2021-06-30:
@persan666 Thanks for reporting. Would you please help share if any updates or details for this issue? Thanks.

persan666, 2021-06-30:
This issue can be closed. I got it working in terms of good buad rate accuracy but I have realized that I need to utilize hardware flow control in order to get the communication stable with the modem at 921600 especially when doing firmware updates. The error was caused by something in my own code.

closed 2021-06-30 by Alvin1Zhang