espressif/esp-idf #7072: PPPoS LCP not working when trying to interrupt connection in the cellular session

JosueSanchezEZM, 2021-05-25:
Hi everybody. I am trying to interrupt the cellular session I already have up. After removing the antenna from my module hardware, I can see that the connectivity is lost from my server. But, after a few minutes, no event is triggered to notify my module (client) using LCP reply that my connection was lost.
Here you can see the connection test I did. The time-outs lines show the moment without the antenna.
[image]

What I would like to know is how can I handle this Kind of connectivity loss. I have already registered de event "IP_EVENT_PPP_LOST_IP" and this event is triggered fine just when I remove the sim card from my sim card slot. I think removing the antenna should also trigger this event. Could you please provide me any feedback or comments about my test?. I will very glad to know your suggestions.

david-cermak, 2021-05-29:
Hi @JosueSanchezEZM 

Could you please try to enable the 
* `LWIP_PPP_NOTIFY_PHASE_SUPPORT`
* `LWIP_PPP_DEBUG_ON`
And share your logs?

The correct way to handle the signal lost, in my opinion, is to register to some PPP protocol phase event (exit of network phase perhaps), but this I think depends on your device (seen devices to retry and re-enter the network phase).

Please note, that the LCP is a link protocol which still operates normally on signal loss. But removing the SIM card brings about a disconnection and consequent exit of PPP mode and thus LCP protocol stops responding (but you'd get a disconnection error before LCP echo fails)

ESP-YJM, 2021-06-16:
@JosueSanchezEZM Is there any update for david's suggestion

JosueSanchezEZM, 2021-06-16:
Thanks for the support I already solved the issue. It was my cellular module configuration.

closed 2021-06-16 by JosueSanchezEZM