espressif/esp-idf #7073: unknown 16 bytes slow mem segment

bhamon, 2021-05-25:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-PICO-V3
- IDF version: v4.3-beta3-195-g6be10fab0
- Build System:         CMake
- Compiler version: (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         external 3.3V

## Problem Description

On an empty project (basic app_main with a simple ESP_LOG) an unknown 16 bytes slow mem segment has appeared on the 4.3 branch (segment 5 on the following output):

```
I (42) boot: Partition Table:
I (46) boot: ## Label            Usage          Type ST Offset   Length
I (53) boot:  0 otadata          OTA data         01 00 0000e000 00002000
I (61) boot:  1 ota_0            OTA app          00 10 00010000 001f0000
I (68) boot:  2 ota_1            OTA app          00 11 00200000 001f0000
I (76) boot:  3 nvs_keys         NVS keys         01 04 003f0000 00001000
I (83) boot:  4 nvs              WiFi data        01 02 003f1000 0000f000
I (91) boot: End of partition table
I (95) boot: No factory image, trying OTA 0
I (100) boot_comm: chip revision: 3, min. application chip revision: 0
I (107) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=070e4h ( 28900) map
I (126) esp_image: segment 1: paddr=0001710c vaddr=3ffb0000 size=0291ch ( 10524) load
I (131) esp_image: segment 2: paddr=00019a30 vaddr=40080000 size=065e8h ( 26088) load
I (143) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=13d94h ( 81300) map
I (174) esp_image: segment 4: paddr=00033dbc vaddr=400865e8 size=056cch ( 22220) load
I (184) esp_image: segment 5: paddr=00039490 vaddr=50000200 size=00010h (    16) load
```

It is then impossible to set the CONFIG_ULP_COPROC_RESERVE_MEM to 8192. The generated link error:

```
/opt/esp/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: erl.elf section `.rtc.data' will not fit in region `rtc_slow_seg'
/opt/esp/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: RTC_SLOW segment data does not fit.
/opt/esp/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: region `rtc_slow_seg' overflowed by 16 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

Also, a CONFIG_ULP_COPROC_RESERVE_MEM value between 8169 an 8176 doesn't trigger a link error, but the bootloader fails to load the segment:

```
I (35) boot: Partition Table:
I (39) boot: ## Label            Usage          Type ST Offset   Length
I (46) boot:  0 otadata          OTA data         01 00 0000e000 00002000
I (53) boot:  1 ota_0            OTA app          00 10 00010000 001f0000
I (61) boot:  2 ota_1            OTA app          00 11 00200000 001f0000
I (68) boot:  3 nvs_keys         NVS keys         01 04 003f0000 00001000
I (76) boot:  4 nvs              WiFi data        01 02 003f1000 0000f000
I (83) boot: End of partition table
I (87) boot: No factory image, trying OTA 0
I (92) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=29814h (170004) map
I (165) esp_image: segment 1: paddr=0003983c vaddr=3ffbdb60 size=048e8h ( 18664) load
I (173) esp_image: segment 2: paddr=0003e12c vaddr=40080000 size=01eech (  7916) load
I (176) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=cf434h (848948) map
I (499) esp_image: segment 4: paddr=0010f45c vaddr=40081eec size=1cf58h (118616) load
I (550) esp_image: segment 5: paddr=0012c3bc vaddr=50001ff0 size=00010h (    16) load
E (550) esp_image: Segment 5 0x50001ff0-0x50002000 invalid: bad load address range
E (556) boot: OTA app partition slot 0 is not bootable
E (562) esp_image: image at 0x200000 has invalid magic byte
W (568) esp_image: image at 0x200000 has invalid SPI mode 255
W (574) esp_image: image at 0x200000 has invalid SPI size 15
E (581) boot: OTA app partition slot 1 is not bootable
E (586) boot: No bootable app partitions in the partition table
```

The bootloader error may be due to a 8-bytes alignment problem. The 16 bytes could fit in the remaining slow mem space (subtracted from ULP reserved memory), but are out of bounds once aligned.

I searched the whole repository for a SLOW_MEM_ATTR match without any luck.

negativekelvin, 2021-05-25:
https://github.com/espressif/esp-idf/blob/1f54d17503ff17d830de75eed4eb47bc6a3aafc3/components/esp_hw_support/esp_clk.c#L51

bhamon, 2021-05-26:
@negativekelvin  Nice catch.

Then the max value for CONFIG_ULP_COPROC_RESERVE_MEM should be changed to `8168` accordingly (unless you find a workaround for the alignment problem).

closed 2021-06-28 by espressif-bot

negativekelvin, 2021-06-28:
@ESP-Marius what about 

> a CONFIG_ULP_COPROC_RESERVE_MEM value between 8169 an 8176 doesn't trigger a link error, but the bootloader fails to load the segment

ESP-Marius, 2021-06-28:
> @ESP-Marius what about
> 
> > a CONFIG_ULP_COPROC_RESERVE_MEM value between 8169 an 8176 doesn't trigger a link error, but the bootloader fails to load the segment

This should also be OK now I think (https://github.com/espressif/esp-idf/commit/2f705136e9ba0e7406a7c99015c314c0f67f8e36)