espressif/esp-idf #7075: Add hooks to the default UART irq handler

smurfix, 2021-05-26:
**Is your feature request related to a problem? Please describe.**

My code needs to process incoming serial data, most notably an interrupt character like ^C, immediately, i.e. from interrupt.

The problem is that the default UART interrupt service routine is a large blob which cannot be partially overridden or hooked into, and I'd rather not re-write the whole UART handler.

**Describe the solution you'd like**

At minimum, the serial handler should have a couple of hooks (TX fifo low water mark reached, RX data available) which optionally call my code.

**Describe alternatives you've considered**

Copying the code of `uart_rx_intr_handler_default` to my own code and modifying it: is not possible because it refers to other static data items, most notably `uart_context`.

Using sync code and `select`: doesn't work for me because my code needs to process an interrupt character.

negativekelvin, 2021-05-26:
What is the required latency and what latency are you seeing with pattern character event and high priority task?

Alvin1Zhang, 2021-05-27:
Thanks for raising this feature request.

txinto, 2024-01-24:
I have the same issue too.  It seems the ability to define your own ISR has been dropped from v4 to v5?