espressif/esp-idf #7077: SNI

Anouer-Kebir, 2021-05-26:
**Is your feature request related to a problem? Please describe.**

A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**

A clear and concise description of what you want to happen.

**Describe alternatives you've considered**

A clear and concise description of any alternative solutions or features you've considered.

Please give as many details as you can. Include suggestions for useful APIs or interfaces if relevant.

**Additional context**

Add any other context or screenshots about the feature request here.

Alvin1Zhang, 2021-05-27:
@Anouer-Kebir Thanks for raising this feature request, would you please help share more details as described in the [template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=Type%3A+Feature+Request&template=feature_request.md&title=)? Thanks.

Alvin1Zhang, 2021-06-01:
Thanks for reporting, will close due to short of details, feel free to reopen with more details, thanks.

closed 2021-06-01 by Alvin1Zhang