espressif/esp-idf #7079: IDF4.3-beta3 random crash when called stat

Zmmfly, 2021-05-27:
## Environment

- Development Kit:  Custom
- Kit version: v4.3-beta3
- Module or chip used:  ESP32-WROVER-E (16Mbyte & 4Mbytes SPIRAM)
- IDF version (run ``git describe --tags`` to find it):
    // v4.3-beta3
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Windows
- (Windows only) environment type: Plain Command Prompt
- Using an IDE?: No
- Power Supply:  external 3.3V

## Problem Description
When used "esp_littlefs", and the "stat" called in http_server uri handler, will cause a random crash.
Sometime, it's will crash; sometime it's fine.

Maybe the stack overflow?

### Expected Behavior
Get stat result

### Actual Behavior
Random crash

### Steps to reproduce

1. Use "esp_littlefs" and update littlefs to v2.4.0
2. WiFi and ble used in same time.
3. SoftAP open
4. https_server started
5. add file downloader uri handler
6. Call stat in file download handler to get file size
7. Random crash

### Code to reproduce this issue
```cpp
    if (stat(file.c_str(), &file_stat) != 0) {
        ESP_LOGE(TAG, "Failed to stat file : %s", file.c_str());
        path += svr->compress_suffix;
        ESP_LOGI(TAG, "Try find gz file: %s", path.c_str());
    } else {
        goto skip_gz; //if uncompressed file exist, skip find gz file
    }
```

## Debug Logs

RAM print code:
```cpp
    ESP_LOGI(TAG, "SPIRAM size: %u", heap_caps_get_total_size(MALLOC_CAP_SPIRAM));
    ESP_LOGI(TAG, "SPIRAM free: %u", esp_get_free_heap_size() - esp_get_free_internal_heap_size());
    ESP_LOGI(TAG, "IntRAM size: %u", heap_caps_get_total_size(MALLOC_CAP_INTERNAL));
    ESP_LOGI(TAG, "IntRAM free: %u", esp_get_free_internal_heap_size());
```

RAM usage:
```
I (136025) esp_web_cxx: SPIRAM size: 4175799
I (136025) esp_web_cxx: SPIRAM free: 3988835
I (136029) esp_web_cxx: IntRAM size: 319284
I (136034) esp_web_cxx: IntRAM free: 87468
```

register dump:
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4009b51b  PS      : 0x00060c33  A0      : 0x8009c1e7  A1      : 0x3ffe7350  
A2      : 0x80eb63fc  A3      : 0x4009c228  A4      : 0x3ffe7390  A5      : 0x3ffc88b0  
A6      : 0xfffffffc  A7      : 0x00000001  A8      : 0x8009b535  A9      : 0x3ffe7330  
A10     : 0x3ffc88b0  A11     : 0x40eedb50  A12     : 0x00000000  A13     : 0x3ffe7390  
A14     : 0xb33fffff  A15     : 0x000000fe  SAR     : 0x00000015  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x80eb6400  LBEG    : 0x40091f76  LEND    : 0x40091f81  LCOUNT  : 0x00000000  

Backtrace:0x4009b518:0x3ffe7350 0x4009c1e4:0x3ffe7370 0x400d6422:0x3ffe7390 0x400d6498:0x3ffe73d0 0x4008cdf2:0x3ffe7410 0x4008cb9f:0x3ffe7430 0x4011b213:0x3ffe7470 0x40115d13:0x3ffe74a0 0x40116301:0x3ffe74d0 0x40116924:0x3ffe7500 0x40117098:0x3ffe75a0 0x40119fce:0x3ffe7600 0x40115120:0x3ffe7650 0x400d469d:0x3ffe7780 0x401a6d4d:0x3ffe77a0 0x401138df:0x3ffe77c0 0x40152c17:0x3ffe7890 0x40151af4:0x3ffe78d0 0x4015215a:0x3ffe7980 0x40150428:0x3ffe79a0
```
And the add2line trace:
```
0x4009b518: block_size at S:/esp-idf-4.3-beta3_path/components/heap/heap_tlsf_block_functions.h:75 (discriminator 4)
 (inlined by) block_is_last at S:/esp-idf-4.3-beta3_path/components/heap/heap_tlsf_block_functions.h:86 (discriminator 4)
 (inlined by) tlsf_walk_pool at S:/esp-idf-4.3-beta3_path/components/heap/heap_tlsf.c:581 (discriminator 4)
0x4009c1e4: multi_heap_get_info_impl at S:/esp-idf-4.3-beta3_path/components/heap/multi_heap.c:370
 (inlined by) multi_heap_get_info_impl at S:/esp-idf-4.3-beta3_path/components/heap/multi_heap.c:361
0x400d6422: heap_caps_get_info at S:/esp-idf-4.3-beta3_path/components/heap/heap_caps.c:453
0x400d6498: heap_caps_get_largest_free_block at S:/esp-idf-4.3-beta3_path/components/heap/heap_caps.c:441
0x4008cdf2: get_buffer_malloc at S:/esp-idf-4.3-beta3_path/components/spi_flash/spi_flash_os_func_app.c:170
0x4008cb9f: esp_flash_read at S:/esp-idf-4.3-beta3_path/components/spi_flash/esp_flash_api.c:642
0x4011b213: esp_partition_read at S:/esp-idf-4.3-beta3_path/components/spi_flash/partition.c:347
0x40115d13: littlefs_api_read at s:\project_path\build/../components/esp_littlefs/src/littlefs_api.c:22
0x40116301: lfs_bd_read at s:\project_path\build/../components/esp_littlefs/src/littlefs/lfs.c:99
 (inlined by) lfs_bd_read at s:\project_path\build/../components/esp_littlefs/src/littlefs/lfs.c:27
0x40116924: lfs_dir_fetchmatch at s:\project_path\build/../components/esp_littlefs/src/littlefs/lfs.c:856
0x40117098: lfs_dir_find at s:\project_path\build/../components/esp_littlefs/src/littlefs/lfs.c:1240
0x40119fce: lfs_rawstat at s:\project_path\build/../components/esp_littlefs/src/littlefs/lfs.c:3184
 (inlined by) lfs_stat at s:\project_path\build/../components/esp_littlefs/src/littlefs/lfs.c:5028
0x40115120: vfs_littlefs_stat at s:\project_path\build/../components/esp_littlefs/src/esp_littlefs.c:1258
0x400d469d: esp_vfs_stat at S:/esp-idf-4.3-beta3_path/components/vfs/vfs.c:568 (discriminator 3)
0x401a6d4d: stat at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/syscalls/sysstat.c:11
0x401138df: esp_web_cxx::file_downloader(httpd_req*) at s:\project_path\build/../components/esp_cxx_lib/esp_web_cxx.cpp:94
0x40152c17: httpd_uri at s:\project_path\build/../components/esp_http_server/src/httpd_uri.c:339
0x40151af4: httpd_parse_req at s:\project_path\build/../components/esp_http_server/src/httpd_parse.c:667
 (inlined by) httpd_req_new at s:\project_path\build/../components/esp_http_server/src/httpd_parse.c:789
0x4015215a: httpd_sess_process at s:\project_path\build/../components/esp_http_server/src/httpd_sess.c:304 (discriminator 15)
0x40150428: httpd_server at s:\project_path\build/../components/esp_http_server/src/httpd_main.c:217
 (inlined by) httpd_thread at s:\project_path\build/../components/esp_http_server/src/httpd_main.c:247
```
## Other items if possible

- [x ] sdkconfig file (attach the sdkconfig file from your project folder)
- [x ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [x ] coredump (This provides stacks of tasks.)

igrr, 2021-05-27:
Hi @Zmmfly, this looks like heap corruption — the crash happens when the memory allocation functions tries to get information from the heap structure, and runs into an invalid pointer. Please try the instructions given in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#finding-heap-corruption to locate the heap corruption.

Regarding the stack overflow possibility, from the panic handler output you can see how much stack was actually used at the point of the crash: 0x3ffe79a0-0x3ffe7350=0x650. It is possible though that the stack got overflowed at some point before the crash happened. To detect stack overflow when it happens, you can enable CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK option: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-freertos-watchpoint-end-of-stack.

tom-borcin, 2021-07-12:
Hi @Zmmfly, did you manage to resolve the issue following help from @igrr ?

Zmmfly, 2021-07-13:
@igrr Thanks for you help, i tried to resolve this problem at idf4.3-beta, but appear again, most at access to web page. When i upgrade to idf4.3 from idf4.3-beta, it's disappear. @tom-borcin I don't known it's as resolved or not.

tom-borcin, 2021-07-13:
Thank you for the update @Zmmfly. If the problem disappeared by upgrading to v4.3 we consider it resolved. Version 4.3 is minor release and is considered complete release as opposed to beta releases.

Thank you for using ESP-IDF.

closed 2021-07-13 by tom-borcin