espressif/esp-idf #7080: Key Trust Chain validation for Signed App Verification Without Hardware Secure Boot

Darsh-Dev, 2021-05-27:
Hello,

I want to implement the key trust chain features for [Signed App Verification Without Hardware Secure Boot](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/security/secure-boot-v2.html#signed-app-verification-without-hardware-secure-boot).
`"In this mode, any public key which is present in the signature block of the currently running app will be used to verify the signature of a newly updated app. (The signature on the running app isn’t verified during the update process, it’s assumed to be valid.) In this way, the system creates a chain of trust from the running app to the newly updated app."
`

Can you please provide the steps to validate multiple public keys to verify the signed image? I would like to change the pub key during the new OTA updates.

Thanks

Alvin1Zhang, 2021-05-28:
Thanks for raising the feature request.

projectgus, 2021-05-28:
Hi @Darsh-Dev,

I'm afraid this is a mistake in the documentation. Only one key can be used to verify a new signature (the first slot), as described in the next paragraph:

> The system only verifies a signature block in the first position and ignores the other (2) appended signatures.

This is necessary because otherwise an attacker could append a new public key in an update and gain control of the updated firmware.

We'll fix the other paragraph to match.

> Can you please provide the steps to validate multiple public keys to verify the signed image? I would like to change the pub key during the new OTA updates.

We can consider this as a feature request, however it's a high cost and risk to implement due to the extra complexity, risk of a critical security loophole as described above, and difficulty of managing OTA update and rollback behaviour in a power-off-safe way. Using the hardware as a root of trust (i.e. Secure Boot) on ESP32-S2 & ESP32-C3 is the accepted and supported way to support multiple signing keys with revocation.

Which chip are you using? May I ask why you don't want to use hardware Secure Boot, and what the reason for the "new trusted public key" requirement is?

closed 2021-06-28 by espressif-bot