espressif/esp-idf #7082: ESP32-C3 ECO3   Connect WiFi and restart

zenghi, 2021-05-28:
Problem debug
```cpp,monokai
I (46771) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 14, mt_pti: 25000, mt_time: 10000
I (46791) wifi:BcnInt:102400, DTIM:1
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x13 (GLITCH_RTC_RST),boot:0xe (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x1658
load:0x403ce000,len:0x8c0
load:0x403d0000,len:0x2928
entry 0x403ce000
I (31) boot: ESP-IDF v4.4-dev-1404-gc13afea63-dirty 2nd stage bootloader
I (31) boot: compile time 09:37:26
```
If the TX power is limited to 14, it will operate normally
or git checkout 7bb5c8cc82c647316802f051cff30a61ddcc2b5e the connection is normal

Use examples\wifi\getting_started\station to check

Our problem prototype has been provided to your business Xie Jin

zenghi, 2021-06-01:
bootloader_glitch_reset_disable
add REG_SET_FIELD(RTC_CNTL_FIB_SEL_REG, RTC_CNTL_FIB_SEL, RTC_CNTL_FIB_SUPER_WDT_RST | RTC_CNTL_FIB_BOR_RST);
to fix gitch reset。

yshestakov, 2021-08-18:
Verified the patch on ESP32-C3 revision 3 based "LilyGo-T-OI-PLUS" board and latest esp-idf code:
```
diff --git a/components/bootloader_support/src/esp32c3/bootloader_esp32c3.c b/components/bootloader_support/src/esp32c3/bootloader_esp32c3.c
index 7f033c0560..684983000b 100644
--- a/components/bootloader_support/src/esp32c3/bootloader_esp32c3.c
+++ b/components/bootloader_support/src/esp32c3/bootloader_esp32c3.c
@@ -273,7 +273,7 @@ static inline void bootloader_glitch_reset_disable(void)
     uint8_t chip_version = bootloader_common_get_chip_revision();
     if (chip_version < 2) {
         REG_SET_FIELD(RTC_CNTL_FIB_SEL_REG, RTC_CNTL_FIB_SEL, RTC_CNTL_FIB_SUPER_WDT_RST);
-    } else if (chip_version == 2) {
+    } else if (chip_version >= 2) {
         REG_SET_FIELD(RTC_CNTL_FIB_SEL_REG, RTC_CNTL_FIB_SEL, RTC_CNTL_FIB_SUPER_WDT_RST | RTC_CNTL_FIB_BOR_RST);
     }
 }
```

closed 2021-09-07 by espressif-bot