espressif/esp-idf #7083: Outdated CMake version in tools.json

b1ackviking, 2021-05-28:
https://github.com/espressif/esp-idf/blob/c13afea635adec735435961270d0894ff46eef85/tools/tools.json#L338-L403

CMake 3.16.4 was released almost 1.5 years ago, and it lacks a lot of new functionality we want to use in our projects.

I'm going to submit a PR for bumping the CMake version to 3.20.3 in the `tools.json` file in the next few days.

Are there any pitfalls I don't know about that require an older version of CMake?

dobairoland, 2021-05-31:
Thank you @b1ackviking for pointing this out. We will update it soon. 

Thank you for offering a PR. I think it is not necessary because we still have to run our internal tests to see if all of our cmake files are compatible. We might settle for an older version in case any serious issue emerges. 

In the meantime, if you already run experiments and experienced any warnings or issue then your comments are welcomed.

closed 2021-07-29 by espressif-bot

MartinPedersenpp, 2021-08-31:
Will this also be changed in 4.3 or other releases? or only on the master branch until a new stable release?

Best regards

igrr, 2021-08-31:
That's correct, we didn't plan on backporting this to existing releases since CMake upgrade might affect projects based on these releases. Is there any specific functionality you are missing in 3.16?

MartinPedersenpp, 2021-08-31:
Not anything super important, but I am trying to run some Azure IoT SDK examples for provisioning and they use the umock library wich requires cmake 3.18 or above.
Example link: [ESP-AZURE Examples](https://github.com/espressif/esp-azure/tree/master/examples/prov_dev_client_ll_sample)