espressif/esp-idf #7084: esp_mesh_send() is blocked forever in router-less environment

dshil, 2021-05-28:
## Environment

- Module or chip used:  ESP32-WROVER-B
- IDF version: master
- Build System: idf.py (cmake - the same)
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Linux
- Using an IDE: No
- Power Supply: USB

## Problem Description

I am experimenting with router-less setup. I use examples/mesh/internal_communication almost unmodified. 
The only difference is that in main function during the initialization phase I added the following lines:

```cpp
    bool root = false;
    if (root) {
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_set_type(MESH_ROOT));
    } else {
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_fix_root(true));
    }
```

and I commented the following lines:

```cpp
    /* router */
    cfg.channel = CONFIG_MESH_CHANNEL;
    cfg.router.ssid_len = strlen(CONFIG_MESH_ROUTER_SSID);
    memcpy((uint8_t *) &cfg.router.ssid, CONFIG_MESH_ROUTER_SSID, cfg.router.ssid_len);
    memcpy((uint8_t *) &cfg.router.password, CONFIG_MESH_ROUTER_PASSWD,
           strlen(CONFIG_MESH_ROUTER_PASSWD));

```

and put `esp_mesh_comm_p2p_start()` into the following block:

```cpp
    switch (event_id) {
    case MESH_EVENT_STARTED: {
        esp_mesh_get_id(&id);
        ESP_LOGI(MESH_TAG, "<MESH_EVENT_MESH_STARTED>ID:"MACSTR"", MAC2STR(id.addr));
        is_mesh_connected = false;
        mesh_layer = esp_mesh_get_layer();

        esp_mesh_comm_p2p_start();
    }
```

For one device I set `root` variable to true and for another one is to false. Thus I have 1 root and 1 node element in the mesh network. All other parts of the code are untouched. When I start the mesh network devices are recognized by each other:

**Root**

```cpp
ESPPORT=/dev/ttyUSB1  pmake monitor
Note: idf_monitor.py will attempt to set baud rate automatically. To specify a baud rate, set the ESPBAUD environment variable.
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7084
load:0x40078000,len:13704
load:0x40080400,len:4000
entry 0x40080688
I (29) boot: ESP-IDF v4.2-beta1-dirty 2nd stage bootloader
I (29) boot: compile time 14:53:38
[... 271 lines cut ...]
I (41633) mesh: 701[monitor]no change, parent:24:6f:28:f0:10:81, rssi:-36
I (41643) mesh: 1990<arm>parent monitor, my layer:2(cap:6)(node), interval:380088ms, retries:3<>
I (41673) mesh_main: layer:2, rtableSize:1, NODE
I (51673) mesh_main: layer:2, rtableSize:1, NODE
I (61673) mesh_main: layer:2, rtableSize:1, NODE
I (71673) mesh_main: layer:2, rtableSize:1, NODE
```

Then I added a few logs to clarify the issue:

```cpp
            ESP_LOGI(MESH_TAG, "before send to= " MACSTR "", MAC2STR(route_table[i].addr));
            err = esp_mesh_send(&route_table[i], &data, MESH_DATA_P2P, NULL, 0);
            ESP_LOGI(MESH_TAG, "after send to= " MACSTR "", MAC2STR(route_table[i].addr));
```

I expect that root element will send data to each element in the network but instead it was blocked forever.

```cpp
            ESP_LOGI(MESH_TAG, "before send to= " MACSTR "", MAC2STR(route_table[i].addr));
            err = esp_mesh_send(&route_table[i], &data, MESH_DATA_P2P, NULL, 0);
            ESP_LOGI(MESH_TAG, "after send to= " MACSTR "", MAC2STR(route_table[i].addr));
```

From the other side when I provide correct router settings all work as expected. Could somebody clarify is it the expected or not? Probably @dongdong004, @ESP-iPENCIL. At the end I want be able to send data from node to root and from root to node without the router.

shenjun7, 2021-05-31:
Please update IDF version to v4.2(d9ec7df3) or v4.3(639e7ad4). All devices in the network shall use esp_mesh_fix_root(true) to keep the same fixed root setting.

dshil, 2021-05-31:
Hi @shenjun7 and thanks for your reply. I checkout to [v4.3](https://github.com/espressif/esp-idf/commit/639e7ad494d93fc82159f9fe854041bc43a96d5f) and apply the same patch mentioned above:

```
diff --git a/examples/mesh/internal_communication/main/mesh_main.c b/examples/mesh/internal_communication/main/mesh_main.c
index 6275acde3..bf96897de 100644
--- a/examples/mesh/internal_communication/main/mesh_main.c
+++ b/examples/mesh/internal_communication/main/mesh_main.c
@@ -101,7 +101,9 @@ void esp_mesh_p2p_tx_main(void *arg)
         }
 
         for (i = 0; i < route_table_size; i++) {
-            err = esp_mesh_send(&route_table[i], &data, MESH_DATA_P2P, NULL, 0);
+            ESP_LOGI(MESH_TAG, "before send to= " MACSTR "", MAC2STR(route_table[i].addr));
+            err = esp_mesh_send(&route_table[i], &data, MESH_DATA_NONBLOCK | MESH_DATA_P2P, NULL, 0);
+            ESP_LOGI(MESH_TAG, "after send to= " MACSTR "", MAC2STR(route_table[i].addr));
             if (err) {
                 ESP_LOGE(MESH_TAG,
                          "[ROOT-2-UNICAST:%d][L:%d]parent:"MACSTR" to "MACSTR", heap:%d[err:0x%x, proto:%d, tos:%d]",
@@ -188,6 +190,8 @@ void mesh_event_handler(void *arg, esp_event_base_t event_base,
         ESP_LOGI(MESH_TAG, "<MESH_EVENT_MESH_STARTED>ID:"MACSTR"", MAC2STR(id.addr));
         is_mesh_connected = false;
         mesh_layer = esp_mesh_get_layer();
+
+        esp_mesh_comm_p2p_start();
     }
     break;
     case MESH_EVENT_STOPPED: {
@@ -247,7 +251,6 @@ void mesh_event_handler(void *arg, esp_event_base_t event_base,
         if (esp_mesh_is_root()) {
             esp_netif_dhcpc_start(netif_sta);
         }
-        esp_mesh_comm_p2p_start();
     }
     break;
     case MESH_EVENT_PARENT_DISCONNECTED: {
@@ -399,37 +402,24 @@ void app_main(void)
     ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &ip_event_handler, NULL));
     ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_FLASH));
     ESP_ERROR_CHECK(esp_wifi_start());
+
     /*  mesh initialization */
     ESP_ERROR_CHECK(esp_mesh_init());
     ESP_ERROR_CHECK(esp_event_handler_register(MESH_EVENT, ESP_EVENT_ANY_ID, &mesh_event_handler, NULL));
-    /*  set mesh topology */
-    ESP_ERROR_CHECK(esp_mesh_set_topology(CONFIG_MESH_TOPOLOGY));
-    /*  set mesh max layer according to the topology */
-    ESP_ERROR_CHECK(esp_mesh_set_max_layer(CONFIG_MESH_MAX_LAYER));
-    ESP_ERROR_CHECK(esp_mesh_set_vote_percentage(1));
-    ESP_ERROR_CHECK(esp_mesh_set_xon_qsize(128));
-#ifdef CONFIG_MESH_ENABLE_PS
-    /* Enable mesh PS function */
-    ESP_ERROR_CHECK(esp_mesh_enable_ps());
-    /* better to increase the associate expired time, if a small duty cycle is set. */
-    ESP_ERROR_CHECK(esp_mesh_set_ap_assoc_expire(60));
-    /* better to increase the announce interval to avoid too much management traffic, if a small duty cycle is set. */
-    ESP_ERROR_CHECK(esp_mesh_set_announce_interval(600, 3300));
-#else
-    /* Disable mesh PS function */
-    ESP_ERROR_CHECK(esp_mesh_disable_ps());
-    ESP_ERROR_CHECK(esp_mesh_set_ap_assoc_expire(10));
-#endif
+
+    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_fix_root(true));
+
+    bool root = false;
+    if (root) {
+      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_set_type(MESH_ROOT));
+    } else {
+      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_set_type(MESH_NODE));
+    }
+
     mesh_cfg_t cfg = MESH_INIT_CONFIG_DEFAULT();
     /* mesh ID */
     memcpy((uint8_t *) &cfg.mesh_id, MESH_ID, 6);
-    /* router */
-    cfg.channel = CONFIG_MESH_CHANNEL;
-    cfg.router.ssid_len = strlen(CONFIG_MESH_ROUTER_SSID);
-    memcpy((uint8_t *) &cfg.router.ssid, CONFIG_MESH_ROUTER_SSID, cfg.router.ssid_len);
-    memcpy((uint8_t *) &cfg.router.password, CONFIG_MESH_ROUTER_PASSWD,
-           strlen(CONFIG_MESH_ROUTER_PASSWD));
-    /* mesh softAP */
     ESP_ERROR_CHECK(esp_mesh_set_ap_authmode(CONFIG_MESH_AP_AUTHMODE));
     cfg.mesh_ap.max_connection = CONFIG_MESH_AP_CONNECTIONS;
     memcpy((uint8_t *) &cfg.mesh_ap.password, CONFIG_MESH_AP_PASSWD,
@@ -437,13 +427,4 @@ void app_main(void)
     ESP_ERROR_CHECK(esp_mesh_set_config(&cfg));
     /* mesh start */
     ESP_ERROR_CHECK(esp_mesh_start());
-#ifdef CONFIG_MESH_ENABLE_PS
-    /* set the device active duty cycle. (default:12, MESH_PS_DEVICE_DUTY_REQUEST) */
-    ESP_ERROR_CHECK(esp_mesh_set_active_duty_cycle(CONFIG_MESH_PS_DEV_DUTY, CONFIG_MESH_PS_DEV_DUTY_TYPE));
-    /* set the network active duty cycle. (default:12, -1, MESH_PS_NETWORK_DUTY_APPLIED_ENTIRE) */
-    ESP_ERROR_CHECK(esp_mesh_set_network_duty_cycle(CONFIG_MESH_PS_NWK_DUTY, CONFIG_MESH_PS_NWK_DUTY_DURATION, CONFIG_MESH_PS_NWK_DUTY_RULE));
-#endif
-    ESP_LOGI(MESH_TAG, "mesh starts successfully, heap:%d, %s<%d>%s, ps:%d\n",  esp_get_minimum_free_heap_size(),
-             esp_mesh_is_root_fixed() ? "root fixed" : "root not fixed",
-             esp_mesh_get_topology(), esp_mesh_get_topology() ? "(chain)":"(tree)", esp_mesh_is_ps_enabled());
 }
```

Nothing changed, I have the same result, esp_mesh_send is blocked forever even if I setup esp_mesh_fix_root(true) for all devices (root and non-roots) as you suggested previously:

```
I (3842) wifi:station: 24:6f:28:f0:0d:5c join, AID=1, bgn, 40U
I (3842) mesh_main: <MESH_EVENT_PS_CHILD_DUTY>cidx:0, 24:6f:28:f0:0d:5c, duty:12
W (3862) mesh_main: <MESH_EVENT_ROUTING_TABLE_ADD>add 1, new:2, layer:1
I (3862) mesh_main: <MESH_EVENT_CHILD_CONNECTED>aid:1, 24:6f:28:f0:0d:5c
I (4762) mesh_main: before send to= 24:6f:28:f0:10:80
W (4762) mesh_main: [#RX:4/4][L:1] parent:00:00:00:00:00:00, receive from 24:6f:28:f0:10:80, size:1460, heap:171432, flag:0[err:0x0, proto:0, tos:0]
I (4762) mesh_main: after send to= 24:6f:28:f0:10:80
I (4772) mesh_main: before send to= 24:6f:28:f0:0d:5c
```

shenjun7, 2021-05-31:
Please tell me the specific commit number.

dshil, 2021-05-31:
@shenjun7 I use the same as you suggested, 639e7ad494d93fc82159f9fe854041bc43a96d5f. v4.3(639e7ad)

dshil, 2021-06-05:
Hi @shenjun7, any updates?

shenjun7, 2021-06-17:
The root cause has been found. We will merge it to master first and then backport to the release/v4.x.

dshil, 2021-07-05:
Hi @shenjun7, any news?

shenjun7, 2021-07-06:
The bug has been fixed in gitlab and is waiting to be synchronized to github.

dshil, 2021-07-19:
Hi @shenjun7, how long normally does it take for the bug fix to be synchronized with github repo?

Alvin1Zhang, 2021-07-19:
Thanks for reporting, and sorry for late reply. Fix on master branch is available https://github.com/espressif/esp-idf/commit/d93849baf86b6de5f08a03ae6aa7006dac58a93b. We are back porting the fix to release/4.3 and release/4.2 branches.

dshil, 2021-07-19:
Hi @Alvin1Zhang, thanks for the feedback, very helpful. Is it possible to get an approximate timeline when fix will be available in release branches?

Alvin1Zhang, 2021-07-20:
Thanks for reporting, and sorry for late reply. Fix on master branch is available https://github.com/espressif/esp-idf/commit/de92d7e15f1033e7c21050a547820db005a733a9, feel free to reopen if the issue still happens.

closed 2021-07-20 by Alvin1Zhang

dshil, 2021-07-21:
Hi @Alvin1Zhang, thanks everybody involved into fixing this issue. I am very appreciate it. Waiting for the back-port to release/v4.2.

Alvin1Zhang, 2021-11-24:
@dshil Thanks for reporting and sorry for the very slow turnaround, fix on release/4.2 is available at https://github.com/espressif/esp-idf/commit/f4a2c5b520b621ed2d9e05f0d75112f248c02732 and fix on release/4.3 is https://github.com/espressif/esp-idf/commit/de92d7e15f1033e7c21050a547820db005a733a9.