espressif/esp-idf #7086: ESP IDF 4.3 fails to install using python3.5

weshouman, 2021-05-30:
## Environment

- IDF version: v4.3-beta3
- Operating System:     Linux

## Problem Description

Installing ESP IDF 4.3 using python3.5 to work, the release docs encourages using python3.6 but not requiring it.  
**Python 3.6 or newer will be required in the future to use ESP-IDF v4.4**
 
### Expected Behavior

- ESP IDF 4.3 gets installed using python3.5
Or
- ESP IDF 4.3 shall require python3.6

### Actual Behavior

ESP IDF 4.3 fails to install using python3.5 resulting in:
```
Installing Python packages from /home/user/esp/esp-idf-v4.3-beta3/requirements.txt
Traceback (most recent call last):
  File "/usr/lib/python3.5/runpy.py", line 184, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/user/.espressif/python_env/idf4.3_py3.5_env/lib/python3.5/site-packages/pip/__main__.py", line 29, in <module>
    from pip._internal.cli.main import main as _main
  File "/home/user/.espressif/python_env/idf4.3_py3.5_env/lib/python3.5/site-packages/pip/_internal/cli/main.py", line 58
    sys.stderr.write(f"ERROR: {exc}")
                                   ^
SyntaxError: invalid syntax
Traceback (most recent call last):
  File "/home/user/esp/esp-idf-v4.3-beta3/tools/idf_tools.py", line 1672, in <module>
    main(sys.argv[1:])
  File "/home/user/esp/esp-idf-v4.3-beta3/tools/idf_tools.py", line 1668, in main
    action_func(args)
  File "/home/user/esp/esp-idf-v4.3-beta3/tools/idf_tools.py", line 1393, in action_install_python_env
    subprocess.check_call(run_args, stdout=sys.stdout, stderr=sys.stderr)
  File "/usr/lib/python3.5/subprocess.py", line 581, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/user/.espressif/python_env/idf4.3_py3.5_env/bin/python', '-m', 'pip', 'install', '--no-warn-script-location', '-r', '/home/user/esp/esp-idf-v4.3-beta3/requirements.txt', '--extra-index-url', 'https://dl.espressif.com/pypi']' returned non-zero exit status 1
```

### Steps to reproduce

1. Install ESP IDF 4.3 beta3 using python3.5

dobairoland, 2021-05-30:
Hi @weshouman. What linux distribution and version are you using?

I believe this might be a very similar issue outlined [here](
https://docs.espressif.com/projects/esp-idf/en/release-v4.3/esp32/get-started/linux-setup.html#fixing-broken-pip-on-ubuntu-16-04) if your Linux distribution (& version) is based on the same Debian/Ubuntu version.

Could you please try the steps described there?

dobairoland, 2021-06-11:
I'm closing this. Please inform me if this is still an unsolved issue.

closed 2021-06-11 by dobairoland

jazzoo2, 2021-11-08:
esp-idf $ ./install.sh
Detecting the Python interpreter
Checking "python" ...
Python 3.5.2
"python" has been detected
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2021r1-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2021r1-8.4.0 (already installed)
Skipping xtensa-esp32s3-elf@esp-2021r1-8.4.0 (already installed)
Skipping riscv32-esp-elf@esp-2021r1-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20210401 (already installed)
Installing Python environment and packages
Creating a new Python environment in /home/james/.espressif/python_env/idf4.3_py3.5_env
Traceback (most recent call last):
  File "/public/ESP/esp-idf/tools/idf_tools.py", line 1675, in <module>
    main(sys.argv[1:])
  File "/public/ESP/esp-idf/tools/idf_tools.py", line 1671, in main
    action_func(args)
  File "/public/ESP/esp-idf/tools/idf_tools.py", line 1373, in action_install_python_env
    import virtualenv  # noqa: F401
  File "/home/james/.local/lib/python3.5/site-packages/virtualenv/__init__.py", line 3, in <module>
    from .run import cli_run, session_via_cli
  File "/home/james/.local/lib/python3.5/site-packages/virtualenv/run/__init__.py", line 7, in <module>
    from ..app_data import make_app_data
  File "/home/james/.local/lib/python3.5/site-packages/virtualenv/app_data/__init__.py", line 9, in <module>
    from platformdirs import user_data_dir
  File "/home/james/.local/lib/python3.5/site-packages/platformdirs/__init__.py", line 27
    result: Type[PlatformDirsABC] = getattr(importlib.import_module(module), name)
          ^
SyntaxError: invalid syntax

dobairoland, 2021-11-08:
@jazzoo2 The solution is linked. It is broken on your side. You are using an old Python & pip which are not supported. As a consequence, pip installs an incompatible virtualenv. You can resolve this issue by following the steps outlined in the documentation.