espressif/esp-idf #7087: Removed esp_bt_controller_is_sleeping() and esp_bt_controller_wakeup_request() APIs

geza-pycom, 2021-05-30:
## Environment

- Module or chip used:  [ESP32]
- IDF version (run ``git describe --tags`` to find it): v4.2.1
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Linux]

## Problem Description
Updating from esp-idf v4.1 to v4.2.1 I found out that the folloiwng 2 APIs have been removed:
- esp_bt_controller_wakeup_request()
- esp_bt_controller_is_sleeping()

I did not find this mentioned in the documentation, and the related commit does not give any guidance about:
- Why were these APIs removed ?
- What to use instead of them ?

Related commit: https://github.com/espressif/esp-idf/commit/f64d0ec8730cb1956f17a50da3dd022b393a0d15 

Could you please help me how should I update the application which previously used these APIs ?

blueMoods, 2021-06-22:
Why were these APIs removed ?
> Bluetooth sleep is managed by controller task, only. If other task do this,  may causing some issues.

What to use instead of them ?
> Bluetooth sleep is automatic， customer need not to care about it.  What do you use these APIs for？

Alvin1Zhang, 2021-06-30:
@geza-pycom Thanks for reporting and sorry for late reply. Would you please help share if any updates or details for this issue? Thanks.

geza-pycom, 2021-07-21:
esp_bt_controller_is_sleeping() is used to fetch the status of the controller and esp_bt_controller_wakeup_request() is used to wake up the controller if it is sleeping.

blueMoods, 2022-03-21:
> esp_bt_controller_is_sleeping() is used to fetch the status of the controller and esp_bt_controller_wakeup_request() is used to wake up the controller if it is sleeping.

Hi @geza-pycom , What did you wakeup controller for? just for send HCI command?
`esp_vhci_host_send_packet` is used for send HCI command to controller, and it will wakeup controller if it is sleeping.

closed 2023-11-14 by Alvin1Zhang