espressif/esp-idf #7088: bootloader_sha256_finish crashes in specific case

Piramma, 2021-05-30:
File components/bootloader_support/src/esp_image_format.c 
Line 308
When calling esp_ota_end() during OTA, (attempting to write an unsigned image where signed image is required). 
The return code at line 305 is ESP_ERR_IMAGE_INVALID. At line 308 'CORRUPT HEAP' error is thrown when calling bootloader_sha256_finish. Skipping this avoid the crash, but the memory sha_handle will end up not freed.

projectgus, 2021-05-31:
Hi @Piramma,

What version of ESP-IDF are you using?

Angus

Piramma, 2021-05-31:
I am using idf version 4.2

projectgus, 2021-05-31:
Hi @Piramma,

Thanks @Piramma .I think this issue was fixed in #6210 and included in the [v4.2.1 release](https://github.com/espressif/esp-idf/releases/v4.2.1). Please try updating and see if the problem still exists.

projectgus, 2021-06-28:
Closing as resolved.

closed 2021-06-28 by projectgus