espressif/esp-idf #7090: How to do a mdns_query() without blocking the thread until finished?

0xFEEDC0DE64, 2021-05-31:
I need to do a mdns query in the main task with a few seconds of runtime, but not block the main task.

Almost all network related work is done by polling which saves us lots of synchronisation and memory issues so far, but it seems like mdns cannot be run in that manner?

I'm looking for a mdns_query() call which tells the espressif mdns task to query for 3 seconds and I poll the result every 100ms or so. Is this somehow possible without too many changes to the esp-idf ?

in mdns.c, it seems like the line `xSemaphoreTake(search->done_semaphore, portMAX_DELAY);` is waiting for it to finish, but the `mdns_search_once_t` is never exposed to my main app code.

david-cermak, 2021-06-04:
Hi @0xFEEDC0DE64 

Yes, we would have to split the `mdns_query()` into the init part for creating a new search and the polling part, that would check the semaphore you've mentioned.

Here's a WIP patch, which does so, but still have to review and decide for final IDF support.
[WIP-mdns-Add-asynchronous-query-API.patch.txt](https://github.com/espressif/esp-idf/files/6596698/WIP-mdns-Add-asynchronous-query-API.patch.txt)

0xFEEDC0DE64, 2021-06-13:
thanks, it works like a charm, now it just needs a PR to the esp-idf, are you doing it or should I?

closed 2021-07-16 by espressif-bot