espressif/esp-idf #7091: Require more clarification on the function esp_ble_mesh_server_model_update_state() in ESP BLE MESH

ankit-thealchemist, 2021-05-31:
## Environment
- IDF version (run ``git describe --tags`` to find it):
    // v4.2.1
- Operating System:     [Windows]

- Using an IDE?: Yes (Vs code)
- Power Supply:         [USB]

I am using the esp ble mesh on nimble. The project is simple, I have 3 led, and I am implementing the simple on off server.  Every thing is fine I can change the change the status of all the led from the client. Also the value of struct   esp_ble_mesh_gen_onoff_srv_t onoff_server_1.state.onoff is according to the value received by the ble mesh. 

But when I do the reverse, ie change the status of the led and updates the on off server model using the function  esp_ble_mesh_server_model_update_state(myModel, ESP_BLE_MESH_GENERIC_ONOFF_STATE, &state), every thing is fine in this also, while reading the state of the model from the client side provide the right value of the server, but reading the state of the server model from the server(node) itself just after using the above function, by inquiring the value of onoff_server_1.state.onoff, provides the wrong value. Seems that the above function does not changes the value of the onoff variable.

Does I miss something?

InfiniteYuan, 2021-06-09:
Please check the onoff variable in the ESP_BLE_MESH_SERVER_MODEL_UPDATE_STATE_COMP_EVT case. Because these operations are asynchronous.

```
    case ESP_BLE_MESH_SERVER_MODEL_UPDATE_STATE_COMP_EVT:
        ESP_LOGI(TAG, "onoff (%d)", onoff_server.state.onoff);
        break;
```

ankit-thealchemist, 2021-06-09:
> Please check the onoff variable in the ESP_BLE_MESH_SERVER_MODEL_UPDATE_STATE_COMP_EVT case. Because these operations are asynchronous.
> 
> ```
>     case ESP_BLE_MESH_SERVER_MODEL_UPDATE_STATE_COMP_EVT:
>         ESP_LOGI(TAG, "onoff (%d)", onoff_server.state.onoff);
>         break;
> ```

Thanks for the reply.  Query is solved, thus closing the issue.

closed 2021-06-09 by ankit-thealchemist