espressif/esp-idf #7093: Health Client model in ble mesh client

sumathi98, 2021-06-01:
## Environment

Development Kit : ESP32-DevKitC
Module or chip used: ESP32-WROOM-32U
IDF version : v4.4.0
Build System : idf.py
Compiler version : 8.4.0
Operating System : Linux
Using an IDE? : No
Power Supply : external 5V

## Problem Description
Iam working on ble mesh. I want to get the health status of all the server nodes in the client node, so Iam trying to use health client model, Is there any example or steps available using health client model. 
Can I use health client model in ble mesh client?

InfiniteYuan, 2021-06-10:
Currently, there are no public examples available.

InfiniteYuan, 2021-06-28:
Have you encountered any problems while developing the functions of this model? 

If you have completed the development, please close this issue.

Alvin1Zhang, 2021-07-13:
Thanks for reporting, will close due to short of updates, feel free to reopen.

closed 2021-07-13 by Alvin1Zhang