espressif/esp-idf #7094: ESP32 WROVER GPIO14 pin causing reset on MCU

Tajpeer-Hassansab, 2021-06-01:
We are using ESP32 WROVER with *** module and UART interface to communicate.
we used GPIO 14 as TX PIN and GPIO 34 as RX PIN. We have used ESP-IDF provided UART API to communicate, in that 
when we are calling uart_write_bytes() function then we are observing Reset on ESP32.

Below are the error logs.
0x40081e68: call_start_cpu1 at C:/Users/RanjeetKr/esp/esp_v4.2/esp-idf/components/esp32/cpu_start.c:287
I (0) cpu_start: App cpu up.
E (1417) spiram: SPI SRAM memory test fail. 131072/131072 writes failed, first @ 3F800000
E (1417) cpu_start: External RAM failed memory test!
fff
Backtrace:0x4008a262:0x3ffe3b80 0x4008a949:0x3ffe3ba0 0x4008e751:0x3ffe3bc0 0x400820a7:0x3ffe3c40 0x400790a6:0x3ffe3c80 |<-CORRUPTED
Whenever we are trying to write we get reset and by above error logs we think GPIO14 pin used by another interface.

Can someone suggest on the possible causes for this issue from the above error logs or GPIO14 pin functionality on ESP32 wROOVER?

Thanks in advance.

ryan4volts, 2021-06-15:
Do you have some basic code to replicate this? If its just uart, perhaps just the app_main() {} with a few lines of initialisation and then attempts to send data?

Also which WROVER variant are you using? Note the WROVER only has 1 'o' and the WROOM has the 2 'o's. You're using a WROVER right?

Tajpeer-Hassansab, 2021-06-15:
Hi Ryan,
We have root caused this issue and it found out to be HW issue in our design.

Thanks for the support.

Tajpeer-Hassansab, 2021-06-15:
Identified the root cause. Closing the ticket

closed 2021-06-15 by Tajpeer-Hassansab