espressif/esp-idf #7095: httpd_queue_work not working for me

sabin-john, 2021-06-01:
## Environment
- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version : v4.3-dev-1901-g178b122c1-dirty
- Build System:         [CMake]
- Operating System:     [Windows]
- -IDE: Eclipse IDE Version: 2020-09 (4.17.0) Build id: 20200910-1200
- Power Supply: USB
## Problem Description
I am running [this project]( https://github.com/espressif/esp-idf/tree/release/v4.3/examples/protocols/http_server/advanced_tests) from esp idf.
What I want to do is send async data from the webserver inside the esp to the Client. I have tested this both with the chrome browser and Postman. As per the example the uri /async_data should return two strings "Hello World!" and “Hello Double World!" but I am receiving only “Hello World!” responses in my client and “Hello Double World!" is dropped completely.  I am attaching my debug logs and also wireshark logs of the network traffic.
[adv_test_log_asc.txt](https://github.com/espressif/esp-idf/files/6575464/adv_test_log_asc.txt)
[image]

### Expected Behavior
Client should receive 2 strings "Hello World!" and “Hello Double World!"
### Actual Behavior
Client is only receiving "Hello World!"
### Steps to reproduce
The code was just complied from the example. Only changes made in sdkconfig are the ssid and password.
### Code to reproduce this issue

```cpp
//These are the 2 functions the are in use here
static void generate_async_resp(void *arg)
{
    char buf[250];
    struct async_resp_arg *resp_arg = (struct async_resp_arg *)arg;
    httpd_handle_t hd = resp_arg->hd;
    int fd = resp_arg->fd;
#define HTTPD_HDR_STR      "HTTP/1.1 200 OK\r\n"                   \
                           "Content-Type: text/html\r\n"           \
                           "Content-Length: %d\r\n"
#define STR "Hello Double World!"

    ESP_LOGI(TAG, "Executing queued work fd : %d", fd);

    snprintf(buf, sizeof(buf), HTTPD_HDR_STR,
         strlen(STR));
    httpd_socket_send(hd, fd, buf, strlen(buf), 0);
    /* Space for sending additional headers based on set_header */
    httpd_socket_send(hd, fd, "\r\n", strlen("\r\n"), 0);
    httpd_socket_send(hd, fd, STR, strlen(STR), 0);
#undef STR
    free(arg);
}

static esp_err_t async_get_handler(httpd_req_t *req)
{
#define STR "Hello World!"
    httpd_resp_send(req, STR, HTTPD_RESP_USE_STRLEN);
    /* Also register a HTTPD Work which sends the same data on the same
     * socket again
     */
    struct async_resp_arg *resp_arg = malloc(sizeof(struct async_resp_arg));
    resp_arg->hd = req->handle;
    resp_arg->fd = httpd_req_to_sockfd(req);
    if (resp_arg->fd < 0) {
        return ESP_FAIL;
    }

    ESP_LOGI(TAG, "Queuing work fd : %d", resp_arg->fd);
    httpd_queue_work(req->handle, generate_async_resp, resp_arg);
    return ESP_OK;
#undef STR
}
```
## Debug Logs

```
01-Jun-21 15:04:46 [RX] - D (211401) httpd: httpd_server: processing listen socket 54<ESC> [0m<CR><LF>
D (211411) httpd: httpd_accept_conn: newfd = 57<ESC> [0m<CR><LF>
D (211411) httpd_sess: httpd_sess_new: fd = 57<ESC> [0m<CR><LF>
D (211411) httpd: httpd_accept_conn: complete<ESC> [0m<CR><LF>
D (211411) httpd: httpd_server: doing select maxfd+1 = 58<ESC> [0m<CR><LF>
D (211421) httpd: httpd_server: processing socket 57<ESC> [0m<CR><LF>
D (211421) httpd_sess: httpd_sess_process: httpd_req_new<ESC> [0m<CR><LF>
D (211431) httpd_txrx: httpd_recv_with_opt: requested length = 128<ESC> [0m<CR><LF>
D (211441) httpd_txrx: httpd_recv_with_opt: received length = 128<ESC> [0m<CR><LF>
D (211441) httpd_parse: read_block: received HTTP request block size = 128<ESC> [0m<CR><LF>
D (211451) httpd_parse: cb_url: message begin<ESC> [0m<CR><LF>
D (211451) httpd_parse: cb_url: processing url = /async_data<ESC> [0m<CR><LF>
D (211461) httpd_parse: verify_url: received URI = /async_data<ESC> [0m<CR><LF>
D (211461) httpd_parse: cb_header_field: headers begin<ESC> [0m<CR><LF>

Detailed logs attached
```

shubhamkulkarni97, 2021-06-09:
Hi @sabin-john,

I checked running async response test and it works fine. I think the issue you are observing is specific to HTTP client. You may want to check implementation of client in the [test script](https://github.com/espressif/esp-idf/blob/1d7068e4be430edd92bb63f2d922036dcf5c3cc1/examples/protocols/http_server/advanced_tests/scripts/test.py#L540).

Please also note that `advanced_tests` example is added for testing purpose, so it is guaranteed to work with the test script only.
Application might need some changes based on your requirements.

Let me know if you still face any issues.

shubhamkulkarni97, 2021-06-14:
@sabin-john, Any updates on the issue?

sabin-john, 2021-06-14:
@shubhamkulkarni97  Yes.. I found that the server is sending the the 2 strings in the wireshark log. I will look at how I can handle it on the client side. 
Thank you for your time! Feel free to close the thread.
[image]

Alvin1Zhang, 2021-06-15:
Thanks fore sharing updates, feel free to reopen.

closed 2021-06-15 by Alvin1Zhang