espressif/esp-idf #7096: a2dp_source回调函数bt_app_a2d_data_cb中的len从何而来，是否可以通过哪里配置或修改len的值？

ShaoMingya, 2021-06-02:
https://github.com/espressif/esp-idf/blob/c13afea635adec735435961270d0894ff46eef85/examples/bluetooth/bluedroid/classic_bt/a2dp_source/main/main.c#L381

这只是一个提问，不是指出程序有问题。
我在做一个项目，系统结构为 ①手机（蓝牙音频源）->② ESP32_1（蓝牙 SINK）---spi---> ③ESP32_2（蓝牙 SOURCE）->④蓝牙音箱。流程为：一、①连接②然后播放音乐；二、②通过SPI总线把音频数据流发送给③；三、③通过SPI接收音频数据流，把这个数据流发送给④。目前问题，②能够通过回调函数获取正确的数据流，每次大概获取2048字节。但是③在使用上述例程回调函数时发现，len的值始终为512，不等于2048。我尝试使用ringbuf使得存取分离，但是播放出的声音不正常，是和源音乐节奏接近的类似电流的脉冲声音。所以我想，如果能够在哪里修改此处的len为2048，我的问题应该能够解决，希望得到专业的回答，谢谢！

boblane1, 2021-06-04:
ShaoMingya 你好, 

A2DP source 是通过 SBC 编码将 PCM 音频数据编码成 SBC 帧，然后通过蓝牙发送给对端的 A2DP sink 进行 SBC 解码再次得到 PCM 音频数据。source 中 bt_app_a2d_data_cb 的参数 len 就是编码成一帧所需要的数据。至于你所说的听到的声音“有类似电流的脉冲声音”，我觉得你可以尝试增大 ringbuffer 看看是否能解决问题。

谢谢！

ShaoMingya, 2021-06-11:
> ShaoMingya 你好,
> 
> A2DP source 是通过 SBC 编码将 PCM 音频数据编码成 SBC 帧，然后通过蓝牙发送给对端的 A2DP sink 进行 SBC 解码再次得到 PCM 音频数据。source 中 bt_app_a2d_data_cb 的参数 len 就是编码成一个 SBC 帧所需要的 PCM 数据长度。至于你所说的听到的声音“有类似电流的脉冲声音”，我觉得你可以尝试增大 ringbuffer 看看是否能解决问题。
> 
> 谢谢！

请问这个len是固定的512字节吗？还有进入这个回调函数的间隔时间是多少？

boblane1, 2021-06-11:
>请问这个len是固定的512字节吗？还有进入这个回调函数的间隔时间是多少？

len 取决于 SBC 的 参数,在默认的 SBC 参数下，这个 len 始终是 512 字节。目前，esp32 是每隔 30 ms 触发一次打包 SBC 帧的操作，SBC encoder 会计算出这 30ms 间隔需要发送的帧数，然后循环调用 data_callback 填充 PCM 数据。

Alvin1Zhang, 2021-06-30:
@ShaoMingya 您好， 请问这个问题还有更多更新么？谢谢。

Alvin1Zhang, 2021-08-18:
@ShaoMingya 您好， 这个问题还有更多更新么？谢谢。

Alvin1Zhang, 2021-11-30:
感谢报此问题，因为长时间没有更新，将会暂时关闭此问题；如您有更多更新，请随时重开这个问题，谢谢。

closed 2021-11-30 by Alvin1Zhang