espressif/esp-idf #7098: PPPoS 2G (GSM ) connection not stablished, suddenly Disconnected

JosueSanchezEZM, 2021-06-02:
Hi everybody. I am experiencing an issue during de PPP connection using PPPoE sample from esp-idf 4.2 version.
I have my cellular module configured only to use GSM bands. What I am looking for is to enable PPP connection to communicate to a remote server. 

While the process of connection in PPP occurs, it gets stuck at PPP EVENT 259 (INITIALIZE). Suddenly the event is raised 262 (ESTABLISH) but after a few more seconds, the connection is LOST (Event 268). I was expecting to see after 262 event the PPP_GOT_IP_EVENT and check through the log the information regarding the PPP connection established (IP, DNS Servers, etc). But this does not occur. As you can see below, what happened is that in some way PPP session is trying to be raise but, between 262 and 268 some verification or validation fails so the connection is lost.

Could you please provide me any comments about this issue I am experiencing. I will very glad to get from you any suggestions or additional comments about this process behind PPP connection. Is this problem related to the signal strength? Is there any threshold to manually configure to allow PPP connection establishment?

Thank you so much for any help you can provide to me.

*** WHEN USING 3G AND 4G EVERYTHING WORKS WELL***

LOG:

D (28096) esp_netif_lwip: LwIP stack has been initialized
D (28106) esp_netif_lwip: esp-netif has been successfully initialized
D (28116) event: running task for loop 0x3ffdf42c
D (28116) event: created task for loop 0x3ffdf42c
D (28126) event: created event loop 0x3ffdf42c
I (28126) main: Inside NIC CELLULAR Handler
I (28136) main: Mode PPP START
D (28136) esp_netif_objects: esp_netif_add_to_list 0x3ffdf8a0
D (28146) esp_netif_objects: esp_netif_add_to_list netif added successfully (total netifs: 1)
D (28146) esp-netif_lwip-ppp: esp_netif_new_ppp: PPP connection created: 0x3ffe0f60
D (28156) esp-netif_lwip-ppp: Phase Dead
D (28166) event: created event loop 0x3ffdfb9c
D (28166) event: running post EZM_NIC_CELLULAR:4 with handler 0x400da448 and context 0x3ffdfd84 on loop 0x3ffdfb9c
I (28166) main: Inside NIC CELLULAR Handler
I (28176) ezm_api_nic_cellular.c: Dentro del NIC CELLULAR PPP Handler
I (28186) main: Mode PPP START in main
I (28186) ezm_api_nic_cellular.c: Mode PPP started.
D (28196) event: running post EZM_NIC_CELLULAR:4 with handler 0x40103840 and context 0x3ffdff18 on loop 0x3ffdfb9c
I (28206) main: ----------------------------------------
D (28206) esp_netif_handlers: esp_netif action has started with netif0x3ffdf8a0 from event_id=4
I (28226) main: Inside del Other Handler
D (28226) esp-netif_lwip-ppp: esp_netif_start_ppp: Starting PPP connection: 0x3ffe0f60
D (28246) esp-netif_lwip-ppp: Phase Start
D (28246) event: running post NETIF_PPP_STATUS:259 with handler 0x400da4f4 and context 0x3ffdffbc on loop 0x3ffdf42c
D (28256) esp-netif_lwip-ppp: Phase Establish
I (28256) ezm_api_nic_cellular.c: PPP state changed event 259
D (28266) esp_netif_lwip: check: remote, if=0x3ffdf8a0 fn=0x40104d5c

D (28276) event: running post NETIF_PPP_STATUS:262 with handler 0x400da4f4 and context 0x3ffdffbc on loop 0x3ffdf42c
D (28276) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffdf8a0
I (28286) ezm_api_nic_cellular.c: PPP state changed event 262
D (28296) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (88266) esp-netif_lwip-ppp: Phase Disconnect
D (88266) event: running post NETIF_PPP_STATUS:268 with handler 0x400da4f4 and context 0x3ffdffbc on loop 0x3ffdf42c
I (88266) ezm_api_nic_cellular.c: PPP state changed event 268
D (88276) esp-netif_lwip-ppp: Phase Dead
D (88276) event: running post NETIF_PPP_STATUS:256 with handler 0x400da4f4 and context 0x3ffdffbc on loop 0x3ffdf42c
I (88276) esp-netif_lwip-ppp: Connection lost
I (88286) ezm_api_nic_cellular.c: PPP state changed event 256
D (88296) event: running post IP_EVENT:6 with handler 0x400da6b0 and context 0x3ffdf864 on loop 0x3ffdf42c
I (88306) ezm_api_nic_cellular.c: IP event! 6

I (88316) ezm_api_nic_cellular.c: Modem Disconnect from PPP Server
I (88326) main: Mode PPP NIC LOST IP.

david-cermak, 2021-06-03:
Hi @JosueSanchezEZM 

Does it always fail or are you able to get an IP from time to time? Is the 2G network still supported by your operator?

I'm afraid there's nothing from the PPP client side, that controls the actual network connection. As a quick check, you can try to use `AT` commands to setup a socket connection (using `AT+NETOPEN` or similar). But I'd suggest asking your operator.

closed 2021-06-10 by JosueSanchezEZM

JosueSanchezEZM, 2021-06-16:
Hi everybody. I already found the problem. I realize that one of the configurations AT commands did not allow the module to connect the module to PPP while being exclusively in 2G. Thank you everybody for your suggestions.

CAIJUN111, 2022-03-29:
> Hi everybody. I already found the problem. I realize that one of the configurations AT commands did not allow the module to connect the module to PPP while being exclusively in 2G. Thank you everybody for your suggestions.

Hi!bro,could you tell me  where  the configurations AT commands is.
Thanks in advance.