espressif/esp-idf #7100: Classic BT AVRC passthrough command not working while media is playing

spleen1981, 2021-06-03:
Hi,
sending AVRC passthrough commands (e.g. ESP_AVRC_PT_CMD_FORWARD) to connected peer while media is playing (works ok if media is not playing) from ESP32 A2DP sink seems to be broken with ESP IDF 4.2.1 (same with 4.3beta2), while with 4.2.0 same code works ok.
esp_avrc_ct_send_passthrough_cmd returns ESP_OK, but no event is received in relevant callback function.
Commands seems to be somehow queued, as when media is stopped from peer device, all previous passthrough commands sent are executed in series.

Alvin1Zhang, 2021-06-03:
@spleen1981 Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

spleen1981, 2021-06-04:
## Environment

- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.2.1
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Windows
- (Windows only) environment type: Plain Command Prompt
- Using an IDE?: No
- Power Supply: USB

## Problem Description
Sending AVRC passthrough commands (e.g. ESP_AVRC_PT_CMD_FORWARD) to connected peer while media is playing (works ok if media is not playing) from ESP32 A2DP sink seems to be broken with ESP IDF 4.2.1 (same with 4.3beta2), while with 4.2.0 same code works ok.
esp_avrc_ct_send_passthrough_cmd returns ESP_OK, but no event is received in relevant callback function.
Commands seems to be somehow queued, as when media is stopped from peer device, all previous passthrough commands sent are executed in series.

### Expected Behavior
esp_avrc_ct_send_passthrough_cmd sent to peer and event received in relevant callback function even if media is playing.

### Actual Behavior
esp_avrc_ct_send_passthrough_cmd not sent to peer and event not received in relevant callback function if media is playing.
Commands seems to be somehow queued, as when media is stopped from peer device, all previous passthrough commands sent are executed in series.

### Steps to reproduce

1. compile and upload attached source (AD2P example with additional task sending passthrough commands every 10 seconds starting when AVRC is connected)
2. connect peer 
3. play media from peer

### Code to reproduce this issue

Refer to attached source (AD2P example with additional task sending passthrough commands every 10 seconds starting when AVRC is connected).

## Debug Logs

```
W (26735) BT_APPL: new conn_srvc id:19, app_id:1
I (26735) BT_AV: A2DP audio state: Started
I (26735) RCCT: AVRC event notification: 1
I (26745) BT_AV: Playback status changed: 0x1
I (26745) RCCT: AVRC event notification: 5
I (26755) BT_AV: Play position changed: 0-ms
I (28035) RCCT: Test passthrough command sent, result 0

I (29015) BT_AV: Audio packet count 100
I (31045) BT_AV: Audio packet count 200
I (33045) BT_AV: Audio packet count 300
I (35045) BT_AV: Audio packet count 400
I (37025) BT_AV: Audio packet count 500
I (38035) RCCT: Test passthrough command sent, result 0

I (39025) BT_AV: Audio packet count 600
I (41025) BT_AV: Audio packet count 700
I (43025) BT_AV: Audio packet count 800
I (45025) BT_AV: Audio packet count 900
I (47035) BT_AV: Audio packet count 1000
I (48035) RCCT: Test passthrough command sent, result 0

I (49025) BT_AV: Audio packet count 1100
I (51025) BT_AV: Audio packet count 1200
I (53025) BT_AV: Audio packet count 1300
I (55025) BT_AV: Audio packet count 1400
I (57025) BT_AV: Audio packet count 1500
I (58035) RCCT: Test passthrough command sent, result 0

I (59025) BT_AV: Audio packet count 1600
I (61015) BT_AV: Audio packet count 1700
I (63045) BT_AV: Audio packet count 1800
I (65045) BT_AV: Audio packet count 1900
I (67045) BT_AV: Audio packet count 2000
I (68035) RCCT: Test passthrough command sent, result 0

I (69045) BT_AV: Audio packet count 2100
I (71025) BT_AV: Audio packet count 2200
I (73025) BT_AV: Audio packet count 2300
I (75025) BT_AV: Audio packet count 2400
I (77025) BT_AV: Audio packet count 2500
I (78035) RCCT: Test passthrough command sent, result 0

I (79025) BT_AV: Audio packet count 2600
I (81035) BT_AV: Audio packet count 2700
I (83025) BT_AV: Audio packet count 2800
I (85025) BT_AV: Audio packet count 2900
I (87025) BT_AV: Audio packet count 3000
I (88035) RCCT: Test passthrough command sent, result 0

I (89025) BT_AV: Audio packet count 3100
I (89525) RCCT: AVRC event notification: 5
I (89525) BT_AV: Play position changed: 10124-ms
I (89525) RCCT: AVRC event notification: 2
I (90125) RCCT: AVRC passthrough rsp: key_code 0x4b, key_state 0
I (90135) RCCT: AVRC passthrough rsp: key_code 0x4c, key_state 0
I (90135) RCCT: AVRC passthrough rsp: key_code 0x4b, key_state 0
I (90135) RCCT: AVRC passthrough rsp: key_code 0x4c, key_state 0
I (90145) RCCT: AVRC passthrough rsp: key_code 0x4b, key_state 0
I (90155) RCCT: AVRC passthrough rsp: key_code 0x4c, key_state 0
I (90155) RCCT: AVRC passthrough rsp: key_code 0x4b, key_state 0
I (91175) BT_AV: A2DP audio state: Suspended
```
[a2dp_sink_passthrough_test.zip](https://github.com/espressif/esp-idf/files/6598354/a2dp_sink_passthrough_test.zip)

boblane1, 2021-06-10:
Hi spleen1981,

Would you please try the following patch to see if the problem still exists?

Thanks.
[0001-fix-a2dp-sink-RxSbcQ-bugs.zip](https://github.com/espressif/esp-idf/files/6630160/0001-fix-a2dp-sink-RxSbcQ-bugs.zip)

spleen1981, 2021-06-11:
@boblane1 problem seems to be fixed with this patch.

closed 2021-06-28 by espressif-bot

pad52, 2023-08-08:
I'm experiencing again the problem with new commits of the btc_a2dp_sink.c
with @boblane1 version problem was gone.