espressif/esp-idf #7102: RMT receive not receiving correctly V4.2.1

MarkEvens, 2021-06-03:
```
#include <stdio.h>
#include <string.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "driver/rmt.h"
#include "ir_tools.h"

static const char *TAG = "example";

static rmt_channel_t example_rx_channel = RMT_CHANNEL_1;

/**
 * @brief RMT Receive Task
 *
 */
static void example_ir_rx_task(void *arg)
{
    uint32_t addr = 0;
    uint32_t cmd = 0;
    uint32_t length = 0;
    bool repeat = false;
    RingbufHandle_t rb = NULL;
    rmt_item32_t *items = NULL;

    rmt_config_t rmt_rx_config = RMT_DEFAULT_CONFIG_RX(CONFIG_EXAMPLE_RMT_RX_GPIO, example_rx_channel);
    rmt_config(&rmt_rx_config);
    rmt_driver_install(example_rx_channel, 1000, 0);

    //get RMT RX ringbuffer
    rmt_get_ringbuf_handle(example_rx_channel, &rb);
    // Start receive
    rmt_rx_start(example_rx_channel, true);
    while (rb) {
         items = (rmt_item32_t *) xRingbufferReceive(rb, &length, 4000);
        if (items) {
            printf("\n items : %d", length);
            length /= 4; // one RMT = 4 Bytes
            printf("\n items : %d", length);
                for(int i = 0; i < length; i++)
                {
                    printf("\n items[%d] duration0 : %d", i, items[i].duration0);
                    printf("\n items[%d] level0 : %d", i, items[i].level0);
                    printf("\n items[%d] duration1 : %d", i, items[i].duration1);
                    printf("\n items[%d] level1 : %d", i, items[i].level1);
                }
            //after parsing the data, return spaces to ringbuffer.
            vRingbufferReturnItem(rb, (void *) items);
        } else {
            break;
        }
    }
    rmt_driver_uninstall(example_rx_channel);
    vTaskDelete(NULL);
}

void app_main(void)
{
    xTaskCreate(example_ir_rx_task, "ir_rx_task", 2048, NULL, 10, NULL);
}
```
This code for receiving so my Remote sending 211 block that means it is only receiving 168 or something
Before it my code is working in V4.2.0

Alvin1Zhang, 2021-06-04:
@MarkEvens Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

MarkEvens, 2021-06-04:
## Environment

- Development Kit:      [ESP32-WROOM-32]
- Kit version (none)
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    v4.2.1
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Linux]

- Using an IDE?: [VS-Code]
- Power Supply:         [USB]

## Problem Description

As my code shown it print received data in form of micro seconds, so my first protocol is sending 211 size of buffer which means total 105 or 106 of "rmt_item32_t" so in above code I try it with memory block 1, so it is not receiving complete data after that i try with 2 memory block so it worked, but again In my next remote It sending 429 raw data and again it not getting complete data. so I change memory block to 3 but it is still not working

### Expected Behavior

I my project I don't have fixed value It is changing and max is 1000 so I want code that can accurately received upto 1000 raw data so 500 of  "rmt_item32_t" item.

### Actual Behavior
one time is show rmt buffer full.
even not getting more than 51 data

### Code to reproduce this issue
- this is my rmt config
```
#define RMT_DEFAULT_CONFIG_RX(gpio, channel_id) \
    {                                           \
        .rmt_mode = RMT_MODE_RX,                \
        .channel = channel_id,                  \
        .gpio_num = gpio,                       \
        .clk_div = 80,                          \
        .mem_block_num = 3,                     \
        .flags = 0,                             \
        .rx_config = {                          \
            .idle_threshold = 12000,            \
            .filter_ticks_thresh = 100,         \
            .filter_en = true,                  \
        }                                       \
    }

```
## Debug Logs
- I am send 429 item means 215 buffer length
```
items : 112
 items : 28
 items[0] duration0 : 361 	 items[0] level0 : 0 	 items[0] duration1 : 532 	 items[0] level1 : 1
 items[1] duration0 : 364 	 items[1] level0 : 0 	 items[1] duration1 : 506 	 items[1] level1 : 1
 items[2] duration0 : 364 	 items[2] level0 : 0 	 items[2] duration1 : 1398 	 items[2] level1 : 1
 items[3] duration0 : 392 	 items[3] level0 : 0 	 items[3] duration1 : 501 	 items[3] level1 : 1
 items[4] duration0 : 340 	 items[4] level0 : 0 	 items[4] duration1 : 557 	 items[4] level1 : 1
 items[5] duration0 : 363 	 items[5] level0 : 0 	 items[5] duration1 : 532 	 items[5] level1 : 1
 items[6] duration0 : 388 	 items[6] level0 : 0 	 items[6] duration1 : 478 	 items[6] level1 : 1
 items[7] duration0 : 418 	 items[7] level0 : 0 	 items[7] duration1 : 478 	 items[7] level1 : 1
 items[8] duration0 : 364 	 items[8] level0 : 0 	 items[8] duration1 : 531 	 items[8] level1 : 1
 items[9] duration0 : 365 	 items[9] level0 : 0 	 items[9] duration1 : 528 	 items[9] level1 : 1
 items[10] duration0 : 365 	 items[10] level0 : 0 	 items[10] duration1 : 531 	 items[10] level1 : 1
 items[11] duration0 : 364 	 items[11] level0 : 0 	 items[11] duration1 : 505 	 items[11] level1 : 1
 items[12] duration0 : 415 	 items[12] level0 : 0 	 items[12] duration1 : 477 	 items[12] level1 : 1
 items[13] duration0 : 391 	 items[13] level0 : 0 	 items[13] duration1 : 504 	 items[13] level1 : 1
 items[14] duration0 : 364 	 items[14] level0 : 0 	 items[14] duration1 : 533 	 items[14] level1 : 1
 items[15] duration0 : 363 	 items[15] level0 : 0 	 items[15] duration1 : 530 	 items[15] level1 : 1
 items[16] duration0 : 391 	 items[16] level0 : 0 	 items[16] duration1 : 504 	 items[16] level1 : 1
 items[17] duration0 : 364 	 items[17] level0 : 0 	 items[17] duration1 : 505 	 items[17] level1 : 1
 items[18] duration0 : 390 	 items[18] level0 : 0 	 items[18] duration1 : 503 	 items[18] level1 : 1
 items[19] duration0 : 389 	 items[19] level0 : 0 	 items[19] duration1 : 507 	 items[19] level1 : 1
 items[20] duration0 : 361 	 items[20] level0 : 0 	 items[20] duration1 : 1403 	 items[20] level1 : 1
 items[21] duration0 : 363 	 items[21] level0 : 0 	 items[21] duration1 : 1373 	 items[21] level1 : 1
 items[22] duration0 : 391 	 items[22] level0 : 0 	 items[22] duration1 : 1370 	 items[22] level1 : 1
 items[23] duration0 : 363 	 items[23] level0 : 0 	 items[23] duration1 : 1401 	 items[23] level1 : 1
 items[24] duration0 : 393 	 items[24] level0 : 0 	 items[24] duration1 : 1342 	 items[24] level1 : 1
 items[25] duration0 : 420 	 items[25] level0 : 0 	 items[25] duration1 : 476 	 items[25] level1 : 1
 items[26] duration0 : 364 	 items[26] level0 : 0 	 items[26] duration1 : 1398 	 items[26] level1 : 1
 items[27] duration0 : 365 	 items[27] level0 : 0 	 items[27] duration1 : 0 	 items[27] level1 : 1

```

MarkEvens, 2021-06-05:
@Alvin1Zhang  have looked in ?

suda-morris, 2021-06-07:
Hi @MarkEvens On esp32, the RMT channel size (hardware) is 64 Words per channel, if you configured the `mem_block_num` to `1`. This is the hardware limitation on esp32. For chips after esp32-s2, the RMT peripheral cans upport ping-pong operations for RX channel, which helps us to receive packages larger than the hardware channel size.
If the package you want to receive should contain `X` items, you have to set the mem_block_num not smaller than `X/64`.

MarkEvens, 2021-06-07:
That means If i want to receive 512 word for that i have to use 8 memory block ?
But I have also sending and RGB led strip How can I get all this thing work together ? any possible way ?

suda-morris, 2021-06-07:
ESP32 doesn't support ping-pong for RX channel, the only way to receive a large package is to increase the `mem_block_num`.
The ringbuffer used in the driver is to store multiple packets for you, in case there comes a lot of packets and the user doesn't retrieve them in time. It can't help eliminate the hardware limitation you met in your project.

closed 2021-06-12 by MarkEvens