espressif/esp-idf #7103: i2s_basic project & any project using i2s seems to not compile for ESP32-C3 via VSCode extension

nicholasdkunes, 2021-06-03:
**CMakeLists.txt:**

```
idf_component_register(
    SRCS "main.c"
    INCLUDE_DIRS "."
)
```

**Code:**

```
#include <stdlib.h>
#include <stddef.h>
#include <inttypes.h>
#include <string.h>
#include <stdio.h>
#include <stdbool.h>
#include <sys/time.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_log.h"
#include "soc/rtc.h"
#include "driver/i2s.h"
#include "driver/gpio.h"

#define TAG "main"

static void init_i2s()
{
	const int sample_rate = 44100;

    /* RX: I2S_NUM_1 */
    i2s_config_t i2s_config_rx = {
	.mode = I2S_MODE_MASTER | I2S_MODE_RX, // Only TX
	.sample_rate = sample_rate,
	.bits_per_sample = I2S_BITS_PER_SAMPLE_32BIT,    // Only 8-bit DAC support
	.channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,   // 2-channels
	.communication_format = I2S_COMM_FORMAT_STAND_MSB,
	.dma_buf_count = 32,                            // number of buffers, 128 max.
	.dma_buf_len = 32 * 2,                          // size of each buffer
	.intr_alloc_flags = ESP_INTR_FLAG_LEVEL1        // Interrupt level 1
	};

	i2s_pin_config_t pin_config_rx = {
		.bck_io_num = GPIO_NUM_2,
		.ws_io_num = GPIO_NUM_3,
		.data_out_num = I2S_PIN_NO_CHANGE,
		.data_in_num = GPIO_NUM_0
	};

	i2s_driver_install(I2S_NUM_0, &i2s_config_rx, 0, NULL);
	i2s_set_pin(I2S_NUM_0, &pin_config_rx);

}

void task_megaphone()
{
	uint16_t buf_len = 1024;
	char *buf = calloc(buf_len, sizeof(char));

	struct timeval tv = {0};
	struct timezone *tz = {0};
	gettimeofday(&tv, &tz);
	uint64_t micros = tv.tv_usec + tv.tv_sec * 1000000;
	uint64_t micros_prev = micros;
	uint64_t delta = 0;

	init_i2s();

	int cnt = 0;

	while(1)
	{
		char *buf_ptr_read = buf;
		char *buf_ptr_write = buf;

		// read whole block of samples
		size_t bytes_read = 0;
		while(bytes_read == 0) {
			i2s_read(I2S_NUM_0, buf, buf_len, &bytes_read, 0);
		}

		uint32_t samples_read = bytes_read / 2 / (I2S_BITS_PER_SAMPLE_32BIT / 8);

		//  convert 2x 32 bit stereo -> 1 x 16 bit mono
		for(int i = 0; i < samples_read; i++) {

			// const char samp32[4] = {ptr_l[0], ptr_l[1], ptr_r[0], ptr_r[1]};

			// left
			buf_ptr_write[0] = buf_ptr_read[2]; // mid
			buf_ptr_write[1] = buf_ptr_read[3]; // high

			// right
			buf_ptr_write[2] = buf_ptr_write[0]; // mid
			buf_ptr_write[3] = buf_ptr_write[1]; // high

			buf_ptr_write += 2 * (I2S_BITS_PER_SAMPLE_16BIT / 8);
			buf_ptr_read += 2 * (I2S_BITS_PER_SAMPLE_32BIT / 8);
		}

		cnt += samples_read;

		if(cnt >= 44100) {
			gettimeofday(&tv, &tz);
			micros = tv.tv_usec + tv.tv_sec * 1000000;
			delta = micros - micros_prev;
			micros_prev = micros;
			printf("%d samples in %" PRIu64 " usecs\n", cnt, delta);

			cnt = 0;
		}
	}
}

/**
 * entry point
 */
void app_main()
{
    printf("starting app_main()\n");
    task_megaphone();
}

```

**Output:**

```
FAILED: template-app.elf 
cmd.exe /C "cd . && C:\Users\nicho\.espressif\tools\riscv32-esp-elf\1.24.0.123_64eb9ff-8.4.0\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe  -march=rv32imc  -nostartfiles -march=rv32imc --specs=nosys.specs @CMakeFiles\template-app.elf.rsp  -o template-app.elf  && cd ."
c:/users/nicho/.espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/driver/libdriver.a(i2s.c.obj): in function `i2s_driver_uninstall':
C:/Users/nicho/Desktop/esp-idf/components/driver/i2s.c:1017: undefined reference to `rtc_clk_apll_enable'
c:/users/nicho/.espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/driver/libdriver.a(i2s.c.obj): in function `i2s_set_clk':
C:/Users/nicho/Desktop/esp-idf/components/driver/i2s.c:397: undefined reference to `rtc_clk_apll_enable'
c:/users/nicho/.espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/hal/libhal.a(i2s_hal.c.obj): in function `i2s_hal_init':
C:/Users/nicho/Desktop/esp-idf/components/hal/i2s_hal.c:273: undefined reference to `I2S0'
c:/users/nicho/.espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: C:/Users/nicho/Desktop/esp-idf/components/hal/i2s_hal.c:273: undefined reference to `I2S0'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

ginkgm, 2021-06-04:
Hi @nicholasdkunes , I'm sorry to say the I2S driver on C3 hasn't been implemented yet.

So the example build is also supposed to fail. About the vscode plugin part, we will see if we have better way to show the failure/unsupported message.

closed 2021-06-04 by nicholasdkunes

Alvin1Zhang, 2021-08-10:
I2S support in ESP32-C3 is available https://github.com/espressif/esp-idf/commit/8131d6f46d690e9cb60cc5cd457863cc5479351f.