espressif/esp-idf #7105: crashed LoadProhibited when close bt

buttonwoodsz, 2021-06-04:
# Environment
- Development Kit: [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used: [ESP32-WROOM-32]
- IDF version (run git describe --tags to find it): v4.2
- Build System: idf.py
- Compiler version (run xtensa-esp32-elf-gcc --version to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

# Problem Description
when close bt platform, call function such as
[image]

then system crashed occasionally ！
- log 1
```
GAP procedure initiated: stop advertising.
GAP procedure initiated: terminate connection; conn_handle=0 hci_reason=19
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008ab08  PS      : 0x00060530  A0      : 0x00000000  A1      : 0x3ff813b0  
A2      : 0x00000000  A3      : 0x3ffc3338  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x4008ab00  A7      : 0x00000000  A8      : 0x800d450b  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x3ff813b0  A12     : 0x3ffb7238  A13     : 0x00000000  
A14     : 0x00000001  A15     : 0x3ff803e8  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40081ca5  LEND    : 0x40081cad  LCOUNT  : 0x00000027  

Backtrace:0x4008ab05:0x3ff813b0 |<-CORRUPTED
```
- log 2

[image]

dhrishi, 2021-06-04:
@buttonwoodsz Thank you for reporting the issue. Can you please update about the application you are using?
If it is a custom application, then is it possible to share a patch/code snippet that can be applied on any of the existing NimBLE examples to re-create this issue?

buttonwoodsz, 2021-06-04:
This is an occasional problem that is difficult to reproduce. Can you check the use of illegal pointers based on the log position?
Actually I’m not sure that turning off ble caused this problem.
BTW, I would try to make a example to reproduce it. Thanks

buttonwoodsz, 2021-06-17:
reproduce once, crashed again when disable bt, this time have  stack information
[image]
it seems that problem at BLE_HS_LOG function

buttonwoodsz, 2021-06-17:
add log
W (1623917428) trans-pdu: bt trans pdu deinit

I (423677) bt: Disable BLE.

GAP procedure initiated: stop advertising.

GAP procedure initiated: terminate connection; conn_handle=0 hci_reason=19Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.

Memory dump at 0x40081a44: 5a651a0c fffa46ff 30004136

Core  0 register dump:

PC      : 0x40081a4b  PS      : 0x00060031  A0      : 0x8008fc5d  A1      : 0x3ffbe5d0 

A2      : 0x3ffc3300  A3      : 0x00000000  A4      : 0x872d04c8  A5      : 0x3ffc32e8 

A6      : 0x00000000  A7      : 0x00000000  A8      : 0x80081aa4  A9      : 0x3ffbe5c0 

A10     : 0x00000003  A11     : 0x00060023  A12     : 0x00060021  A13     : 0x3f42bbc0 

A14     : 0x3ffdf130  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x00000000 

EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff 

Core  0 was running in ISR context:

EPC1    : 0x40081a4b  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

Backtrace:0x40081a48:0x3ffbe5d0 0x4008fc5a:0x3ffbe5f0 0x4008f137:0x3ffbe610 0x4008f2e5:0x3ffbe630 0x40082132:0x3ffbe640 0x401203f1:0x3ffdec80 0x40120cfd:0x3ffdeca0 0x400d2931:0x3ffdecc0 0x4000bd83:0x3ffdece0 0x4000117d:0x3ffded00 0x400592fe:0x3ffded20 0x4005937a:0x3ffded40 0x40058bbf:0x3ffded60 0x4019832f:0x3ffded90 0x4019d913:0x3ffdedc0 0x4019da95:0x3ffdf0d0 0x401a94c2:0x3ffdf100 0x400926ae:0x3ffdf130 0x400f8c46:0x3ffdf180 0x40101a08:0x3ffdf1b0 0x40101a53:0x3ffdf1d0 0x400fc2ba:0x3ffdf1f0 0x400ea8f3:0x3ffdf210

ELF file SHA256: 21a7ba325665fc31

Entering gdb stub now.

RahulEspressif, 2021-06-17:
Hi @buttonwoodsz,   Thanks for sharing this log. Can you also please let us know the commit id you are on ( first commit information of "git log" should be enough)  , so we can match the backtrace and debug this further ?

buttonwoodsz, 2021-06-17:
[image]
tag: v4.2

RahulEspressif, 2021-06-17:
Hi @buttonwoodsz  , this is a 6 month old commit. I wanted to check if it is possible for you to move latest headsync ?  I have put a simulation code as yours , but don't see this issue yet in my testing. Will continue to check more. Please let me know in case it is possible to pull in changes of v4.2 and move to headsync at your end and see if issue is still seen ?

Alvin1Zhang, 2021-06-30:
@buttonwoodsz Thanks for reporting. Would you please help share if any updates or details for this issue? Thanks.

buttonwoodsz, 2021-07-27:
after fix multi-thread safe at ble callback function, this bug seems to have been resolved, so close it now

closed 2021-07-27 by buttonwoodsz