espressif/esp-idf #7107: ESP-IDF 4.4 on arduino platform breaks ETH PHY support

Simonefardella, 2021-06-04:
Hi guys, as i've reported in the esp32-arduino repo, starting from the IDF 4.x version, there are in place a lot of changes that breaks Ethernet on most of third party boards (Like Olimex, etc...).

For example, the driver initialization doesn't have the CLK GPIO option selectable and nowhere is documented that.

This represent for the overall esp32-arduino users a big problem to face...: here there are all details, scope traces and informations: [ISSUE HERE](https://github.com/espressif/arduino-esp32/issues/5239)

negativekelvin, 2021-06-05:
It's a compile time option so when libraries are precompiled for Arduino the choice is locked.  I think it could be made configurable at runtime, there's not really any penalty. This could also benefit micropython and other platforms that need to support multiple hardware configurations.  What do you think @suda-morris ?

https://github.com/espressif/esp-idf/blob/1f54d17503ff17d830de75eed4eb47bc6a3aafc3/components/hal/esp32/emac_hal.c#L84-L130

atanisoft, 2021-06-05:
@negativekelvin in IDF v3.x it was runtime configurable.

suda-morris, 2021-06-07:
Thanks for reporting, IDF will add that runtime "mac clock config" option back soon.

Simonefardella, 2021-06-07:
Thanks @suda-morris, we are waiting for that! :)

Simonefardella, 2021-06-10:
@suda-morris, any news about this issue?

kostaond, 2021-06-11:
@Simonefardella It's going to be worked very soon. There is EMAC HAL code refactor update pending to be merged which needs to be finished first. It's been already review so we could start "mac clock config" soon. Thank you for your patience.

kostaond, 2021-06-23:
@Simonefardella just FYI, the update is in review.

closed 2021-07-16 by espressif-bot