espressif/esp-idf #7109: Wifi SoftAP example crashes when client connects

IVData, 2021-06-06:
## Environment

- Development Kit: esp32-d1 mini
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.2.1
- Build System: cmake
- Compiler version: 8.4.0
- Operating System:  Windows
- (Windows only) environment type: vscode extension
- Using an IDE?: Yes
- Power Supply: Tried usb, external

## Problem Description

Wifi AP example crashes immediately when a client connects

### Steps to reproduce

1. Flash wifi AP example
2. Connect client to AP

## Debug Logs

```
I (31) boot: ESP-IDF v4.2.1-dirty 2nd stage bootloader
I (31) boot: compile time 15:36:04
I (31) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x15d30 ( 89392) map
I (148) esp_image: segment 1: paddr=0x00025d58 vaddr=0x3ffb0000 size=0x03228 ( 12840) load
I (154) esp_image: segment 2: paddr=0x00028f88 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at C:/Users/MystX/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (155) esp_image: segment 3: paddr=0x00029394 vaddr=0x40080404 size=0x06c84 ( 27780) load
I (176) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x6f908 (456968) map
0x400d0020: _stext at ??:?

I (350) esp_image: segment 5: paddr=0x0009f930 vaddr=0x40087088 size=0x0e4e0 ( 58592) load
0x40087088: reconfigure_all_wdts at C:/Users/MystX/esp/esp-idf/components/esp_system/panic.c:123

I (388) boot: Loaded app from partition at offset 0x10000
I (388) boot: Disabling RNG early entropy source...
I (388) cpu_start: Pro cpu up.
I (392) cpu_start: Application information:
I (397) cpu_start: Project name:     wifi_softAP
I (402) cpu_start: App version:      1
I (406) cpu_start: Compile time:     Jun  6 2021 15:35:43
I (412) cpu_start: ELF file SHA256:  3190b2953952bbf4...
I (418) cpu_start: ESP-IDF:          v4.2.1-dirty
I (424) cpu_start: Starting app cpu, entry point is 0x400818a4
0x400818a4: call_start_cpu1 at C:/Users/MystX/esp/esp-idf/components/esp32/cpu_start.c:287

I (415) cpu_start: App cpu up.
I (434) heap_init: Initializing. RAM available for dynamic allocation:
I (441) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (447) heap_init: At 3FFB77E0 len 00028820 (162 KiB): DRAM
I (454) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (460) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (466) heap_init: At 40095568 len 0000AA98 (42 KiB): IRAM
I (473) cpu_start: Pro cpu start user code
I (491) spi_flash: detected chip: generic
I (492) spi_flash: flash io: dio
I (492) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (542) wifi softAP: ESP_WIFI_MODE_AP
I (562) wifi:wifi driver task: 3ffbffc4, prio:23, stack:6656, core=0
I (562) system_api: Base MAC address is not set
I (562) system_api: read default base MAC address from EFUSE
I (572) wifi:wifi firmware version: c44a5e7
I (572) wifi:wifi certification version: v7.0
I (572) wifi:config NVS flash: enabled
I (572) wifi:config nano formating: disabled
I (582) wifi:Init data frame dynamic rx buffer num: 32
I (582) wifi:Init management frame dynamic rx buffer num: 32
I (592) wifi:Init management short buffer num: 32
I (592) wifi:Init dynamic tx buffer num: 32
I (602) wifi:Init static rx buffer size: 1600
I (602) wifi:Init static rx buffer num: 10
I (602) wifi:Init dynamic rx buffer num: 32
I (612) wifi_init: rx ba win: 6
I (612) wifi_init: tcpip mbox: 32
I (622) wifi_init: udp mbox: 6
I (622) wifi_init: tcp mbox: 6
I (622) wifi_init: tcp tx win: 5744
I (632) wifi_init: tcp rx win: 5744
I (632) wifi_init: tcp mss: 1440
I (642) wifi_init: WiFi IRAM OP enabled
I (642) wifi_init: WiFi RX IRAM OP enabled
I (652) phy_init: phy_version 4660,0162888,Dec 23 2020
W (792) phy_init: saving new calibration data because of checksum failure, mode(0)
I (812) wifi:mode : softAP (a4:cf:12:6d:c9:e9)
I (812) wifi:Total power save buffer number: 16
I (812) wifi:Init max length of beacon: 752/752
I (812) wifi:Init max length of beacon: 752/752
I (822) wifi softAP: wifi_init_softap finished. SSID:myssid password:mypassword channel:1
I (16462) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (16472) wifi:station: 1a:cc:52:73:2d:d8 join, AID=1, bgn, 40U
I (16512) wifi softAP: station 1a:cc:52:73:2d:d8 join, AID=1
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x00000000  PS      : 0x00060930  A0      : 0x800fc0b8  A1      : 0x3ffbfd60
A2      : 0x3ffb8dfc  A3      : 0x3ffb87d0  A4      : 0x3ffb87e8  A5      : 0x3ffb87f1
A6      : 0x0000002d  A7      : 0x000000d8  A8      : 0x80110aec  A9      : 0x3ffb19ec
A10     : 0x3ffb8dfc  A11     : 0x3ffb87d0  A12     : 0x3ffb87e8  A13     : 0x3ffb87f1
A14     : 0x00000009  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x00000014
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace:0xfffffffd:0x3ffbfd60 0x400fc0b5:0x3ffbfd80 0x400fc373:0x3ffbfe90 0x40091e09:0x3ffbfeb0 0x4008fd78:0x3ffbff00 0x40087851:0x3ffbff30
0x400fc0b5: hostap_input at ??:?

0x400fc373: ap_rx_cb at ??:?

0x40091e09: ppRxPkt at ??:?

0x4008fd78: ppTask at ??:?

0x40087851: vPortTaskWrapper at C:/Users/MystX/esp/esp-idf/components/freertos/xtensa/port.c:143

ELF file SHA256: 3190b2953952bbf4

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7104
ho 0 tail 12 room 4
load:0x40078000,len:13212
load:0x40080400,len:4568
0x40080400: _init at ??:?

entry 0x400806f4
```

## Other items if possible

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/6603603/sdkconfig.txt)
[wifi_softAP.elf.txt](https://github.com/espressif/esp-idf/files/6603604/wifi_softAP.elf.txt)

felmue, 2021-06-07:
Hello @IVData 

hmm, I can run the softAP example just fine. That said, I am using the same IDF version 4.2.1 as you do, but on different hardware: building on MacOSX (command line) and running on an ESP32 LyraT board.

- have you tried different clients? (I've tested with an iPadMini.)
- have you tried a different ESP32 board?
- have you considered trying with the latest IDF master?
- have you tried any other example using WiFi? Do they work?

BTW: I wonder if this line from your log could be the issue (I don't see that in my log.)

`W (792) phy_init: saving new calibration data because of checksum failure, mode(0)`

Good luck!
Felix

For reference - my log when the client connects:
```
I (783) wifi softAP: wifi_init_softap finished. SSID:myssid password:mypassword channel:1
I (13743) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (13743) wifi:station: ea:ae:23:85:8e:2e join, AID=1, bgn, 20
I (13763) wifi softAP: station ea:ae:23:85:8e:2e join, AID=1
I (13973) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
```

IVData, 2021-06-07:
Annoyingly, it seems to only happen for my android phone (OnePlus 5T). When my laptop or my partner's phone connects there is no issue.
I have also just compiled on a different PC (Linux vs Windows) and I don't have the phy_init message but still get the crash. So I don't think it's that.

1. So yes, some clients connect successfully, some (at least 1) don't.
2. I've tried 2 different kinds of boards (an ESP32 D1 mini and the product board I'm working on)
3. Not yet
4. I've tried various code online for bringing up the AP and had this crash with all of them. So I tried to just run the example before reporting here.

nishanth-radja, 2021-06-07:
@IVData can we have a sniffer capture while you are trying to connect,So that we can find out what is the configuration on the phone that is causing this crash.

IVData, 2021-06-07:
@nishanth-radja sure. Could you kindly let me know what the easiest way to do that is? I only have a linux laptop and android phones with wifi in them.

nishanth-radja, 2021-06-07:
@IVData use the following script on the linux laptop.you should be able to capture the packets on laptop.
Hope wire shark is installed on the laptop.else Pls install the wireshark using 
"sudo add-apt-repository universe 
sudo apt install wireshark"

replace the wlan0 with the wireless interface on your laptop. 
ifconfig wlan0 down
iwconfig wlan0 mode monitor
ifconfig wlan0 up
iwconfig wlan0 channel 1
wireshark

IVData, 2021-06-07:
Yeah that is what I did and I don't get any output. I have a USB wifi adapter that I think supports monitor mode well. But I will have to try that tomorrow.

IVData, 2021-06-07:
Hi,
I managed to get this file, which should contain a connection from my phone that caused the crash. Hope this helps.

It also seems like the phone will sometimes connect successfully if it is not switching from another wifi network. I will try to get a capture of a successful connection for comparison

sagb2015, 2021-06-09:
Hi @IVData, We acknowledge the bug. We will fix it soon. It is related to "20/40 coex management" public action frame handling which is specific to this client. As per our analysis latest master should not have this bug.

IVData, 2021-06-09:
Thanks very much for the response.

Alvin1Zhang, 2021-06-30:
Thanks for reporting, sorry for late reply, the fix is available at https://github.com/espressif/esp-idf/commit/1e8fca74f254801b5b90df8608c958ee6bdafabe, feel free to reopen.

closed 2021-06-30 by Alvin1Zhang