espressif/esp-idf #7114: ESP32C3 i2s.c and i2s_hal.c don't compile

eyewy, 2021-06-07:
esp-idf-v4.3-rc/components/driver/i2s.c:1017: undefined reference to `rtc_clk_apll_enable'
esp-idf-v4.3-rc/components/hal/i2s_hal.c:273: undefined reference to `I2S0'

looking at  esp-idf/components/esp_hw_support/port/esp32c3/rtc_clk.c I can see that rtc_clk_apll_enable is not defined in there, which brings me to conclusion that I2S does not work at all on ESP32C3?

ESP-Marius, 2021-06-08:
Unfortunately I2S support for C3 did not make it into v4.3, but it'll be there for v4.4 (and hopefully on master soon)

The README.md for each example lists which targets are supported, but I agree we need to make this more obvious for users if they are compiling for a non-supported target.

xw901103, 2021-07-10:
for rtc_clk_apll_enable, according to their unfinished tech reference manual, there is no such thing is Audio PLL. you can create an empty definition to bypass this in %ESP-IDF%/esp_hw_support/port/esp32c3/rtc_clk.c

for I2S0, it should be I2S1 according to the %ESP-IDF%/esp32c3/ld/esp32c3.peripherals.ld file. You can modify the I2S_LL_GET_HW macro in file %ESP-IDF%/hal/esp32c3/include/hal/i2s_ll.h to match the peripherals.Id file. In addition, please make sure that the %ESP-IDF%/soc/esp32c3/include/soc/i2s_struct.h file is also modified to match I2S1 definition instead of I2S0.

Furthermore, there is %ESP-IDF%/soc/esp32c3/i2s_periph.c file that is missing values. Based on %ESP-IDF%/soc/esp32c3/include/soc/gpio_sig_map.h file, it seems to be the following.

.o_bck_in_sig = I2SO_BCK_IN_IDX,
.o_ws_in_sig = I2SO_WS_IN_IDX,
.o_bck_out_sig = I2SO_BCK_OUT_IDX,
.o_ws_out_sig = I2SO_WS_OUT_IDX,
.o_data_out_sig = I2SO_SD_OUT_IDX,
.i_bck_in_sig = I2SI_BCK_OUT_IDX,
.i_ws_in_sig = I2SI_WS_OUT_IDX,
.i_bck_out_sig = I2SI_BCK_IN_IDX,
.i_ws_out_sig = I2SI_WS_IN_IDX,
.i_data_in_sig = I2SI_SD_IN_IDX

After all these changes in esp-idf folder, you should be able to compile your project now.

eyewy, 2021-07-10:
Have you got it working on C3? This may compile, but perhaps won't work as many definitions are incomplete in:
esp-idf/components/hal/esp32c3/include/hal/i2s_ll.h

xw901103, 2021-07-10:
attached gist is what I got so far. I have checked their WIP ESP32-S3 (it is using that GDMA as well) tech reference manual too and nothing there can guide us for low level I2S register manipulation. In addition, %ESP-IDF%/soc/esp32c3/include/soc/i2s_struct.h seems to be incomplete/incorrect as well. Please someone enlighten us on how to make this I2S (with GDMA) works.

https://gist.github.com/xw901103/5ff1c37590d2a4659ef09fe2a4babc3d

L-KAYA, 2021-07-12:
Exactly, I2S on ESP32-C3 and ESP32-S3 uses GDMA instead of a dedicated DMA, and APLL is useless on these two chips. 

Additionally, new I2S will support TDM mode which means you can use up to 16 channels and more flexible slot configurations.

The good news is that the new I2S driver will be available on master soon (I guess won't be more than two weeks).

If you still want to use I2S by yourself anyway, generally, 
1. You should switch to master branch and use GDMA interface to transport data. 
2. Please register  I2S GDMA in `i2s_driver_install` using `gdma_new_channel`, `gdma_connect` and `gdma_register_rx_event_callbacks`
3. Remember you should write a new gdma callback function for `gdma_register_rx_event_callbacks`. 
4. Than use `gdma_start` to start the GDMA transmition in `i2s_start`.

It's hard to tell all the details here, also I don't recommend you to use GDMA by yourself, because hardware has been changed due to the new features. My suggestion is waiting for the new I2S driver on the master branch.

Alvin1Zhang, 2021-11-30:
Thanks for reporting, this has been implemented at https://github.com/espressif/esp-idf/commit/8131d6f46d690e9cb60cc5cd457863cc5479351f, and will be part of IDF v4.4 release, feel free to reopen. Thanks.

closed 2021-11-30 by Alvin1Zhang