espressif/esp-idf #7115: esp idf PPPOS example not working

PIYUSHGIRI2021, 2021-06-08:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit): |v4]
- Module or chip used:  ESP32-WROVER-E
- IDF version (run ``git describe --tags`` to find it):
    v4.4-dev-1594-g1d7068e4b
- Build System:         |idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
     xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: PowerShell
- Using an IDE?: Yes (VS code)
- Power Supply:         [USB]

## Problem Description

hello everyone,
  i am using esp idf example protocol PPPOS code and i am using quectel M95 GSM module,i run this code before 20 days it will work fine ,but same code i compile and monitor using idf.py it will get below error,i download new expressif esp idf from github...in that some changes done,but i monitor this code than get error and crash esp32,please let me know how can i resolve this issue,thanks. please below attache log .this error comes (xQueueGenericReceive)- assert failed!

### Expected Behavior
I (1276) pppos_example: Module: SIMCOM_SIM800L
I (1276) pppos_example: Operator: "CHINA MOBILE"
I (1276) pppos_example: IMEI: 865992039850864
I (1276) pppos_example: IMSI: 460007454185220
I (1476) pppos_example: rssi: 25, ber: 0
I (1676) pppos_example: Battery voltage: 4674 mV
I (1876) pppos_example: Modem PPP Started
I (2806) pppos_example: Modem Connect to PPP Server
I (2806) pppos_example: ~~~~~~~~~~~~~~
I (2806) pppos_example: IP          : 10.188.173.2
I (2806) pppos_example: Netmask     : 255.255.255.255
I (2816) pppos_example: Gateway     : 192.168.254.254
I (2816) pppos_example: Name Server1: 211.136.112.50
I (2826) pppos_example: Name Server2: 211.136.150.66
I (2826) pppos_example: ~~~~~~~~~~~~~~
I (2836) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (2846) pppos_example: MQTT other event id: 7
I (8156) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
I (8826) pppos_example: MQTT_EVENT_CONNECTED
I (8826) pppos_example: sent subscribe successful, msg_id=26237
I (9526) pppos_example: MQTT_EVENT_SUBSCRIBED, msg_id=26237
I (9526) pppos_example: sent publish successful, msg_id=0
I (10326) pppos_example: MQTT_EVENT_DATA
TOPIC=/topic/esp-pppos
DATA=esp32-pppos
I (10376) pppos_example: Modem PPP Stopped
I (14526) pppos_example: Send send message [Welcome to ESP32!] ok
I (15076) pppos_example: Power down
### Actual Behavior
0x400818e4: call_start_cpu1 at C:/Users/Frank/Desktop/esp-idf/components/esp32/cpu_start.c:287

I (379) cpu_start: App cpu up.
I (397) heap_init: Initializing. RAM available for dynamic allocation:
I (404) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (410) heap_init: At 3FFB39F8 len 0002C608 (177 KiB): DRAM
I (417) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (423) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (429) heap_init: At 4008B744 len 000148BC (82 KiB): IRAM
I (436) cpu_start: Pro cpu start user code
I (454) spi_flash: detected chip: generic
I (455) spi_flash: flash io: dio
W (455) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (465) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (478) uart: queue free spaces: 30
I (828) pppos_example: Module: Quectel_M95
I (828) pppos_example: Operator: "Vodafone - Gujarat"
I (828) pppos_example: IMEI: 862462032305913
I (828) pppos_example: IMSI: 404242306491057
I (878) pppos_example: rssi: 31, ber: 0
I (938) pppos_example: Battery voltage: 3873 mV
I (1148) pppos_example: Modem PPP Started
I (1148) pppos_example: PPP state changed event 259
I (1148) pppos_example: PPP state changed event 262
I (1248) pppos_example: PPP state changed event 263
C:/Users/Frank/Desktop/esp-idf/components/freertos/queue.c:1448 (xQueueGenericReceive)- assert failed!

abort() was called at PC 0x40088441 on core 0
0x40088441: xQueueGenericReceive at C:/Users/Frank/Desktop/esp-idf/components/freertos/queue.c:1448 (discriminator 3)

Backtrace:0x4008501f:0x3ffb8b40 0x400856bd:0x3ffb8b60 0x4008938a:0x3ffb8b80 0x40088441:0x3ffb8bf0 0x40088591:0x3ffb8c30 0x40132b2d:0x3ffb8c50 0x40132c49:0x3ffb8c90 0x400856c5:0x3ffb8cb0
0x4008501f: panic_abort at C:/Users/Frank/Desktop/esp-idf/components/esp_system/panic.c:330

0x400856bd: esp_system_abort at C:/Users/Frank/Desktop/esp-idf/components/esp_system/system_api.c:106

0x4008938a: abort at C:/Users/Frank/Desktop/esp-idf/components/newlib/abort.c:46

0x40088441: xQueueGenericReceive at C:/Users/Frank/Desktop/esp-idf/components/freertos/queue.c:1448 (discriminator 3)

0x40088591: xQueueTakeMutexRecursive at C:/Users/Frank/Desktop/esp-idf/components/freertos/queue.c:634

0x40132b2d: esp_event_loop_run at C:/Users/Frank/Desktop/esp-idf/components/esp_event/esp_event.c:560

0x40132c49: esp_event_loop_run_task at C:/Users/Frank/Desktop/esp-idf/components/esp_event/esp_event.c:115 (discriminator 15)

0x400856c5: vPortTaskWrapper at C:/Users/Frank/Desktop/esp-idf/components/freertos/xtensa/port.c:143

ELF file SHA256: 9f2e89dffc21e72c

Rebooting...
ets Jul 29 2019 12:21:46

### Steps to reproduce

1.i attache esp32 UART1 for AT command transfer to gsm module.
2. Quectel m95 GSM module connect using usb to DB9 converter cable.

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

static void on_ppp_changed(void *arg, esp_event_base_t event_base,
                           int32_t event_id, void *event_data)
{
    ESP_LOGI(TAG, "PPP state changed event %d", event_id);
    if (event_id == NETIF_PPP_ERRORUSER) {
        /* User interrupted event from esp-netif */
        esp_netif_t *netif = *(esp_netif_t**)event_data;
        ESP_LOGI(TAG, "User interrupted event from netif:%p", netif);
    }
}

i think here ,when PPP started after this error will come.

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
I (1148) pppos_example: PPP state changed event 262
I (1248) pppos_example: PPP state changed event 263
C:/Users/Frank/Desktop/esp-idf/components/freertos/queue.c:1448 (xQueueGenericReceive)- assert failed!

abort() was called at PC 0x40088441 on core 0
0x40088441: xQueueGenericReceive at C:/Users/Frank/Desktop/esp-idf/components/freertos/queue.c:1448 (discriminator 3)

Backtrace:0x4008501f:0x3ffb8b40 0x400856bd:0x3ffb8b60 0x4008938a:0x3ffb8b80 0x40088441:0x3ffb8bf0 0x40088591:0x3ffb8c30 0x40132b2d:0x3ffb8c50 0x40132c49:0x3ffb8c90 0x400856c5:0x3ffb8cb0
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

david-cermak, 2021-06-11:
Hi @PIYUSHGIRI2021 

Are you really running the same example code from IDF as your reference: 1d7068e4be430edd92bb63f2d922036dcf5c3cc1? Anything changed, updated, added?

The code you've attached to reproduce this has lost its formatting and some non-ascii characters, but they still look like the unchanged example:

https://github.com/espressif/esp-idf/blob/1d7068e4be430edd92bb63f2d922036dcf5c3cc1/examples/protocols/pppos_client/main/pppos_client_main.c#L170-L179

Based on the backtrace, it looks like a memory corruption of some kind, as a separate thread runs another instance of event handler (which apparently ran before, per your log) trying to lock its mutex (again) but now, the mutex ptr is a `nullptr`. Either the event loop has been deleted and destroyed (somehow?) when trying to establish the connection or some other task has written to its memory.

Do you use any authentication protocol? (asking as it crashes just after the `NETIF_PPP_PHASE_AUTHENTICATE` event).

I've tested the example on that version of IDF and everything looks good.

Alvin1Zhang, 2021-08-18:
@PIYUSHGIRI2021 Thanks for reporting and will close due to short of feedback, feel free to reopen with more updates.

closed 2021-08-24 by Alvin1Zhang