espressif/esp-idf #7116: Cannot install ESP IDF tools 2.3 on Windows

balix-microej, 2021-06-08:
## Environment

- Development Kit: none
- IDF version (run ``git describe --tags`` to find it): v4.2.1
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): N/A
- Operating System: Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: No

## Problem Description

The ESP IDF tools 2.3 cannot be installed on Windows.

### Expected Behavior

ESP IDF tools installation.

### Actual Behavior

Failed to install ESP IDF tools.

### Steps to reproduce

Following the environment setup instructions I downloaded the IDF tools setup as instructed [here](https://docs.espressif.com/projects/esp-idf/en/v4.2.1/esp32/get-started/windows-setup.html#esp-idf-tools-installer).

The installation failed on the first try due to Visual Studio Code C++ 14 missing:

`error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/`

After installing Visual Studio Code with some warnings (there are files that I cannot download) it failed with the same error and same message as before.

I modified the VSC installation to keep only the optional MSVC compilers and libraries. This time it failed during a python dependency installation:

```
                            creating build\temp.win-amd64-3.9\Release\src\gevent\libev
                            C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -DFD_SETSIZE=1024 -D_WIN32=1 -DEV_STANDALONE=1 -DLIBEV_EMBED=1 -DEV_COMMON= -DEV_CLEANUP_ENABLE=0 -DEV_EMBED_ENABLE=0 -DEV_PERIODIC_ENABLE=0 -Isrc/gevent/libev -IC:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\include -IC:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\Include -IC:\Users\balix\AppData\Local\Temp\pip-install-qppw7a7c\gevent_34b48c843146446b823951a1298009f5\deps -IC:\Users\balix\AppData\Local\Temp\pip-install-qppw7a7c\gevent_34b48c843146446b823951a1298009f5\src\gevent\libev -IC:\Users\balix\AppData\Local\Temp\pip-install-qppw7a7c\gevent_34b48c843146446b823951a1298009f5\deps\libev -Isrc/gevent -Isrc/gevent/libev -Isrc/gevent/resolver -I. -IC:\Users\balix\.espressif\python_env\idf4.2_py3.9_env\include -IC:\Program Files\WindowsApps  C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -DFD_SETSIZE=1024 -D_WIN32=1 -DEV_STANDALONE=1 -DLIBEV_EMBED=1 -DEV_COMMON= -DEV_CLEANUP_ENABLE=0 -DEV_EMBED_ENABLE=0 -DEV_PERIODIC_ENABLE=0 -Isrc/gevent/libev -IC:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\include -IC:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\Include -IC:\Users\balix\AppData\Local\Temp\pip-install-qppw7a7c\gevent_34b48c843146446b823951a1298009f5\deps -IC:\Users\balix\AppData\Local\Temp\pip-install-qppw7a7c\gevent_34b48c843146446b823951a1298009f5\src\gevent\libev -IC:\Users\balix\AppData\Local\Temp\pip-install-qppw7a7c\gevent_34b48c843146446b823951a1298009f5\deps\libev -Isrc/gevent -Isrc/gevent/libev -Isrc/gevent/resolver -I. -IC:\Users\balix\.espressif\python_env\idf4.2_py3.9_env\include -IC:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520  C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -DFD_SETSIZE=1024 -D_WIN32=1 -DEV_STANDALONE=1 -DLIBEV_EMBED=1 -DEV_COMMON= -DEV_CLEANUP_ENABLE=0 -DEV_EMBED_ENABLE=0 -DEV_PERIODIC_ENABLE=0 -Isrc/gevent/libev -IC:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\include -IC:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\Include -IC:\Users\balix\AppData\Local\Temp\pip-install-qppw7a7c\gevent_34b48c843146446b823951a1298009f5\deps -IC:\Users\balix\AppData\Local\Temp\pip-install-qppw7a7c\gevent_34b48c843146446b823951a1298009f5\src\gevent\libev -IC:\Users\balix\AppData\Local\Temp\pip-install-qppw7a7c\gevent_34b48c843146446b823951a1298009f5\deps\libev -Isrc/gevent -Isrc/gevent/libev -Isrc/gevent/resolver -I. -IC:\Users\balix\.espressif\python_env\idf4.2_py3.9_env\include -IC:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\include -IC:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\include /Tcsrc/gevent/libev/callbacks.c /Fobuild\temp.win-amd64-3.9\Release\src/gevent/libev/callbacks.obj
                            callbacks.c
                            src/gevent/libev/callbacks.c(2): fatal error C1083: Cannot open include file: 'stddef.h': No such file or directory
                            error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\BuildTools\\VC\\Tools\\MSVC\\14.29.30037\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2
```

The complete log file is attached to this post.

### Code to reproduce this issue

None.

## Debug Logs

[Setup Log 2021-06-08 #003.txt](https://github.com/espressif/esp-idf/files/6614918/Setup.Log.2021-06-08.003.txt)

## Other items if possible

None.

georgik, 2021-06-08:
@balix-microej  Please use Installer 2.8 or newer which contains embedded Python: https://dl.espressif.com/dl/esp-idf/

The dependency on Visual Studio C++ is caused by Python 3.9 and missing Python Wheels. Both problems are solved in installer 2.8 which deployes Embedded Python 3.8 and the installer downloads wheels from https://dl.espressif.com/pypi/.

georgik, 2021-06-08:
Thank you @balix-microej for information about documentation. We updated also the link on server so the installer 2.8 is downloaded even when somebody is accessing older documentation. 
Installer 2.8 supports v4.2.1 release.

georgik, 2021-06-08:
@balix-microej Thank you for the confirmation. Enjoy ESP-IDF.

closed 2021-06-21 by dobairoland