espressif/esp-idf #7117: Enc28j60 hangs. PHSTAT2 failed

ELERKSOSY, 2021-06-08:
I have been trying to use Enc28j60 module with esp32 I used the example in the git hub repo

https://github.com/espressif/esp-idf/tree/master/examples/ethernet/enc28j60

It works fine for 2 or 3 hours then it hangs and IP not reply (Requested timeout).

I have tried unplugged the ethernet cable or restarting the connection from my pc but nothing works the only way get it to work is resetting whole system (esp reset button) or unplug the power from the board.

In 5 minutes before the connection goes down this errors starts to popup in the logs,  before that the socket accepts the connection for 100,000 times (there is a desktop app used for testing).

[image]

[image]

Does anyone knows the reason or had similar problem?

P.s any other are function are working fine even when the ethernet hangs.

kostaond, 2021-06-14:
What IDF version do you use? Were you able to reproduce the issue multiple times?

gendy-98, 2021-06-14:
@kostaond 
Idf version 4.3
Yes, we have been debugging this issue for a while now

kostaond, 2021-06-14:
Could you please share the complete log?

kostaond, 2021-06-14:
What is your ENC28J60 silicon revision? There is errata to SPI interface so let's start with excluding that. https://ww1.microchip.com/downloads/en/DeviceDoc/80349c.pdf

gendy-98, 2021-06-15:
[session3_6_21.log](https://github.com/espressif/esp-idf/files/6654961/session3_6_21.log)

you will find some changes in the logs like  "line 468 ret = 259" ,I added this line to know which line throws the error 468 is the line number in the code and 259 is the return value

kostaond, 2021-06-15:
Now I'm little bit confused because it seems there is more than one session. 
- I noticed that in one session the Link status started oscillating Up and Down. Is this another issue or did you plug and unplug the Ethernet cable?
- When "read PHSTAT2 failed" fails started occurring, it ended up with stack overflow. Is this something new? You previously mentioned that other function are working fine even when the Ethernet hangs.
- There are some restarts which are not preceded by any logged event. What happened there?

kostaond, 2021-06-15:
Anyway, it seems the issues occur pretty much random. Could you please check the ENC28J60 silicon revision?

gendy-98, 2021-06-15:
- yes, I was plugging and unplugging the ethernet cable
- the stack overflow was one time thing l checked other log files 
- the revision id is 6 "B7"

kostaond, 2021-06-15:
Could you please try to configure your PC Ethernet interface to 10 Mbps Full duplex mode (disable autonegotiation)? Please ensure that the example output shows "enc28j60 mac: working in full duplex".

gendy-98, 2021-06-15:
[image]
[image]
it still connects as half duplix

kostaond, 2021-06-16:
My bad, it seems the link status might be updated silently. Unfortunately, I don't have this module at my desk yet to test it. Therefore let's give it to the last try. After that we will have to wait once I receive the board. Try to get Duplex mode by calling esp_eth_ioctl with ETH_CMD_G_DUPLEX_MODE command after the link is reported to be up.

Also note that the issue with readying PHY registers (the "phy is busy" in our case) should be fixed by updating SPI driver setting, see https://github.com/espressif/esp-idf/issues/7156, credits to @MTsimon.

gendy-98, 2021-06-18:
I updated the code to resolve the "phy is busy" issue but it wasn't successful, I gave it some runs and changed the value ".cs_ena_posttrans = 5" to 10 and gave it a try but nothing happened  

Tring to get Duplex mode by calling esp_eth_ioctl  didn't work
[image]

So I gave ETH_CMD_G_SPEED a try but the speed was zero
[image]

MTsimon, 2021-06-18:
The ".cs_ena_posttrans" depends on the SPI clock frequency. What SPI clock frequency are you using?

gendy-98, 2021-06-18:
5Mhz

MTsimon, 2021-06-18:
> 5Mhz

Can you try this configuration: 
`.clock_speed_hz = SPI_MASTER_FREQ_8M,
.cs_ena_posttrans = 10,
`

kostaond, 2021-06-18:
> Tring to get Duplex mode by calling esp_eth_ioctl didn't work
> [image]

Get Duplex option was added to IDF in January, see https://github.com/espressif/esp-idf/commits/master/components/esp_eth/include/esp_eth_com.h. You indicated that you use IDF 4.3 so it should be present. Please check IDF version you use.
The reason why I want to test your setup with Full Duplex is that there is errata to your chip revision "When transmitting in Half-Duplex mode with some link partners, the PHY will sometimes incorrectly interpret a received link pulse as a collision event. ... ...In some cases, the abort will fail to reset the transmit state machine"

gendy-98, 2021-06-18:
@kostaond 
My bad the idf version is 4.2 I will download 4.3 and redo the tests

gendy-98, 2021-06-18:
@MTsimon 
I was running a test with 8Mhz and until now it worked without a problem

gendy-98, 2021-06-19:
@kostaond 
After downloading version 4.3 the duplex still half

gendy-98, 2021-06-20:
> The reason why I want to test your setup with Full Duplex is that there is errata to your chip revision "When transmitting in Half-Duplex mode with some link partners, the PHY will sometimes [...]
@kostaond 
when you said that there is a problem with my chip revision and Half Duplex mode, I tried to change the connection from my PC but it ended up connecting in Half Duplex mode so I changed the code in _esp_eth_phy_enc28j60.c_  and _esp_eth_mac_enc28j60.c_ to force the connection to be Full Duplex  and double checked that the connection is Full Duplex from the command prompt, but unfortunately after 5 hours of running  the ethernet just hanged again. I redid the test multiple times but with no luck

Do you have any suggestion for a way to debug the code ? 
 
[image]

[image]

kostaond, 2021-06-21:
@gendy-98 did you perform multiple trials? Does the ENC hangs after 5 hours now instead of 2-3?

I quickly check on the internet and it seems that multiple users report similar problems on different platforms. We need to wait until we receive the boards with the chip. Then we try to reproduce the issue but I cannot promise anything... I am not sure if we will be even able to reproduce it. We will need more details about your test scenario. Could you please also share which exact board do you use?

Last thing which crossed my mind, could you please try your test in different network (ENC connected to different PC/switch/router)?

gendy-98, 2021-06-21:
- yes I performed multiple tries with slightly different hanging time from 3 hr to 5 hr and one time it lasted for 12 hr with no error in the logs just hangs

- I use a custom designed board that contains the esp32, enc28j60 and some other components I thought that my design was the problem so I uploaded the code to esp32 divKit v1 and used a store bought  enc module and connected them using som jumper wires and it gave me the same output 

-  Can I offer sending you esp and enc for testing instead of waiting?

AndriiFilippov, 2021-06-28:
Hi @gendy-98 ! 
We already have Enc28j60 module on site and we were able to reproduce the issue. 
We will provide you updates soon!

Alvin1Zhang, 2021-07-22:
Thanks for reporting, we have a fix which is now under internal reviewing, the issue will close automatically once it is available on GitHub. Thanks.

kostaond, 2021-07-26:
Just to summarize the issue. The driver was improved, several ENC28J60 erratas was addressed and it is going to be available soon in master branch.

During the driver update, we made important observation which needs to be taken into account when using this chip. **Proper input power source needs to be used** since ENC28J60 is quite power consuming device (it consumes more than 200 mA in peaks when transmitting). If improper power source is used, input voltage may drop and ENC28J60 may either provide nonsense response to host controller via SPI (fail to read registers properly) or it may enter to some strange state in the worst case. There are several options how to resolve it:
  * Power ESP32 board from `USB 3.0`, if board is used as source of power to ENC board.
  * Power ESP32 board from external 5V power supply with current limit at least 1 A, if board is used as source of power to ENC board.
  * Power ENC28J60 from external 3.3V power supply with common GND to ESP32 board. Note that there might be some ENC28J60 boards with integrated voltage regulator on market and so powered by 5 V. Please consult documentation of your board for details.

  If a ESP32 board is used as source of power to ENC board, ensure that that particular board is assembled with voltage regulator capable to deliver current up to 1 A. This is a case of ESP32 DevKitC or ESP-WROVER-KIT, for example. Such setup was tested and works as expected. Other boards may use different voltage regulators and may perform differently.

closed 2021-07-29 by espressif-bot