espressif/esp-idf #7119: ESP_PROV: ESP BLE Prov Android App is not working on wifi_prov_mgr example

carlessole, 2021-06-08:
Hi,
I'm running the stable version of ESP-IDF, [v4.2.1](https://github.com/espressif/esp-idf/tree/v4.2.1).
I'm trying the [wifi_prov_mgr example](https://github.com/espressif/esp-idf/tree/v4.2.1/examples/provisioning/wifi_prov_mgr) and testing it against the [ESP BLE Prov Android App](https://play.google.com/store/apps/details?id=com.espressif.provble&hl=es&gl=US)

Although the APP finds the device:

[image]

Later it is unable to connect with the device:

[image]

No ESP32 console LOGS appears neither :(.

If using the legacy example [ble_prov](https://github.com/espressif/esp-idf/tree/v4.2.1/examples/provisioning/legacy/ble_prov) everything is OK.

dhrishi, 2021-06-08:
Hi @carlessole Are you using BLE as the provisioning transport (default used in the example is BLE)? If yes, there is a known issue on the ESP32 side with BLE (NimBLE host) in release v4.2.1. 

You can use the latest `release/v4.2` branch (also update submodules) to test it out as the fix is added on that branch.

The fix will be a part of v4.2.2 release which should be out soon.

carlessole, 2021-06-08:
Thanks @dhrishi,
Yes, the error occurs when using BLE. Using SoftAP there is no problem.
I'm gonna try with 4.2 to check it.

And is it possible to configure the example on this version to use Bluedroid instead of NimBLE? (I tried it but it is still not working)
[image]

carlessole, 2021-06-08:
Hi @dhrishi ,
I have just tried with the "release/v4.2" and same error is being triggered.

[image]

`(17:22:42.254) <ESC>[0;32mI (28) boot: ESP-IDF v4.2.1-116-gfa734e6af 2nd stage bootloader<ESC>[0m
`

Is it "normal"?

Thanks

dhrishi, 2021-06-08:
>And is it possible to configure the example on this version to use Bluedroid instead of NimBLE? (I tried it but it is still not working)

This is quite strange. I will check this internally and update

dhrishi, 2021-06-08:
>(17:22:42.254) <ESC>[0;32mI (28) boot: ESP-IDF v4.2.1-116-gfa734e6af 2nd stage bootloader

This looks fine. Did you also update the submodules using `git submodule update --init --recursive`

carlessole, 2021-06-08:
> 
> 
> > (17:22:42.254) [0;32mI (28) boot: ESP-IDF v4.2.1-116-gfa734e6af 2nd stage bootloader
> 
> This looks fine. Did you also update the submodules using `git submodule update --init --recursive`

I'm quite sure, but just in case, tomorrow I will try again.

Thanks

dhrishi, 2021-06-11:
Hello @carlessole  I just checked -
ESP32 `examples/provisioning/wifi_prov_mgr` with ESP BLE Prov Android App (v2.0.8 - e8734ca)

1. IDF Release v4.2.1 - Using Bluedroid Host : Successful
`I (30) boot: ESP-IDF v4.2.1 2nd stage bootloader` 

2. IDF release/v4.2 branch - Using default (NimBLE Host) : Successful
I (27) boot: ESP-IDF v4.2.1-116-gfa734e6af 2nd stage bootloader

3. IDF release/v4.2 branch - Using Bluedroid Host : Successful
I (27) boot: ESP-IDF v4.2.1-116-gfa734e6af 2nd stage bootloader

If you are still seeing this issue, please share the sdkconfig and debug enabled logs from the ESP32 side.

dhrishi, 2021-06-18:
@carlessole ESP-IDF v4.2.2 is released that has the fix for NimBLE and it is functional as expected. You can find it here: https://github.com/espressif/esp-idf/releases/tag/v4.2.2

Please give it a try and let us know in case of any issues

carlessole, 2021-06-21:
Hello!
I have test it with current stable version ([4.3](https://github.com/espressif/esp-idf/releases/tag/v4.3)) and now it works when using the ESP32-WROOM-32E. 
But trying it with ESP32-WROVER-E on (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-wrover-kit.html) I am still unable to connect -> the same error pops up.

carlessole, 2021-06-21:
> 
> 
> Hello!
> I have tested it with current stable version ([4.3](https://github.com/espressif/esp-idf/releases/tag/v4.3)) and now it works when using the ESP32-WROOM-32E.
> But trying it with ESP32-WROVER-E on (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-wrover-kit.html) I am still unable to connect -> the same error pops up.

I also have tested it with [4.2.2](https://github.com/espressif/esp-idf/releases/tag/v4.2.2). The same is happening:

ESP-WROOM-32 -> OK
ESP-WROVER -> NOK

dhrishi, 2021-06-24:
@carlessole Can you please share the generated sdkconfig file with v4.2.2? As mentioned above, you are using the wifi_prov_mgr example with no changes, right?

closed 2021-07-28 by Alvin1Zhang