espressif/esp-idf #7122: ledc_example_main.c doesn't compile

grantcroft, 2021-06-09:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version   Arduino 1.8.15
- Operating System:     [Windows]
- (Windows only) environment type: [Arduino].
- Power Supply:         [USB]

## Problem Description
ledc_example_main.c doesn't compile. Throws ERROR 1 with the code as-is from the git. 
I googled the error and found: https://www.mikrocontroller.net/topic/496993#6284479
adding the "#define LEDC_AUTO_CLK 0" near the top made ERROR 2.
Both ERROR 1 and 2 are referencing line 74. 

Commenting that out makes the code progress to ERROR 3 which happens on lines: 106, 115, 125, and 134.

### Expected Behavior
Compiling 

### Actual Behavior
Error 1: " 'LEDC_AUTO_CLK' was not declared in this scope" 

ERROR 2: "'ledc_timer_config_t' has no non-static data member named 'ledc_clk_cfg_t'"

ERROR 3:  "expected primary-expression before '.' token" 

### Steps to reproduce

1.  Download and compile code for ERROR 1
2. add "#define LEDC_AUTO_CLK 0" and compile for ERROR 2
3. comment out line 74 to get ERROR 3

negativekelvin, 2021-06-09:
>IDF version Arduino 1.8.15

No that's the IDE version.  You probably need to use an older version of the example code https://github.com/espressif/esp-idf/blob/release/v3.3/examples/peripherals/ledc/main/ledc_example_main.c

You may have other issues building esp-idf examples against precompiled Arduino libraries.

0xjakob, 2022-01-05:
@grantcroft Does the issue still persist?

0xjakob, 2022-01-24:
@grantcroft I'm closing this issue since there hasn't been any feedback. If you come across this issue again, please let us know. Feel free to re-open this issue then.

closed 2022-01-24 by 0xjakob