espressif/esp-idf #7123: Wifi station/access point example

Benedito821, 2021-06-09:
Hello! Is there an example that shows how to configure **wifi in station/access point mode** **with a simple web server**? I runned the **station** example and after this I can'topen the web server by the IP adress.

MaxwellAlan, 2021-06-09:
hi @Benedito821, You can try example/protocols/http_server/simple example and set in menuconfig using WiFi connection.

Benedito821, 2021-06-09:
thank you for the tip
I'll check that @MaxwellAlan

Benedito821, 2021-06-09:
could I use also this example to control my board with some commands(e.g, leds,peripherals,...)?

MaxwellAlan, 2021-06-15:
hi @Benedito821 yes, you can, but you need to add some more code for your commands, may be you can also refer to other esp-idf/examples.

Alvin1Zhang, 2021-07-14:
Thanks for reporting and sharing updates, feel free to reopen.

closed 2021-07-14 by Alvin1Zhang