espressif/esp-idf #7124: mDNS sends reply to multicast instead of unicast if request source port is 5353 (breaks avahi-resolve)

DaStoned, 2021-06-09:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit: ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version: v4.2.1
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply:  USB

## Problem Description

The mDNS resolution of device name to IP has stopped working in v4.2.1 (don't remember in which version it was OK, maybe v4.0?). Specifically, the ESP IDF mDNS service responds to queries from `avahi-resolve -n -4 UBIK-71F164.local` with a packet destined to the multicast group 224.0.0.251 which is not received by `avahi-resolve`. As a result avahi-resolve doesn't resolve the name.

As contrast, running `dig +short -p 5353 @224.0.0.251 UBIK-71F164.local` does resolve the name perfectly. Details and packet capture below.

NB! The problem raises only 2 minutes after ESP32 boots, probably because on boot mDNS broadcasts the relevant info which gets cached by my machine.

The network setup is fairly common, a Technicolor domestic router/AP with ESP32 connected via WiFi and my computer via Ethernet.

### Expected Behavior

Running `avahi-resolve` after 2 minutes after ESP32 boot resolves the name to IP:
```
tarmo@lumi:~$ avahi-resolve -n -4 UBIK-71F164.local -v
Server version: avahi 0.8; Host name: lumi.local
UBIK-71F164.local       192.168.224.120
```
### Actual Behavior

`avahi-resolve` fails to resolve the name:
```
tarmo@lumi:~$ avahi-resolve -n -4 UBIK-71F164.local -v
Server version: avahi 0.8; Host name: lumi.local
Failed to resolve host name 'UBIK-71F164.local': Timeout reached
```

At the same time `dig` correctly resolves the name:
```
tarmo@lumi:~$ dig +short -p 5353 @224.0.0.251 UBIK-71F164.local
192.168.224.120
```

### Steps to reproduce

1. Start ESP32, init mDNS and set name per official instructions
2. Wait 2+ minutes
3. Run `avahi-resolve -n -4 <name>.local` 
4. Resolution times out

### Code to reproduce this issue

```cpp
bool NetworkHub::runMdns() {
    esp_err_t result = mdns_init();
    if (ESP_OK != result) {
        err("Fail mDNS init: %s (%d)", esp_err_to_name(result), result);
        return false;
    }
    result = mdns_hostname_set(mTrait.id());
    if (ESP_OK != result) {
        err("Fail mDNS set hostname to [%s]: %s (%d)", mTrait.id(), esp_err_to_name(result), result);
        return false;
    }
    char buf[50];
    const int printed = snprintf(buf, sizeof(buf), "Solar panel inverter/converter %s", mTrait.id());
    assert(printed > 0 && printed < sizeof(buf));
    result = mdns_instance_name_set(buf);
    if (ESP_OK != result) {
        err("Fail mDNS set instance name to [%s]: %s (%d)", buf, esp_err_to_name(result), result);
        return false;
    }
    result = mdns_service_add(nullptr, "_https", "_tcp", 443, NULL, 0);
    if (ESP_OK != result) {
        err("Fail mDNS add service HTTPS: %s (%d)", esp_err_to_name(result), result);
        return false;
    }
    return true;
}
```
## Debug Logs
There is nothing at all in the ESP32 logs.

## Initial analysis

A screenshot from Wireshark shows two different mDNS queries (request-response pairs). 
[image]

First query-response pair is from running `dig +short -p 5353 @224.0.0.251 UBIK-71F164.local` which successfully resolves the name. You can see that the request's source port is a random ephemeral port 44238. The response is sent as unicast to source IP of resolve query (192.168.224.100). All good.

Second query-response pair is from running `avahi-resolve -n -4 UBIK-71F164.local -v` which fails. The request's source port is 5353. The response is sent to multicast group 224.0.0.53 instead of source IP of resolve query. Not good.

The packet capture for those 4 packets is here: 
[packetcapture.zip](https://github.com/espressif/esp-idf/files/6623370/packetcapture.zip)

negativekelvin, 2021-06-09:
> The cache-flush bit is only set in records in the Resource Record
>    Sections of Multicast DNS responses sent to UDP port 5353.
> 
>    The cache-flush bit MUST NOT be set in any resource records in a
>    response message sent in legacy unicast responses to UDP ports other
>    than 5353.

Note that neither request sets the unicast response bit.  I don't know if the fact that a unicast response is forced when an alternate port is used is per spec or not.

Alvin1Zhang, 2021-06-10:
Thanks for sharing the detailed report, we will look into.

david-cermak, 2021-06-10:
Thanks again for this report and sharing the analysis, this is helpful :+1: 

The actual difference between the `dig` query and the one sent by `avahi` is not only the source port, but mainly the fact that `dig` (and some other libraries, lwip, too) sends so called *One shot queries* defined in https://datatracker.ietf.org/doc/html/rfc6762#section-5.1, whereas `avahi` is a fully compliant mDNS querier.

The mDNS library doesn't (correctly) support both scenarios, trying to comply with full featured mDNS responder and at the same time responds to lwip mDNS queries introduced `MDNS_REPEAT_QUERY_IN_RESPONSE` and `CONFIG_MDNS_STRICT_MODE` configuration options.
Enabling the strict mode in the project configuration menu would make it respond to `avahi`'s queries (after those 2 minute TTLs), but stop responding to `dig`'s (note, that avahi deviates from the spec here, a little, as well: https://datatracker.ietf.org/doc/html/rfc6762#page-14)

Here's a quick fix, which should work for both queriers:
[mdns-Support-for-One-Shot-mDNS-queries.patch.txt](https://github.com/espressif/esp-idf/files/6631752/mdns-Support-for-One-Shot-mDNS-queries.patch.txt)

> I don't know if the fact that a unicast response is forced when an alternate port is used is per spec or not.

https://datatracker.ietf.org/doc/html/rfc6762#section-6.7

DaStoned, 2021-06-10:
Thank you very much for looking into it. My conclusions probably threw a bunch of red herrings on your path, sorry for that.

I confirm that the patch works. I tested on Linux with every tool I could think of (avahi-resolve, dig, ping, Chrome, Firefox). All successfully resolve the name of the patched device. At the same time the unpatched device is not resolved.

Just in case I also booted Windows and tested there (ping, Chrome, Edge) - works like a charm using whatever native thing that does mDNS on Windows. Unfortunately I don't have any Apple devices, so can't test with Mac or iOS.

Side note: the system name resolver on Linux uses avahi (most frequently, systemd also has something which I haven't tested), so if that doesn't resolve my device's name it's quite unfortunate. Can't access the web interface by name, can't ping to verify connectivity. If I access the web interface by IP, the certificate no longer applies so the browser is screaming murder and hackers...

Alvin1Zhang, 2021-06-28:
Thanks for reporting, the fix on master branch is available https://github.com/espressif/esp-idf/commit/f601cb0fd73de13f5f23293d99f3e8fdb9c30d6c. We are back porting the fix to release/4.3 and release/4.2. Thanks.

DaStoned, 2021-06-30:
Great news! Can't test/confirm with master, as it's frankly a pain in the butt to build. Waiting for the backports.

Alvin1Zhang, 2021-07-14:
@DaStoned Thanks for being patient while waiting for the fix, fix on release/4.3 has been available https://github.com/espressif/esp-idf/commit/3a588d7d1960b0749727360036f41b48dc7afb80, fix on release/4.2 is now being synchronized onto GitHub, will update once release/4.2 fix is available. Thanks.

Alvin1Zhang, 2021-07-19:
Thanks for being patient while waiting for the fix, fix on release/4.2 has been available https://github.com/espressif/esp-idf/commit/93921e042cd8e1c21ad4b39f3f6e6a1f05a69ad2, feel free to reopen if the issue still happens. Thanks.

closed 2021-07-19 by Alvin1Zhang