espressif/esp-idf #7126: Cannot use APLL clock with 48000 Hz, 32-bit dual channel I2S in SDK v4.2.x on PICO-D4

HanVertegaal, 2021-06-09:
I am porting a program from SDK 3.3.5 to 4.2. Note that the program works fine with SDK 3.3.5.

## Environment

- Development Kit:      none
- Module or chip used:  ESP32-PICO-D4 (revision 1)
- IDF version: v4.2
- Build System:         [Make]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     WSL
- Using an IDE?: No
- Power Supply: external 5V

## Problem Description

When initialising the I2S controller(s) I need to use the APPL feature to get an accurate clock. Without APLL the code runs, but the signal received from the ADC/I2S contains spikes. On SDK 3.3.5 selecting the APLL worked fine, and gave the correct clock rate. However, when I try this on SDK 4.2 the log shows a lot of garbage and eventually the program crashes - during the I2S initialisation.

Remotely related to [IDFGH-432](https://github.com/espressif/esp-idf/issues/2634), which shows why use of the APLL clock is a requirement.

### Expected Behavior

I expected the I2S code to work the same between SDK 3.3.5 and SDK 4.2. Certainly I didn't expect the code to crash.

### Actual Behavior

On SDK 4.2 the code crashes in the I2S initialisation (see logs). Note that when using SDK 4.2.1 the log output becomes illegible at some point.

### Steps to reproduce

Inititialise the I2S controllers as in the code below. (The InitI2S function is called from a separate task running on Core 1, because reading the from I2S controller also runs on Core 1.)

### Code to reproduce this issue

```cpp

#define SAMPLE_RATE         (48000)
#define I2S_VADC            I2S_NUM_0
#define I2S_IADC            I2S_NUM_1
#define BITS_PER_SAMPLE     I2S_BITS_PER_SAMPLE_32BIT
#define FRAMES_PER_BUFFER   (480)
#define DMA_BUFFERS         (8)
#define CHANNELS            (2)

void AEI2S::InitI2S(void* args)
{
    ESP_LOGD(LOG_TAG, ">> InitI2S");
    i2s_config_t i2s_config;
    i2s_config.mode = (i2s_mode_t)(I2S_MODE_SLAVE | I2S_MODE_RX);
    i2s_config.sample_rate = SAMPLE_RATE;
    i2s_config.bits_per_sample = BITS_PER_SAMPLE;
    i2s_config.channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT; // 2-channels
    i2s_config.communication_format = I2S_COMM_FORMAT_STAND_I2S;
    i2s_config.dma_buf_count = DMA_BUFFERS;
    i2s_config.dma_buf_len = FRAMES_PER_BUFFER;
    i2s_config.intr_alloc_flags = ESP_INTR_FLAG_LEVEL1;
    // i2s_config.tx_desc_auto_clear = true;
    i2s_config.use_apll = true;
    i2s_config.fixed_mclk = 0;

    ESP_ERROR_CHECK(i2s_driver_install(I2S_VADC, &i2s_config, 0, NULL));
    ESP_ERROR_CHECK(i2s_driver_install(I2S_IADC, &i2s_config, 0, NULL));

    i2s_pin_config_t pin_config;
    pin_config.bck_io_num = 32;
    pin_config.ws_io_num = 33;
    pin_config.data_out_num = I2S_PIN_NO_CHANGE;
    pin_config.data_in_num = 34;
    ESP_ERROR_CHECK(i2s_set_pin(I2S_VADC, &pin_config));
    pin_config.data_in_num = 35;
    ESP_ERROR_CHECK(i2s_set_pin(I2S_IADC, &pin_config));
[... 116 lines cut ...]
I (138) cpu_start: ELF file SHA256:  8e4c382ab4812e1b...
I (144) cpu_start: ESP-IDF:          v4.2-dirty
I (149) cpu_start: Starting app cpu, entry point is 0x40081290

## Other items if possible

On SDK 4.2.1 I see this, where most of the log is unreadable:

D (5679) efuse: coding scheme 0
D (5682) efuse: coding scheme 0;31mE
D (55) t ek_we: coding atchme  got
D (568d. ehe following tchemedi0 rese5 the watsedog oding e:
1mD (5685) easke: codin  scheme 0U[0m
Ese: c5)itask_wdm: Ta[ks
 schemem085) t5s85) e: CPU c:ding schem
ED (6855)tefk_e: codin 1s Init 0Sm
 D (␌┤85␊┼␊ ␌⎺⎼␊) ␉▒␍␋┼± ⎽␌┼├ C└

▒D ├⎼5685) 400⎽␊: ␌⎺␍␋3FFB␌A␤␊└␊ 400822
D:0│6FF)0␊E0 0: ␌⎺␍7AB9:0│␊└␊ 00│400D875) 0│┤FFB45E0 ┼│ ⎽␌␤␊└␊9:0│3FFBD (5 8│40␊°┤D8: ␌│␍␋┼± ⎽20 
0│ 0
4DA0 6│5) ␊°788: ␌3␍␋┼± ⎽1:D 3FFB56)  °┤⎽0057902:0│⎽␌␤␊└␊0000│4005D (5:05)F␊B47␊: ␌⎺4␋┼± ⎽␌1:0│347C
D (5608) ␊°:0␊3FFB4␋┼± ⎽│400␊ 00[:0
F (4845) │°┤0D:E52:0│± ⎽␌␤␊80 0│40└
␤␊└␊ 31└E (568556├5) ␊┬␍├␊: ␌⎺┼├ CPU␌1 ␉▒ ┐├[0└

HanVertegaal, 2021-06-23:
This problem also occurs on an ESP32 WROOM module, so it is not specific to the PICO.
When the apll mode is enabled in the i2s_config structure, the function **i2s_apll_get_fi2s()** in esp-idf-v4.3/components/driver/i2s.c queries the chip revision, and changes the calculation if the revision = 0. As both my ESP32 and the PICO have a revision higher than 0, I have removed the call to **esp_efuse_get_chip_ver()**, and that fixes the problem.
Obviously, this is not a proper solution, as the call to **esp_efuse_get_chip_ver()** should work.

L-KAYA, 2022-09-23:
It has been solved in v4.4 but no plan to backport it to v4.2 because it's somehow a risky refactor. You can upgrade to v4.4 to get the bugfix or even v5.0 for the new I2S driver.

closed 2022-09-23 by HanVertegaal