espressif/esp-idf #7127: Can't connect to ESP's AP Mode while it is powered via the 5V input

GeorgeFlorian, 2021-06-09:
## Environment

- Development Kit:      ESP32 Wemos
- Module or chip used:  ESP-WEOOM-32
- IDF version: v4.2.1
- Build System:         [Make|CMake|idf.py]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:    Linux
- Using an IDE?: VS Code 1.56
- Power Supply:         External 5V

## Problem Description

I am trying to implement this project: https://github.com/DroneBridge/ESP32 and I can't connect to the ESP's AP mode.
I can see the AP (DroneBridge ESP32) in the WiFi list but it doesn't connect to it. I've gotten two different error messages:
```
Couldn't authenticate connection
Couldn't get IP address
```
The project basically puts the ESP32 in AP Mode and makes it a bridge between the user and the drone.
The ESP is powered by a drones battery at 5V.
**If I power it via the USB,it works.**
I have measured and the power input is 5V. I have also tried the other 5V input on the ESP board but to no avail.

[image]

Ignore the RX and TX wires as they are not connected to anything.

GeorgeFlorian, 2021-06-09:
> It is very bad idea in any electronic device to connect 5V to pin labeled 3V.
> That is also answer to your issue.

I don't know if that label is 3 or 5 because it's slightly erased.
But the point is that I have tried the 5V labeled input on the left and I still have the same issues.

chegewara, 2021-06-09:
It is 100% 3V labeled.

The thing is that esp32 can survive 5V connected to 3/3.3V, at least for some time, but you can see "magic smoke". Even if it will survive and it seems to work the wifi wont work (i know it from experience). Radio is just not tolerant.

My advice is to use multimeter and check voltage on pins labeled 3 and 3.3V when you connect board to battery.

GeorgeFlorian, 2021-06-09:
> It is 100% 3V labeled.
>
> The thing is that esp32 can survive 5V connected to 3/3.3V, at least for some time, but you can see "magic smoke". Even if it will survive and it seems to work the wifi wont [...]

But why does it work when I connect it via USB ?

chegewara, 2021-06-09:
Because you have LDO on board and stable 5V from USB.
Start with reading board schematics and using multimeter.

If code works with USB that means it is not issue with code, no matter what power source you are using, as long as it is stable should be sufficient.
This issue should be moved to forum anyway.

GeorgeFlorian, 2021-06-10:
> Because you have LDO on board and stable 5V from USB.
> Start with reading board schematics and using multimeter.
>
> If code works with USB that means it is not issue with code, no matter what power [...]

I have switched to a new ESP32: the DevKit V4 Wroover-B . The red LED is on, but there is no AP listed. This one also works only when connected to USB.
[image]

I've got a multi-meter and it measures 5.05V on the 5V pin and 3.3V on the 3V3 pin.

I have moved this issue on the ESP32 forum: https://www.esp32.com/viewtopic.php?f=12&t=21386

closed 2021-06-10 by GeorgeFlorian