espressif/esp-idf #7128: ESP32C3 SDSPI

rsn50, 2021-06-09:
I am trying to interface ESP32C3 module with the MicroSD Card module (operating at 3.3V). 
The code which I am executing is from the Master Branch, sdspi. (ESP32 Example code for 'SD Card Example (SDSPI)).

Initially it gave me an error:
```
I (231) cpu_start: ESP-IDF:          v4.4-dev-1404-gc13afea63-dirty
I (239) heap_init: Initializing. RAM available for dynamic allocation:
I (246) heap_init: At 3FC8D810 len 000327F0 (201 KiB): DRAM
I (252) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (259) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM
I (265) spi_flash: detected chip: generic
I (270) spi_flash: flash io: dio
I (274) sleep: Configure to isolate all GPIO pins in sleep state
I (280) sleep: Enable automatic switching of GPIO sleep configuration
I (288) cpu_start: Starting scheduler.
I (292) example: Initializing SD card
I (292) example: Using SPI peripheral
E (292) spi: spi_bus_initialize(634): invalid dma channel, chip only support spi dma channel auto-alloc
E (302) example: Failed to initialize bus.
```
After that I changed the define statement in the code to:
```
#define SPI_DMA_CHAN SPI_DMA_CH_AUTO 
```
But I am getting the following error:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x1658
load:0x403ce000,len:0x8c0
load:0x403d0000,len:0x2928
SHA-256 comparison failed:
Calculated: 7a664bd4e7efa15fbbd256bcd27f435e0ab3e9cb5e691b1cf65fdd4ce1555e33
Expected: ff1d8365de40ca83c5da51eaa736d3d543a613d4efac975fa518a8d9d9cb1e52
Attempting to boot anyway...
entry 0x403ce000
I (48) boot: ESP-IDF v4.4-dev-1404-gc13afea63-dirty 2nd stage bootloader
I (49) boot: compile time 19:58:42
I (49) boot: chip revision: 3
I (52) boot.esp32c3: SPI Speed      : 80MHz
I (57) boot.esp32c3: SPI Mode       : DIO
I (62) boot.esp32c3: SPI Flash Size : 4MB
I (67) boot: Enabling RNG early entropy source...
I (72) boot: Partition Table:
I (76) boot: ## Label            Usage          Type ST Offset   Length
I (83) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (90) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (98) boot:  2 factory          factory app      00 00 00010000 00100000
I (105) boot: End of partition table
I (109) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=09e90h ( 40592) map
I (124) esp_image: segment 1: paddr=00019eb8 vaddr=3fc8b400 size=01550h (  5456) load
I (127) esp_image: segment 2: paddr=0001b410 vaddr=40380000 size=04c08h ( 19464) load
I (138) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=2a398h (172952) map
I (169) esp_image: segment 4: paddr=0004a3c0 vaddr=40384c08 size=066fch ( 26364) load
I (174) esp_image: segment 5: paddr=00050ac4 vaddr=50000000 size=00010h (    16) load
I (178) boot: Loaded app from partition at offset 0x10000
I (181) boot: Disabling RNG early entropy source...
I (198) cpu_start: Pro cpu up.
I (207) cpu_start: Pro cpu start user code
I (207) cpu_start: cpu freq: 160000000
I (207) cpu_start: Application information:
I (210) cpu_start: Project name:     sd_card
I (215) cpu_start: App version:      1
I (219) cpu_start: Compile time:     Jun  9 2021 19:57:58
I (226) cpu_start: ELF file SHA256:  d96cd9031dd7ae4a...
I (231) cpu_start: ESP-IDF:          v4.4-dev-1404-gc13afea63-dirty
I (239) heap_init: Initializing. RAM available for dynamic allocation:
I (246) heap_init: At 3FC8D810 len 000327F0 (201 KiB): DRAM
I (252) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (259) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM
I (265) spi_flash: detected chip: generic
I (270) spi_flash: flash io: dio
I (274) sleep: Configure to isolate all GPIO pins in sleep state
I (280) sleep: Enable automatic switching of GPIO sleep configuration
I (288) cpu_start: Starting scheduler.
I (292) example: Initializing SD card
I (292) example: Using SPI peripheral
I (292) gpio: GPIO[19]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (342) sdspi_transaction: cmd=5, R1 response: command not supported
E (352) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (352) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
I (352) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (362) example: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place.

```
Also I have connected pullup resistors (10K) on SD card pins.  
Also I tried to change the frequency by adding the line:
```
host.max_freq_khz = SDMMC_FREQ_PROBING;
```
But the code is giving the same error:
```
E (362) example: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place.
```
I request you to please guide.
Thank you.

chegewara, 2021-06-09:
https://esp32.com/viewtopic.php?f=13&p=75316#p75316

rsn50, 2021-06-10:
There was some connection/wiring problem in the circuit. Now it is working fine.
Thank you.

closed 2021-06-10 by rsn50