espressif/esp-idf #7129: Default argument to esp_prov.get_security causes a TypeError

alexhenrie, 2021-06-09:
The `get_security` function in [esp_prov.py](https://github.com/espressif/esp-idf/blob/master/tools/esp_prov/esp_prov.py) is defined as follows:

```
def get_security(secver, pop=None, verbose=False):
    if secver == 1:
        return security.Security1(pop, verbose)
    elif secver == 0:
        return security.Security0(verbose)
    return None
```

However, if attempting to use the function from an external script, the default `pop` argument results in a TypeError:

```
import sys
from os import environ
sys.path.append(environ['IDF_PATH'] + '/tools/esp_prov')
from esp_prov import *

security = get_security(secver=1)
```

```
TypeError: cannot convert 'NoneType' object to bytes
```

A workaround is to explicitly specify `get_security(secver=1, pop='')` when not using proof-of-possession.

Alvin1Zhang, 2021-07-16:
Thanks for reporting and patience, we have a fix which is now under internal reviewing, the issue will close automatically once the fix is available on GitHub. Thanks.

mahavirj, 2021-07-16:
Fixed with https://github.com/espressif/esp-idf/commit/5ed53c272b2fac5c052d2d097b2fd6e9957ffd01

closed 2021-07-16 by mahavirj