espressif/esp-idf #7133: [performance] pick a patch to speed up mbedtls ec multiplication

Kxuan, 2021-06-10:
**Is your feature request related to a problem? Please describe.**
The ec point multiplication operation is slow with mbedTLS.

[This patch](https://github.com/ARMmbed/mbedtls/pull/4315/commits/ba9cb76e9f18be7fc763b86295c9f445d1c2d280) changes the behavior of option `MBEDTLS_ECP_FIXED_POINT_OPTIM`. It speeds up ecc multiplication, and reduces RAM usage.

The mbedTLS use time to calculate a comb table to speed up ec point multiplication.
This patch makes mbedTLS store a pre-computed comb table on ROM. So mbedTLS does not need take time to calculate the table.
It expects speed up about 40% time on TLS handshake (using ECC certificate) and reduce peak memory usage.

The patch will be used in mbedtls 3.0 instead of the stable release v2.x, because it changes the behavior of an existing option.

**Describe the solution you'd like**
Since the ESP-IDF major release allows break API, I think we can talk about porting this patch into next ESP-IDF major release to gain perforamnce.

The possiblility of breaking user application is low. The only user visible changes is not an official declared feature.
Before applying this patch, after doing a ec point multiplication operation, the `T_size` field on `mbedtls_ecp_group` is setted to non-zero. While after applying this patch, the `T_size` field will always keep zero.

If ESP-IDF want this patch, I can make a pull request.

mahavirj, 2021-06-11:
@Kxuan 

> The patch will be used in mbedtls 3.0 instead of the stable release v2.x, because it changes the behavior of an existing option.

Thank you for notifying on this!

>  Since the ESP-IDF major release allows break API, I think we can talk about porting this patch into next ESP-IDF major release to gain perforamnce.

Currently we are still in v4.4-dev phase, so it will be difficult to accept this change. However we have plans to integrate mbedtls-3.0 in our next major release, so that should readily bring in this optimization as well. Does that sound okay to you?

Kxuan, 2021-06-15:
Thank you @mahavirj .

I think it will be better to backport this patch into mbedtls v2.x and use the patch in the ESP-IDF v4.4 release.

It will be too long to wait the release of mbedtls v3.x. By backporting this patch into mbedtls v2.x, we could gain the performance improvement without wait that long.

The ESP-IDF v4.4 allows minor "breaking changes", so that will be an opportunity to backport this patch. Could you please reconsider about this idea?

mahavirj, 2021-06-16:
> The ESP-IDF v4.4 allows minor "breaking changes", so that will be an opportunity to backport this patch. Could you please reconsider about this idea?

It is true that we do allow "minor" breaking changes. IMO, this case is little different. I will try to explain:

- We typically rely on mbedtls-2.16.x upstream "official" release, we rebase it on periodic basis
- Currently we are using v2.16.9 release, it has "minor" patches of our own to fix few build warnings, and one change to enable hardware MPI implementation. Ref: https://github.com/espressif/mbedtls/commits/mbedtls-2.16.9-idf
- This allows us to still use qualified/tested upstream mbedtls release

In summary, it will be difficult to accept and maintain this change for us on v-2.16.x branches. Hence, my recommendation would be to wait for mbedtls-v3.x release and then go through that route.

AxelLin, 2021-06-16:
hi @mahavirj 

mbedtls-2.16.10 has been released for a while, and I think a new 2.16.x version will be available soon.
Would you still updating mbedtls-v2.16.x stable branch to esp-idf-v4.3 (and earlier versions)?

mahavirj, 2021-06-16:
@AxelLin 

>  Would you still updating mbedtls-v2.16.x stable branch to esp-idf-v4.3 (and earlier versions)?

Yes. I have added internal tracker for this.

Kxuan, 2021-06-16:
Thanks for your detailed explaination. @mahavirj

closed 2021-06-16 by Kxuan

AxelLin, 2021-07-07:
> @AxelLin
> 
> > Would you still updating mbedtls-v2.16.x stable branch to esp-idf-v4.3 (and earlier versions)?
> 
> Yes. I have added internal tracker for this.

Hi @mahavirj 
FYI, mbedtls-2.16.11 is released (with security fixes):
https://github.com/ARMmbed/mbedtls/releases/tag/v2.16.11

AxelLin, 2021-07-07:
> In summary, it will be difficult to accept and maintain this change for us on v-2.16.x branches. Hence, my recommendation would be to wait for mbedtls-v3.x release and then go through that route.

FYI: mbedTLS-v3.0.0 is released.
https://github.com/ARMmbed/mbedtls/releases/tag/v3.0.0

mahavirj, 2021-07-07:
@AxelLin 

Thanks for notifying.

>  FYI, mbedtls-2.16.11 is released (with security fixes):

This will be our immediate priority, we had internal on-going work for 2.16.10 merge but we will now move all release branches to 2.16.11

> FYI: mbedTLS-v3.0.0 is released.

We expect breaking changes here, so our target will be ESP-IDF v5.0 release.

AdityaHPatwardhan, 2021-11-25:
Hi @AxelLin @Kxuan We have recently released the `mbedtls-3.0 Integration preview`.
Please refer to the [blog post](https://www.esp32.com/viewtopic.php?f=10&t=24451&sid=1fda35c1de8104175a3dab99d7f88f61) for more details