espressif/esp-idf #7136: Obtaining the clock counter for a specific core.

ammaree, 2021-06-11:
We have a component that provided duration timing with very low overhead.
Essentially, to mark the starting time of an activity we store the core number and clock counter of the core.
When the activity is completed we read the clock time of the core and save the difference as the duration.

In ~99% of the cases the activity is started and stopped from the same MCU.
In the other ~1% the activity is stopped from a different MCU as that started from, leading to completely incorrect duration calculations.

Is there any way of reading the clock counter for MCY X from MCU Y?

igrr, 2021-06-14:
Unfortunately there isn't a low-overhead way to read the CPU cycle count from another core. Assuming that you are using the ESP32, the CPU cycle counter there a register internal to each CPU. The only way to obtain it from the other CPU would involve some form of cross-core communication, such as raising an interrupt on the other CPU and making the interrupt handler read the cycle counter and store it somewhere.

You could consider using the timer group peripheral counter as an alternative, then the two CPUs would be able to observe the same shared counter.

ammaree, 2021-06-14:
@igrr 

Working on the assumption, possibly wrong, that once the 2 MCU clock counters have been syncronized they remain in sync, is it technically feasible to add functionality that would ensure the 2 clocks are synchronised right at the start, when the APP MCU is started?

igrr, 2021-06-14:
It is feasible, I think. You could implement a mechanism similar to NTP: two tasks, one on each CPU, would communicate over queues or shared memory, adjusting the cycle counters to match, and compensating for roundtrip delay in communication. After that the cycle counters will stay in sync (*).

(*) with the exception of the case when dynamic frequency scaling is enabled (CONFIG_PM_ENABLE) on the ESP32. In this case, the cycle counters will go out of sync.

ammaree, 2021-07-06:
@igrr 

We were hoping that there might be a simple mechanism whereby the clock counters for both MPU's could be synchronised prior to app_main() being called, without any need for the complex logic described above. And, if the clocks are synchronised once during APP MCU startup, we are hoping that the clocks would remain in synch from then on.

From our application, and for our current needs, we are not going to implement the proposed logic but remain hopeful that the IDF team will find the correct location and hopefully a simpler mechanism to implement this functionality.

In the mean time we will close this issue, but remain hopeful.

closed 2021-07-06 by ammaree