espressif/esp-idf #7137: Update for FreeRTOS

DawidDiaco, 2021-06-11:
**Is your feature request related to a problem? Please describe.**

Hello, I would like to request an update for FreeRTOS. The current release/4.3 apears to be working on version 10.2.1, which is quite outdated at this point. In the current version of FreeRTOS they have some interesting features that I would like to use in my project.

The one that I'm really looking forward to use is ulTaskNotifyTakeIndexed function, because they claim, that this function is 45% faster than standard ISR handling by using semaphore.

https://www.freertos.org/RTOS_Task_Notification_As_Counting_Semaphore.html

**Describe the solution you'd like**

I would like to have an updated version of FreeRTOS in the closest release. I'm not sure at this moment if ESP-IDF is using just the kernel of FreeRTOS, or the whole FreeRTOS subsystem, but when I look a those releases, they look juicy as hell! 

https://github.com/FreeRTOS/FreeRTOS/releases
https://github.com/FreeRTOS/FreeRTOS-Kernel/releases

So updating FreeRTOS to version 10.4.4, or 202104.00 seems to have the most sense for now.

**Describe alternatives you've considered**

For now I'm sticking with ulTaskNotifyTake, which is fine, but if I can provide a better way to utilize my ESP, then I want to do this.

I've also considered to update freertos by my self, but my lack of experience with embeded systems may result in more problems than benefits. That's why I would like that this component got updated by experienced team. Especialy it looks like the freertos required some porting for it to work with esp.

projectgus, 2021-06-15:
Hi @DawidDiaco,

Thanks for bringing this up. Agree there are some nice features in these newer FreeRTOS releases.

ESP-IDF uses the FreeRTOS kernel, with [modifications for SMP (dual core) support](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/freertos-smp.html). Because of the kernel modifications, updating is more risky than just using the new kernel source - new features may need modifications to work correctly and careful testing to make sure they work in the SMP configuration and don't introduce subtle bugs or regressions. It's quite a lot of work to update.

That said, we will consider updating to pick up these new features. I can't give you any ETA on this, sorry.

Long term, FreeRTOS team has recently [created a development branch with an official SMP kernel](https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/smp/) and we're looking to move ESP-IDF to this, but such a change will require a major version update as the behaviour is different and some APIs will break. If/when we use the official upstream SMP kernel then updating to new upstream versions will become much simpler!

Angus

DawidDiaco, 2021-06-17:
Thank you for your explanation. I think I can live with the current version of FreeRTOS, but I will be patiently waiting for updates. 

For now you can close this ticket, or mark it as whatever is appropriate in this case.

AxelLin, 2021-07-15:
Hi @projectgus 

I notice below issues was fixed in upstream FreeRTOS.
Would you mind take a look if the esp-idf needs the fix.

https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/117
  https://github.com/FreeRTOS/FreeRTOS-Kernel/commit/82df39764a813970ba1071ac064063ea010ebed8
https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/118
  https://github.com/FreeRTOS/FreeRTOS-Kernel/commit/b9748e50eaafdb20a0c2438c0286a255c3f59e4c

I create a gist to make it easier to check the diff (without whilte space change) for above fixes:
https://gist.github.com/AxelLin/500cdc3414fac7ba291d2df54239fd6c

Dazza0, 2022-02-14:
@DawidDiaco Current IDF master (20847eeb96c486629a678d6ffbe73670013fb771) should already have updated FreeRTOS to v10.4.3. Closing this issue, please feel free to reopen.

ammaree, 2022-02-14:
@DawidDiaco
@Dazza0

Can please reopen this issue.
The ability to change ulRunTimeCounter to uint64_t is a very valuable feature for us since it will enable far longer periods of per task processor utilization trending.

Currently we have to run a routine is an idle task to detect and handle wrapping of the runtime counter, an unnecessary kludge.

Dazza0, 2022-02-15:
@ammaree Could you specific exactly which FreeRTOS version you want?

ammaree, 2022-02-15:
10.4.5 or newer

AxelLin, 2022-02-15:
> @ammaree Could you specific exactly which FreeRTOS version you want?

I think @ammaree want this feature:
https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/History.txt#L31-L34

ammaree, 2022-02-15:
Absolutely correct

https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/History.txt#L31-L34

ammaree, 2022-08-05:
@Dazza0 

Any chance of on update on this functionality?

Dazza0, 2022-08-10:
@ammaree We're still discussing whether to upgrade to v10.4.3. Our main concern right now is that v10.4.3 is an LTS version. Given that we are going to move over to the [SMP Kernel](https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/smp/), preferably want to leave the legacy kernel on an LTS version.

If this ends up being the case, then will backport specific functionality from later versions (such as the `configRUN_TIME_COUNTER_TYPE` feature in this case). However, if there are numerous functions required from v10.4.5, then we may also consider upgrading.

@ammaree do you have any preference w.r.t the two approaches mentioned above?

ammaree, 2022-08-10:
Our only requirement is for the 64bit runtime counters so you welcome to choose the option that would be the least effort and quickest.

Other users might requirement additional 10.4.5 functionality but cannot speak for them.

AxelLin, 2022-08-11:
> @ammaree We're still discussing whether to upgrade to v10.4.3. Our main concern right now is that v10.4.3 is an LTS version. Given that we are going to move over to the [SMP Kernel](https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/smp/) in the future, we preferably want to leave the legacy kernel on an LTS version.

According to the FAQ https://www.freertos.org/faq-lts.html#question2 , the support period for FreeRTOS LTS libraries is two years.
V10.4.3 was released on 15 Dec 2020, so it will reach the end of TLS support period by the end of this year.
Given such timing it's probably fine to move to newer version (or next TLS version).

AxelLin, 2022-09-18:
FYI, V10.5.0 released.
https://github.com/FreeRTOS/FreeRTOS-Kernel/releases/tag/V10.5.0

AxelLin, 2022-12-07:
> @ammaree We're still discussing whether to upgrade to v10.4.3. Our main concern right now is that v10.4.3 is an LTS version. 

@Dazza0 
FYI, New FreeRTOS Long Term Support version released
https://freertos.org/2022/10/new-freertos-long-term-support-version-released.html

ammaree, 2022-12-07:
@Dazza0 as per @AxelLin post above

+99999 for for leaving legacy kernel on the just released new FreeRTOS LTS version.

That will leave those that are remaining on v4.x.x on a version with a long lifetime ahead...
AND
It will allow those of us on v5.x to continue with the same latest version until you have completed the move to SMP kernel

AxelLin, 2023-02-01:
@Dazza0 

Just remind that the FreeRTOS LTS 202012 will be EOL by end of next month.
I think you should prepare upgrading to new LTS version ASAP to avoid using an EOL version of FreeRTOS for esp-idf-4.4+.

Cc @igrr (in case if no response from @Dazza0 )

igrr, 2023-02-01:
We will evaluate this, but since IDF 4.4.x service period is ending soon, we might end up keeping the existing version of FreeRTOS in release/v4.4, to avoid regressions or app behavior changes due to an upgrade.
Even though it will be EOL for support from upstream, we will still provide support in terms of fixes for security issues and critical issues, by backporting relevant fixes from the newer upstream releases.

AxelLin, 2023-02-03:
> We will evaluate this, but since IDF 4.4.x service period is ending soon, we might end up keeping the existing version of FreeRTOS in release/v4.4, to avoid regressions or app behavior changes due to an upgrade. 

FYI: There is no API change (compared to previous LTS version)  for the FreeRTOS Kernel part.
https://freertos.org/2022/10/new-freertos-long-term-support-version-released.html

AxelLin, 2023-04-07:
> We will evaluate this,

AFAICT, the espressif's response is too slow.
**4 Months ago** I already pointed out new FreeRTOS LTS version is available by https://github.com/espressif/esp-idf/issues/7137#issuecomment-1340461360 .
The esp-idf is using a EOL version of FreeRTOS now.

ammaree, 2023-05-29:
@Dazza0  @igrr 
I notice in master repo that we have 'FreeRTOS-Kernel-V10.5.1' there. this OK to use?
Any instructions on selecting and using this new kernel?

igrr, 2023-05-29:
Not yet, it is in progress at the moment:
https://github.com/espressif/esp-idf/blob/903af13e847cd301e476d8b16b4ee1c21b30b5c6/components/freertos/Kconfig#L6-L11

https://github.com/espressif/esp-idf/blob/903af13e847cd301e476d8b16b4ee1c21b30b5c6/components/freertos/CMakeLists.txt#L34-L35

ammaree, 2023-05-29:
Thanks @igrr 

Any estimated timeframe for having a "testable" version in master?

AxelLin, 2023-07-09:
> Not yet, it is in progress at the moment:

I'm wondering if FreeRTOS-10.5.1 fixes this issue: https://github.com/espressif/esp-idf/issues/8849

zikalino, 2023-07-14:
@AxelLin I will check
@ammaree we are actively working on this at the moment, at the end of Q3 we should have version for RISC-V chips

ammaree, 2023-07-14:
Thanks for the feedback but an estimate of Q3 for the RISC-V is of no value to us.
The requirement is for the ESP32[S2/S3] not for RISC-V.
Leaving customer still supporting the original ESP32 out for this long on 10.4.3 is very painful and sad.

AxelLin, 2023-08-15:
> @AxelLin I will check 

So how is the status? Any update?

AxelLin, 2023-09-22:
> @AxelLin I will check 

However, I don't get further response at all. @zikalino

ammaree, 2023-09-22:
@igrr 
@Dazza0 
@zikalino 

This issue is almost 20 months old, from the date of it being reopened at my request. We are STILL, 20 months later, on an LTS version of FreeRTOS that is EOL.

We accept the argument of the older version being retained with the IDF 4.4.x releases/updates, but to still be using the EOL RTOS with IDF v5.2, that is a serious problem.

Also not sure why the "focus" on upgrading the RTOS for C3 series, surely ESP32/S2/S3 should receive same if not higher priority?

Dazza0, 2023-09-22:
@AxelLin @ammaree The FreeRTOS upgrade has already been merged internally and is waiting sync to GitHub.

AxelLin, 2023-09-22:
> @AxelLin @ammaree The FreeRTOS upgrade has already been merged internally and is waiting sync to GitHub.

@Dazza0 
But that is not the question I asked, see https://github.com/espressif/esp-idf/issues/7137#issuecomment-1627668795
What I asked is if the fixes in newer FreeRTOS release fixes the bug in current FreeRTOS in esp-idf.
What I care is if current release branches are stable because the reported issue is still open.

Dazza0, 2023-09-22:
@AxelLin 

- For issue #8849, no this has not been fixed in v10.5.1 (and there is no easy way to fix this as I've described in the issue comments)
- For the other v10.6.0 fixes, we'll need to selectively backport them to v10.6.0 depending on severity.

ammaree, 2023-09-25:
> @AxelLin @ammaree The FreeRTOS upgrade has already been merged internally and is waiting sync to GitHub.

Any idea how long the delay to finally get to GitHub?
We are already another 3 days later and still nothing...

closed 2023-09-27 by espressif-bot

AxelLin, 2023-12-20:
FYI, FreeRTOS-v11.0.0 released. (SMP merged into the mainline)
https://github.com/FreeRTOS/FreeRTOS-Kernel/releases/tag/V11.0.0

AxelLin, 2024-07-02:
FYI, FreeRTOS 202406.00-LTS
https://github.com/FreeRTOS/FreeRTOS-LTS/releases/tag/202406.00-LTS

AxelLin, 2025-03-05:
FRI, FreeRTOS-v11.2.0 released.
https://github.com/FreeRTOS/FreeRTOS-Kernel/releases/tag/V11.2.0