espressif/esp-idf #7138: Subtle bug in SDSPI driver when sharing with another SPI device

khoek, 2021-06-11:
## Environment

- Development Kit:      Adafruit Huzzah32
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.2.1
- Build System:         CMake (PlatformIO)
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     Linux
- Using an IDE?: PlatformIO with vscode.
- Power Supply: USB

## Problem Description

I think I've found a subtle bug in the SDSPI interface when sharing with another SPI device, but I think I need a bit of help to understand how something is going wrong inside the SDSPI/SPI driver. (I'm connecting my ESP32 to this board (https://www.adafruit.com/product/3357), on which an SD card and VS1053 audio chip share the same SPI bus.) Note throughout that what I'm trying to do here does work with `arduino-esp32` SD-card interface.

I've looked all over message boards and issue on this repo and others, and I think there are some traces of people having a problem very similar to this after the changes to esp-idf making sharing an SDSPI device with other SPI devices were merged.

I have logic analyzer traces at the bottom while hopefully helps to understand what is happening. The problem arises in the following situation: 

1. Call `spi_bus_initialize()` to set up the bus,
2. Call `esp_vfs_fat_sdspi_mount()` to configure the SD card and mount it (which because of nuances of the SD SPI spec must be done before anything else),
3. _At this point, performing a directory listing in the mounted directory works._
4. Call `spi_bus_add_device()` to register the other device (in my case an audio chip), where the driver will manage `CS` asserting for us. Perform a single SPI transaction with `spi_device_polling_transmit()` where we exchange 48 bits on the bus.
5. _At this point, performing a directory listing in the mounted directory will always fail._ The following error is consistently emitted (this is on verbose log level):
````
V (4483) sdmmc_cmd: sending cmd slot=1 op=17 arg=4be000 flags=1c50 data=0x3ffbe7f4 blklen=512 datalen=512 timeout=1000
V (4493) sdspi_host: sdspi_host_start_command: slot=1, CMD17, arg=0x004be000 flags=0x5, data=0x3ffbe7f4, data_size=512 crc=0x1b
V bus_lock: dev 5 acquired.
D (4503) spi_master: device5 locked the bus
V bus_lock: SPI dev changed from 0 to 5
V (4513) spi_master: polling trans
V (4513) spi_master: polling trans done
V (4523) spi_master: polling trans
V (4523) spi_master: polling trans done
V (4533) spi_master: polling trans
V (4533) spi_master: polling trans done
V (4533) spi_master: polling trans
V (4543) spi_master: polling trans done
V (4543) spi_master: polling trans
V (4543) spi_master: polling trans done
V (4553) spi_master: polling trans
V (4553) spi_master: polling trans done
V (4553) spi_master: polling trans
V (4563) spi_master: polling trans done
V (4563) spi_master: polling trans
V (4573) spi_master: polling trans done
V (4573) spi_master: polling trans
V (4573) spi_master: polling trans done
D (4583) sdspi_host: poll_data_token: received 0x40 while waiting for data
V (4583) spi_master: polling trans
V (4593) spi_master: polling trans done
D (4593) spi_master: device5 release bus
V (4603) bus_lock: dev 5 released.
D (4603) sdspi_host: sdspi_host_start_command: cmd=17 error=0x108
D (4603) sdmmc_cmd: cmd=17, sdmmc_req_run returned 0x108
E (4613) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x108
E (4623) diskio_sdmmc: sdmmc_read_blocks failed (264)
D (4623) vfs_fat: vfs_fat_readdir_r: fresult=1
````

One problem is that the log line `sdspi_host: poll_data_token: received 0x40 while waiting for data` appears to be nonsense---here is a logic analyser output taken immediately after the SPI transaction to the audio chip:

[image]

On the other hand, here is a logic analyser output taken when I comment out the SPI transaction to the audio chip:

[image]

There are a few curious differences between the two:

1. Recall that in the SD SPI protocol when a `READ_SINGLE_BLOCK` command is issued we have to poll the SD card, expecting `0xFF` over and over, until we get a `0xFE`. Then the next bytes start being data. On the ESP32 there is then a much longer SPI transaction initiated by the SDSPI driver to read the rest of the waiting (512-byte block).
2. In the successful logic analyser trace you can see three "bursts"; the first is issuing the command, the second is the the first poll after the command was issued, during which at some point we got back a `0xFE`, and the third is a much longer (depicted thicker) burst where we get the rest of the data.
3. But now, in the logic analyser trace where we have problem we see multiple strange things. First, the SDSPI driver error log means that it read `0xFF`s until it encountered a `0x40` which it did not expect (it was expecting a `0xFE`). This is already very strange, because the logic analyser dump of the SPI data transmitted during the transaction:
````
FF FF FF FF FF FF FF 00 FF FF FF FF FF FF FF FF 
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
FF FF FF FF 90 5D 00 1C 80 18 40 18 C0 1A C0 03 
C0 22 8C 00 0C 00 0C 40 0B 80 1B 40 1C 00 00 00 
0C C0 00 00 15 14 90 50 D2 CC 0C 0C 53 54 0C C8 
00 0D C9 15 55 D4 55 D4 40 00 08 D5 55 D4 51 81 
AB 3E 54 80 10 5D 00 1C 80 18 40 18 C0 1A C0 03 
C0 1A 8C 00 0C 00 0C 80 0B 80 1B 40 1C 00 00 00 
0C C0 00 00 15 14 90 50 D2 CC 0C 0C 93 54 0C C8 
00 27 88 D5 55 D4 55 D4 40 00 08 D5 55 D4 40 C0 
````
does not even contain a `0x40` in an obvious place. (Note that showhow the logic analyser is has gotten confused and is one bit out of sync for this hex dump, so you actually have to left shift all of the hex dump by one bit. This is why we see `0x90` instead of `0xFE`, if we left shift by 1 everything is fixed.)

Next, and also very strange, is that by my reading of the "bad" logic analyser trace, the SDSPI driver must be reading the `0x40` during the very last burst of SPI transaction depicted before which plenty of data which was not `0xFF` was transmitted by SD card. I also have no idea why this is happening: (after reading the code, especially `poll_data_token()` in `components/driver/sdspi.c`, it seems clear that the driver will abort as soon as the `0x40` is read, only stopping to transmit an additional 8-bits of binary 1s in the abort path, to get the SD card to release MISO).

My hope is that someone who understands the SD card driver could explain what is going on. As I said above, all of this *does* work when I use the `adruino-esp32`'s SD library, which as I understand it actually reimplements a lot of the SD card SPI protocol, and this is why I believe this is a bug in `esp-idf` (if I did not have this working example I would still just be confused).

Thank you for very much for your help.

khoek, 2021-06-11:
Extra info: It seems that in the "bad" version, the `READ_SINGLE_BLOCK` transaction has twice as many clock pulses:

Bad:

[image]

Good:

[image]

khoek, 2021-06-11:
I also don't have any idea why this double-length thing is happening, since it seems to me that the code (https://github.com/espressif/esp-idf/blob/1d7068e4be430edd92bb63f2d922036dcf5c3cc1/components/driver/sdspi_host.c#L667) has a fixed length for that transaction. I suppose it seems like the "good" case is either not reading at all, or reading in full-duplex mode instead of half-duplex?

khoek, 2021-06-11:
Yes, communicating with the other device on the bus using full duplex mode fixes this problem. There is a bug inside the SPI driver handling the duplex mode somewhere.

iosix, 2021-06-11:
Yes, I submitted a fix for this to the ESP32 forum several months ago.

[image: IOSiX Logo]

Robert Vogt IV

CEO

IOSiX, LLC ᐧ www.iosix.com

1300 Tefft Ct #1

Saline MI, 48176

855.623.1939 ext. 700

khoek, 2021-06-12:
@iosix That's great to hear. Did you submit a pull request? Was the fix backported to idf v4.2?

iosix, 2021-06-12:
I’m pretty sure no one bothered to do anything with it, but here it is:

https://www.esp32.com/viewtopic.php?t=13719

Robert Vogt IV
CEO
IOSiX, LLC
1300 Tefft Ct #1
Saline, MI  48176
P: +1-855-OBD-1939
C: +1-734-730-9690
***@***.***

Alvin1Zhang, 2021-06-15:
Thanks for the detailed report, we will look into.

vamshi51, 2022-03-25:
Dear @khoek, I have created a similar test using examples\peripherals\lcd\tjpgd and examples\storage\sd_card\sdspi. Created two tasks, one for displaying image on LCD and the other for writing/read-verify patterns to a file with fat-fs. This test runs indefinitely. One of the tasks will be using SPI master for communicating with the respective device.

[image]

Please see if this helps.

Alvin1Zhang, 2022-04-12:
Thanks for reporting, would you please help share if any updates for the issue? Thanks.

pacucha42, 2022-05-02:
No response from the reporter for few weeks, closing the issue. Please, feel free to reopen if needed

closed 2022-05-02 by pacucha42