espressif/esp-idf #7139: esp_bt_hidh_init()

guo-max, 2021-06-11:
**Is your feature request related to a problem? Please describe.**
```C++
esp_err_t esp_bt_hidh_init(const esp_hidh_config_t *config)
{
    if (config == NULL) {
        ESP_LOGE(TAG, "Config is NULL");
        return ESP_FAIL;
    }
    esp_event_loop_args_t event_task_args = {
        .queue_size = 5,
        .task_name = "esp_bt_hidh_events",
        .task_priority = uxTaskPriorityGet(NULL),
        .task_stack_size = 2048,
        .task_core_id = tskNO_AFFINITY
    };
    esp_err_t ret = esp_event_loop_create(&event_task_args, &event_loop_handle);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "esp_event_loop_create failed!");
        return ret;
    }
    esp_event_handler_register_with(event_loop_handle, ESP_HIDH_EVENTS, ESP_EVENT_ANY_ID, config->callback, NULL);
    BTA_HhEnable(0, bta_hh_cb);
    return ESP_OK;
}
```
In the function 
```C++
esp_err_t esp_bt_hidh_init(const esp_hidh_config_t *config)
```
 there is a function call
```C++
 esp_event_handler_register_with(event_loop_handle, ESP_HIDH_EVENTS, ESP_EVENT_ANY_ID, config->callback, NULL);
```
And the last param "void* event_handler_arg" is given "NULL".

Because of this, the callback function is always has a "NULL" for the "void* event_handler_arg"
```C++
typedef void         (*esp_event_handler_t)(void* event_handler_arg,
                                        esp_event_base_t event_base,
                                        int32_t event_id,
                                        void* event_data); /**< function called when an event is posted to the queue */
```

**Describe the solution you'd like**

Can you add another member to the config like config->event_handler_arg and pass it to the function?

boblane1, 2021-06-15:
Hi guo-max, 

Your suggestion is reasonable, but  adding `config->event_handler_arg` will cause the change of interface. Would you mind working around the problem by software?

Thanks!

guo-max, 2021-06-15:
I need to use the callback function and the "void* event_handler_arg" argument. But the esp_bt_hidh_init function is always set it to NULL.  I do not know how to work around this problem by software except make my own hidh_init function and not use the library. 
Do you have a easier way to solve this problem?

boblane1, 2021-06-16:
Hi guo-max,

I think there are two ways to solve this problem:
1. You can just change the `esp_hidh_config_t` structure and code on your local branch, the code is open source;
2. Replace `event_handler_arg` with a global variable.

What about?

Alvin1Zhang, 2021-11-30:
@guo-max Thanks for reporting. Would you please help share if any updates or details for this feature request? Thanks.

Alvin1Zhang, 2021-11-30:
Thanks for reporting, this has been implemented in https://github.com/espressif/esp-idf/commit/80c3aaab7c8caaa5ff572bec0c3b27a2a82a4724, and will be part of IDF v4.4 release, feel free to reopen.

closed 2021-11-30 by Alvin1Zhang