espressif/esp-idf #7140: installation esp-idf on vs code

isakovM, 2021-06-11:
hello dear supporter. i am trying to install esp-idf on vs code. but at the end of the installation some problems are appearing. please have a look at this picture
[image]

dobairoland, 2021-06-21:
Hi @isakovM. Thank you for your report. We will fix this. Until then you can avoid this issue by installing `git` and making sure it is available everywhere (not just from git bash). Git can be installed with the [Windows Tool Installer](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html#esp-idf-tools-installer) or from https://git-scm.com.

brianignacio5, 2021-06-22:
Can you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/69521854.zip)?

[Installing extension from vsix in Visual Studio Code](https://code.visualstudio.com/docs/editor/extension-marketplace#_install-from-a-vsix)

isakovM, 2021-06-22:
definitely. i will try

closed 2021-06-28 by espressif-bot